evcc-io/evcc #18167: Tasmota charger device brokes evcc sometimes

luckylefti, 2025-01-10:
### Describe the bug

I use a Tasmota plug ( Nous A6T ) to control charging my car ( Volvo ).
For a while everthing runs well ( differently, sometimes minutes, sometimes hours ). 
Then somtimes i can see some errors in logs which depends on plug connection errors.
GUI is then no more reachable.
Restarting evcc works again for a while until errors appear again

This are the errors

Calling these requests manually works.
Looping ping to the device works also without problems.
So connection is not lost

### Steps to reproduce

1. Use Tasmota device for charging control
2.
3.
...

### Configuration details

```yaml
interval: 120s

mqtt:
  broker: 192.168.1.19:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: evcc
  # user:
  # password:

meters:
  - name: my_pv
    type: template
    template: sunspec-inverter
    usage: pv      
    modbus: tcpip
    id: 1
    host: 192.168.1.181 # Hostname
    port: 502 # Port 
  - name: my_grid
    type: template
    template: sunspec-inverter
    usage: grid
    host: 192.168.1.181
    port: 502
    modbus: tcpip
    id: 240
    
chargers:
  - name: my_wallbox
    type: template
    template: tasmota
    host: 192.168.1.131
    standbypower: 15
    user: admin # Standard-User ist admin (optional)
    password: XXX! 
    channel: 1 # Meterkanal (1-8)
    
vehicles:
- name: my_car
  type: template
  template: volvo2mqtt
  title: Volvo # Wird in der Benutzeroberfläche angezeigt (optional)
  capacity: 18.8 # Akkukapazität in kWh (optional)
  vin: XXXXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
  icon: car # Icon in der Benutzeroberfläche (optional)
  phases: 1 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
  mode: PV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
  minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
  maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)

loadpoints:
- title: Loadpoint1
  charger: my_wallbox
  vehicle: my_car
  mode: pv
  enable:
    # einschalten, wenn 2 Minuten lang mindestens 1500 W Überschuss vorhanden ist
    delay: 2m
    threshold: -1500
  disable:
    # ausschalten, wenn 15 Minuten lang mehr als 1500 W aus dem Netz bezogen werden
    delay: 15m
    threshold: 1500

site:
  title: My Home
  meters:
    grid: my_grid
    pv:
    - my_pv
```

### Log details

```text
[lp-1  ] ERROR 2025/01/10 08:14:41 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:46478->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/10 08:20:38 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:42330->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/10 08:32:42 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/10 08:32:51 charge rater: charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/10 08:32:51 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/10 08:32:51 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/10 08:38:42 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/10 09:14:31 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/10 09:26:30 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:40232->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/10 17:08:26 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/10 17:08:35 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/10 17:08:35 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.1

andig, 2025-01-11:
Please add full log

andig, 2025-01-11:
> So connection is not lost

According to log there‘s no answer from device.

luckylefti, 2025-01-12:
Yes, but manually query same request works and also loop ping works without erors.
Only evcc sometimes has timeouts with this device

luckylefti, 2025-01-12:
This is log until evvc brokes:
`main  ] INFO 2025/01/12 08:12:19 evcc 0.132.1
[main  ] INFO 2025/01/12 08:12:19 using config file: /app/evcc.yaml
[db    ] INFO 2025/01/12 08:12:19 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/01/12 08:12:19 connecting evcc at tcp://192.168.1.19:1883
[main  ] INFO 2025/01/12 08:12:19 listening at :7070
[site  ] INFO 2025/01/12 08:12:31 site config:
[site  ] INFO 2025/01/12 08:12:31   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2025/01/12 08:12:31     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/01/12 08:12:31     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/01/12 08:12:31   vehicles:
[site  ] INFO 2025/01/12 08:12:31     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2025/01/12 08:12:31 loadpoint 1:
[lp-1  ] INFO 2025/01/12 08:12:31   mode:        pv
[lp-1  ] INFO 2025/01/12 08:12:31   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/01/12 08:12:31   meters:      charge ✓
[lp-1  ] INFO 2025/01/12 08:12:31     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2025/01/12 08:12:31 vehicle updated: unknown -> XC90 MQTT
[lp-1  ] INFO 2025/01/12 08:12:35 car connected
[lp-1  ] ERROR 2025/01/12 08:20:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 08:20:54 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:20:54 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:32:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 08:38:44 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:38:44 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:40:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:40:49 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:40:49 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:42:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:52:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 09:04:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 09:04:54 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:04:54 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:10:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 09:24:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:32:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:56:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 10:02:44 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:41186->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:08:41 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:50166->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:14:41 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:42874->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:20:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 10:26:42 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:48904->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:32:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[... cut ...]
[lp-1  ] ERROR 2025/01/12 11:52:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 11:52:54 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 11:52:54 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout`

luckylefti, 2025-01-12:
- main  ] INFO 2025/01/12 08:12:19 evcc 0.132.1
[main  ] INFO 2025/01/12 08:12:19 using config file: /app/evcc.yaml
[db    ] INFO 2025/01/12 08:12:19 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/01/12 08:12:19 connecting evcc at tcp://192.168.1.19:1883
[main  ] INFO 2025/01/12 08:12:19 listening at :7070
[site  ] INFO 2025/01/12 08:12:31 site config:
[site  ] INFO 2025/01/12 08:12:31   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2025/01/12 08:12:31     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/01/12 08:12:31     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/01/12 08:12:31   vehicles:
[site  ] INFO 2025/01/12 08:12:31     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2025/01/12 08:12:31 loadpoint 1:
[lp-1  ] INFO 2025/01/12 08:12:31   mode:        pv
[lp-1  ] INFO 2025/01/12 08:12:31   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/01/12 08:12:31   meters:      charge ✓
[lp-1  ] INFO 2025/01/12 08:12:31     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2025/01/12 08:12:31 vehicle updated: unknown -> XC90 MQTT
[lp-1  ] INFO 2025/01/12 08:12:35 car connected
[lp-1  ] ERROR 2025/01/12 08:20:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 08:20:54 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:20:54 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:32:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 08:38:44 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:38:44 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:40:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:40:49 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:40:49 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:42:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 08:52:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 09:04:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 09:04:54 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:04:54 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:10:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 09:24:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:32:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 09:56:40 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 10:02:44 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:41186->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:08:41 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:50166->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:14:41 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:42874->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:20:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 10:26:42 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": read tcp 192.168.1.11:48904->192.168.1.131:80: read: connection reset by peer
[lp-1  ] ERROR 2025/01/12 10:32:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[... cut ...]
[lp-1  ] ERROR 2025/01/12 11:52:45 charge power: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2025/01/12 11:52:54 charge total import: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout
[lp-1  ] ERROR 2025/01/12 11:52:54 charger status: Get "http://192.168.1.131/cm?cmnd=Status+8&password=XXX%21&user=admin": dial tcp 192.168.1.131:80: i/o timeout

luckylefti, 2025-01-12:
When i ping the device i got response always under 100-200ms

andig, 2025-01-12:
Ping is IP- that‘s irrelevant. Theres no answer on HTTP which is TCP.