evcc-io/evcc #18186: Error in log : grid currents invalid syntax

MrBLJ, 2025-01-12:
### Describe the bug

Hello everyone,

For a few days now, I've been faced with a recurring error message in EVCC logs that caught my attention. The line in the log is as follow :

> [site ] ERROR 2025/01/12 13:03:20 grid currents: strconv.ParseFloat: parsing "<nil>": invalid syntax

From the error message, I think it's related to the site configuration. However using the trace I can't pinpoint where to look.

### Steps to reproduce

Just turn on EVCC, the error message appears every 1 mins approximately.

Here's what I've done so far.

1. Make sure my Envoy device is functionnal. I can access the web UI and the app to see real-time production.
2. Make sure my Envoy token is valid. I can use the token to login into the web UI.
3. Make sure that the services I use for my cars are fonctionnal. Teslamate, my ble proxy and the proxy for the wall connectors are all working as expected.

I'm a bit lost, since I truly don't know where to look. Any help on that regard would be greatly appreciated.

### Configuration details

```yaml
site:
  - title: Home
    meters:
      grid: enphaseGrid
      pv: enphasePV
    residualPower: 0

mqtt:
  broker: localhost:1883
  topic: evcc
  user: user
  password: password
  
meters:
  - name: enphaseGrid
    type: template
    template: enphase
    usage: grid
    host: 192.168.1.94
    token: token
 
  - name: enphasePV
    type: template
    template: enphase
    usage: pv
    host: 192.168.1.94
    token: token

interval: 30s

tariffs:
  currency: EUR
  feedin:
    type: fixed
    price: 0.10
  grid:
    type: custom
    forecast:
      source: mqtt
      topic: tempo/grid/forecast

chargers:
  - name: modelY_charger
    type: template
    template: twc3
[... 115 lines cut ...]
[influx] TRACE 2025/01/12 13:03:16 write pvEnergy=map[value:3096.278222] (map[id:1])
[influx] TRACE 2025/01/12 13:03:16 write pvExcessDCPower=map[value:0] (map[id:1])
[influx] TRACE 2025/01/12 13:03:16 write pvCapacity=map[value:<nil>] (map[id:1])
[influx] TRACE 2025/01/12 13:03:16 write pvSoc=map[value:<nil>] (map[id:1])
[influx] TRACE 2025/01/12 13:03:16 write pvControllable=map[value:<nil>] (map[id:1])
[enphaseGrid] TRACE 2025/01/12 13:03:18 {"production":[{"type":"inverters","activeCount":8,"readingTime":1736683625,"wNow":1067,"whLifetime":3147923},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1736683642,"wNow":407.682,"whLifetime":3096278.458,"varhLeadLifetime":1114579.864,"varhLagLifetime":773384.062,"vahLifetime":4362258.002,"rmsCurrent":2.068,"rmsVoltage":229.812,"reactPwr":-147.255,"apprntPwr":478.18,"pwrFactor":0.85,"whToday":2526.458,"whLastSevenDays":11110.458,"vahToday":3982.002,"varhLeadToday":892.864,"varhLagToday":1033.062,"lines":[{"wNow":407.682,"whLifetime":3096278.458,"varhLeadLifetime":1114579.864,"varhLagLifetime":773384.062,"vahLifetime":4362258.002,"rmsCurrent":2.068,"rmsVoltage":229.812,"reactPwr":-147.255,"apprntPwr":478.18,"pwrFactor":0.85,"whToday":2526.458,"whLastSevenDays":11110.458,"vahToday":3982.002,"varhLeadToday":892.864,"varhLagToday":1033.062}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1736683642,"wNow":711.229,"whLifetime":12418646.722,"varhLeadLifetime":1605695.379,"varhLagLifetime":319513.973,"vahLifetime":12633209.985,"rmsCurrent":3.933,"rmsVoltage":229.722,"reactPwr":-288.612,"apprntPwr":903.605,"pwrFactor":0.79,"whToday":24101.722,"whLastSevenDays":210372.722,"vahToday":24762.985,"varhLeadToday":677.379,"varhLagToday":3687.973,"lines":[{"wNow":711.229,"whLifetime":12418646.722,"varhLeadLifetime":1605695.379,"varhLagLifetime":319513.973,"vahLifetime":12633209.985,"rmsCurrent":3.933,"rmsVoltage":229.722,"reactPwr":-288.612,"apprntPwr":903.605,"pwrFactor":0.79,"whToday":24101.722,"whLastSevenDays":210372.722,"vahToday":24762.985,"varhLeadToday":677.379,"varhLagToday":3687.973}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1736683642,"wNow":303.547,"whLifetime":9322663.062,"varhLeadLifetime":2720275.242,"varhLagLifetime":1092898.035,"vahLifetime":12633209.985,"rmsCurrent":1.866,"rmsVoltage":229.722,"reactPwr":-141.357,"apprntPwr":428.612,"pwrFactor":0.71,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":303.547,"whLifetime":9322663.062,"varhLeadLifetime":2720275.242,"varhLagLifetime":1092898.035,"vahLifetime":12633209.985,"rmsCurrent":1.866,"rmsVoltage":229.722,"reactPwr":-141.357,"apprntPwr":428.612,"pwrFactor":0.71,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
[enphaseGrid] TRACE 2025/01/12 13:03:18 GET http://192.168.1.94/production.json?details=1
[enphaseGrid] TRACE 2025/01/12 13:03:18 <html> <head><title>301 Moved Permanently</title></head> <body> <center><h1>301 Moved Permanently</h1></center> <hr><center>openresty/1.17.8.1</center> </body> </html>
[enphaseGrid] TRACE 2025/01/12 13:03:18 GET https://192.168.1.94/production.json?details=1
[enphaseGrid] TRACE 2025/01/12 13:03:20 {"production":[{"type":"inverters","activeCount":8,"readingTime":1736683625,"wNow":1067,"whLifetime":3147923},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1736683644,"wNow":398.213,"whLifetime":3096278.69,"varhLeadLifetime":1114579.947,"varhLagLifetime":773384.062,"vahLifetime":4362258.274,"rmsCurrent":2.022,"rmsVoltage":230.084,"reactPwr":-142.888,"apprntPwr":467.021,"pwrFactor":0.84,"whToday":2526.69,"whLastSevenDays":11110.69,"vahToday":3982.274,"varhLeadToday":892.947,"varhLagToday":1033.062,"lines":[{"wNow":398.213,"whLifetime":3096278.69,"varhLeadLifetime":1114579.947,"varhLagLifetime":773384.062,"vahLifetime":4362258.274,"rmsCurrent":2.022,"rmsVoltage":230.084,"reactPwr":-142.888,"apprntPwr":467.021,"pwrFactor":0.84,"whToday":2526.69,"whLastSevenDays":11110.69,"vahToday":3982.274,"varhLeadToday":892.947,"varhLagToday":1033.062}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1736683644,"wNow":709.937,"whLifetime":12418647.134,"varhLeadLifetime":1605695.38,"varhLagLifetime":319513.973,"vahLifetime":12633210.238,"rmsCurrent":3.9,"rmsVoltage":230.004,"reactPwr":-287.248,"apprntPwr":897.11,"pwrFactor":0.79,"whToday":24102.134,"whLastSevenDays":210373.134,"vahToday":24763.238,"varhLeadToday":677.38,"varhLagToday":3687.973,"lines":[{"wNow":709.937,"whLifetime":12418647.134,"varhLeadLifetime":1605695.38,"varhLagLifetime":319513.973,"vahLifetime":12633210.238,"rmsCurrent":3.9,"rmsVoltage":230.004,"reactPwr":-287.248,"apprntPwr":897.11,"pwrFactor":0.79,"whToday":24102.134,"whLastSevenDays":210373.134,"vahToday":24763.238,"varhLeadToday":677.38,"varhLagToday":3687.973}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1736683644,"wNow":311.724,"whLifetime":9322663.243,"varhLeadLifetime":2720275.327,"varhLagLifetime":1092898.035,"vahLifetime":12633210.238,"rmsCurrent":1.878,"rmsVoltage":230.004,"reactPwr":-144.36,"apprntPwr":431.971,"pwrFactor":0.72,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":311.724,"whLifetime":9322663.243,"varhLeadLifetime":2720275.327,"varhLagLifetime":1092898.035,"vahLifetime":12633210.238,"rmsCurrent":1.878,"rmsVoltage":230.004,"reactPwr":-144.36,"apprntPwr":431.971,"pwrFactor":0.72,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
[site ] ERROR 2025/01/12 13:03:20 grid currents: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site ] DEBUG 2025/01/12 13:03:20 site power: 262W
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.321.1

VolkerK62, 2025-01-12:
you have a 1-phase system?
With 132.1. there was a change regarding currents. https://github.com/evcc-io/evcc/pull/17193

Please try, if 132.0 is working.

MrBLJ, 2025-01-12:
Thanks a lot for your quick answer. i'm indeed using 0.132.1 (sorry for the typo in the OP I'll fix that). I'll try downgrading to 0.132.0.

andig, 2025-01-12:
/cc @pdeliot

pre5ton, 2025-01-13:
Experiencing same issue after upgrading to 0.132.1. Not using Home Automation, but do have an Enphase system.

pdeliot, 2025-01-13:
@ivoks 
I think the code is returning <nil> instead of 0.

`if .consumption[] | select(.measurementType == "net-consumption").activeCount >= 1 then .consumption[] | select(.measurementType == "net-consumption").lines[1].rmsCurrent else 0 end`

ivoks, 2025-01-13:
@pdeliot yes. As mentioned (https://github.com/evcc-io/evcc/pull/17193#issuecomment-2511005056), you should really check for the length of the array before assuming that the lines[1] or lines[2] exist.

ivoks, 2025-01-13:
Let me know if you want to work on this. I have some time today and I can come up with a patch. I don't have CTs, so I can't test it on a live system, but I have some JSONs from other cases that can be of a use for testing purposes.