evcc-io/evcc #18212: Plan: improve next plan selection

naltatis, 2025-01-13:
Currently the next plan is picked by plan time. This is suboptimal, when a plan with low target (30% at 8:00) is followed by a higher plan (100% at 9:00).

_Originally posted by @Balujuhu in https://github.com/evcc-io/evcc/discussions/17746_

andig, 2025-01-13:
Wie realistisch ist das? Tatsächlich müssten alle Pläne evaluiert werden um dann den mit dem frühesten Start auszuwählen. Das könnte evtl. auch der Algorithmus für den übergelaufenen Plan sein. Da gilt es analog. Falls aber der „nächste“ Plan noch nicht begonnen hat (erster Slot in der Zukunft) und das Ladeziel noch nicht erreicht ist dann gilt weiter der letzte Plan als „laufend“.

andig, 2025-01-14:
> Ist durchaus realistisch

I see. Passiert wohl am ehesten, wenn sie wiederkehrende und Einzelpläne überschneiden?

> Das verstehe ich noch nicht. 

Das Problem "welcher Plan ist der nächste" (Algovorschlag oben) haben wir "rückwärts" für die übergelaufenen Pläne auch,  also "welcher Plan ist eigentlich übergelaufen"?

Hier müssen wir uns den aktiven Plan entweder merken oder (mein Favorit) könnten wir es uns einfacher machen und schlicht sagen "wenn der letzte Plan nicht erfüllt wurde und kein neuer (zukünftiger) Plan aktiv ist, dann wählen wir den zeitlich (!) letzten Plan als aktiven, übergelaufenen Plan aus". Dieser Plan bestimmt dann das Planziel. 

Es kann dann aber- wie in die Zukunft auch- passieren, dass z.B. ein 80% Plan bis 7:00 Uhr durch einen 20% Plan bis 7:30 übersteuert wird und wir damit bei einem Ladeziel von 20% landen. Das _richtig_ zu machen geht nur wenn wir uns das alte Ladeziel merken (dann kennen wir aber nicht die Ursprungszeit für die Anzeige) oder den alten Plan selbst.

andig, 2025-01-14:
Ich habe mal versucht, die Szenarien aufzumalen:

[image]

naltatis, 2025-01-14:
Find ich eine guten Idee weil wir dann nicht zusätzlichen State brauchen. 
Wir könnten auch ein festes Interval (max 1-2h) in die Vergangenheit schauen, wenn wir uns im "Overrun" Zustand befinden.
Diese Diskussion gehört aber hier ins Ticket: https://github.com/evcc-io/evcc/issues/17704

closed 2025-01-15 by naltatis