evcc-io/evcc #18223: Laden von kleinen disconnected Batteries

hanborns, 2025-01-14:
Hi,

mittels Shelly Plug S habe ich versucht neben den eBikes auch den Staubsauger und Akkuschrauber mit PV-Strom zu laden, aber noch wichtiger ist mir dass man die Akkus nicht kaputtlädt. Während bei den eBike Akkus evcc die Kapazität (750 bzw. 625 Wh) in sinnvolle prozentuale Einheiten unterteilt funktioniert die Logik bei kleinen Akkus (Staubsauger hat 63 Wh, der Akkuschrauber sogar nur 27 Wh) nicht. Ich fände es klasse, wenn die Logik auch für kleine Akkus anwendbar wäre ...

Danke!

Viele Grüße, Hanno

andig, 2025-01-14:
Aktuell nicht geplant

closed 2025-01-14 by andig

naltatis, 2025-01-14:
Ist in der Tat ein eher nieschiger Anwendungsfall. Über die API kannst du beim Energy Limit schon heute beliebige Werte setzen. Ich hab in https://github.com/evcc-io/evcc/pull/18226 eine kleine UI Anpassung gemacht mit der die Skala im UI bis auf 5W Schritte runter geht.

Kenne mich mit er Effizienz von Haushaltsakkus nicht aus, aber ich vermute, dass hier Ladeverluste noch mal ne ganz andere Rolle spielen und das alles schnell relativ ungenau werden kann.