evcc-io/evcc #18224: Nicht CO2 sparend laden bloss weil CO2-Berechnung konfiguriert ist

hanborns, 2025-01-14:
Hi,
wenn man die CO2-Berechnung einschaltet, wird automatisch CO2-sparend geladen. Ich fände es gut, wenn man das separat einschalten könnte, falls man zwar den CO2-Ausstoss wissen will, aber nicht CO2-basierend laden will.
Eine Alternative wäre CO2 sparend zu laden, aber die letzten n% dann vor der Abfahrt nachzuladen, so dass der Akku für die Abfahrt warm ist und die Standbeheizung nicht aus der Traktionsbatterie erfolgen muss.
Vielen Dank und viele Grüße, Hanno

StefanSchoof, 2025-01-14:
Ähnlich zu https://github.com/evcc-io/evcc/issues/10950 (was in https://github.com/evcc-io/evcc/issues/11654 aufgenommen wurde)

SolarPower2024, 2025-01-14:
Wenn du nur die Co2 Werte einpflegen willst für die Darstellung und Berechnung der Einsparung, dann funktioniert das bereits.
Einfach die "free tier" Variante von electricitymap wie in der Doku beschrieben einpflegen.
Der free tier gibt keine Werte der Zukunft aus, wird also nicht für die Planung verwendet. Lediglich die live Daten werden hier zur Verfügung gestellt und berücksichtigen.
Habe ich auch so laufen.

naltatis, 2025-01-15:
Ja, den Wunsch einen Teil der Ladung auf "kurz vor Abfahrt" zu legen kann ich verstehen. Ist, wie @StefanSchoof schreibt auch auf der Verbesserungsliste für den Planer.

Weiterer Workaround für deinen konkreten Fall ist das explizite setzten eines `planner` Tarifs. Du kannst parallel zu `grid` und `co2` auch `planner` als Tariff konfigurieren. Dieser Eintrag wird dann vom Planungsalgorithmus und in der UI (Netzladen) benutzt. Beispiel:

```yaml
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.294
  co2:
    ...
  planner:
    type: fixed
    price: 0.294
```

Aus der Dokumentation haben wir das rausgenommen. Gut möglich, dass wir den Mechanismus später durch etwas besseres (weniger redundantes) ersetzen.

closed 2025-01-15 by naltatis

vh2002, 2025-03-14:
Der Workaround ist eine super Möglichkeit, weil ich dasselbe Problem hatte. Solange es nicht geplant ist, wäre der Gedanke gar nicht schlecht, es doch wieder in die Doku mit aufzunehmen, denn da habe ich auch zunächst nach einer Möglichkeit gesucht.