evcc-io/evcc #18227: Dynamic pricing broken when scheduling for next day

bsmout, 2025-01-14:
### Describe the bug

I'm charging using dynamic tariffs and I see the pricing until midnight. However when I try to schedule to charge for tomorrow morning pricing is still unknown, even after restarting EVCC. I'm using ENTSOE but also tried with energy-charts API.

### Steps to reproduce

1. Schedule chargeing for tomorrow 7:00
2. Pricing unknown
3. 
...

### Configuration details

```yaml
currency: EUR
grid:
    type: entsoe
    domain: BZN|BE
    securitytoken: ****
```

### Log details

```text
No log errors
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

RenatusRo, 2025-01-14:
This depends on the moment of the day.
After 14u the prices after midnight will be known and charging will be scheduled as expected.

bsmout, 2025-01-14:
Priced are updated at 13:00. At 16:00 I still get the message that prices are not known for the next day.

closed 2025-01-14 by andig