evcc-io/evcc #18230: Anzeige der Ladevorgänge nicht korrekt

Micha70, 2025-01-14:
### Describe the bug

Mehrfach habe ich schon festgestellt, dass die Anzeige der Ladevorgänge nicht korrekt ist. Hier z.B. in der Zeile vom 12.1.05
[image]
Seltsamerweise gleiche Lademenge 35,2kWh wie am 10.1.. Interessanterweise mit 23,2kW bei 100% Sonne geladen, meine PV Anlage hat eine Leistung von 5,7kW ;-)

Wenn ich die beiden Vorgänge einzeln anzeige: 10.1:
[image]
Ist plausibel, Zählerstand passt auch.

Der Fehler ist am 12.1.:
[image]
Der Zählerstand ist plausibel, ich habe per PV ca. 2,2kWh geladen. Aber die geladene Energiemenge 35,2kWh wird vom Vortag übernommen und für die Berechnung der Kosten herangezogen....

### Steps to reproduce

Ladevorgang durchführen

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: warn
  lp-1: warn
  lp-2: warn
  cache: warn
  db: warn
  ocpp: warn

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# javascript:
#   - vm: shared
#     script: |
#       state = {
#         residualpower: 500,
#         pvpower: 5000,
#         batterypower: -750,
#         batterySoc: 55,
#         gridpower: -1000,
#         loadpoints: [
#           { enabled: true, vehicleSoc: 62, maxcurrent: 6, phases: 1, chargepower: 0 },
#           { enabled: false, vehicleSoc: 22, maxcurrent: 0, phases: 3, chargepower: 0 }
#         ]
#       };
#       function logState() {
#         console.log("state:", JSON.stringify(state));
#       }

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle

      
chargers:

  - name: my_charger
    type: template
    template: sungrow
    # Modbus TCP
    modbus: tcpip
    # modbus: rs485tcpip
    id: 248
    # id: 2
    host: 192.168.178.112 # Hostname
    port: 502 # Port
    
    

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:

# site describes the EVU connection, PV and home battery
site:
#   title: Home # display name for UI
#   meters:
#     grid: e3dc-grid # grid meter
#     pv: e3dc-pv # list of pv inverters/ meters
#     battery: e3dc-battery # list of battery meters
#     aux:
#     #   - aux # list of auxiliary meters for adjusting grid operating point
    residualPower: 100 # additional household usage margin
#   maxGridSupplyWhileBatteryCharging: 50 # ignore battery charging if AC consumption is above this value

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    circuit: main
    #charger: wallbe # charger
    charger: my_charger
    mode: "pv" # default charge mode to apply when vehicle is disconnected; use "off" to disable by default if charger is publicly available
    #meter: charge # charge meter
    phases: 0

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      #threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
      threshold: -3000       # einschalten, wenn 1 Minute lang mindestens 2000 W Überschuss vorhanden ist

    disable: # pv mode disable behavior
      # geaendert 15.06.2024  delay: 3m # threshold must be exceeded for this long
      delay: 3m
      #threshold: 0 # maximum import power (W)
      threshold: 1000 # ausschalten, wenn 3 Minuten lang mehr als 2000 W aus dem Netz bezogen werden

    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: XXXXXX
  password: XXXXXXXX
  

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

hems:
  type: relay
  maxPower: 4200 # single SteuVE
  limit:
     source: mqtt
     #topic: monitor-rst-empfaenger/binary_sensor/rse_leistungseinschraenkung/state
     topic: esphome/wallbox/leistungseinschraenkung
     
circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hauptstromkreis # Anzeige in der UI (später)
    # maxCurrent: 63 # 63A (optional)
    # maxPower: 30000 # 30kW (optional)
    # meter: grid # optional

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
```

### Log details

```text
Kann man ein log von vor zwei Tagen generieren?
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.1

zachelnet, 2025-01-14:
Kein es sein das in deiner Konfiguration kein Wechselrichter definiert ist?

zachelnet, 2025-01-14:
Intressantes Phanomen, anschein kommt die Calucaltion bei Tagwechseln durcheinander. Siehe dein erster Screenshot da ist der Zeitraum vom Fr. 10 Jan 10:32 auf Sa. 11 Jan 07:05, rein rechnerish sollten das 20:03h gewesen sein. Irgendwas passt da von hinten und vorne nicht mit den Werten.

andig, 2025-01-14:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

Micha70, 2025-01-15:
@andig is possible to generate this logfile now, or do I need todo this directly if problem appeared? How long data is stored in the past?

Micha70, 2025-01-15:
I was able now to create a log file by 
`docker logs 23fc9a55a0fd > ~/evcc_20250115.log`
but assume is "outdated".

[evcc_20250115.log](https://github.com/user-attachments/files/18419759/evcc_20250115.log)

If I see topic again, will try to generate another log.

andig, 2025-01-15:
We'll need a trace log for `db` to see which queries produce the result you're seing.

Micha70, 2025-01-15:
got it now:

[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 10:19:50.26" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 10:19:50.26" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 10:19:50.26" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 10:19:50.26" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 10:19:50.267" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 10:19:50.267" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 10:19:50.267" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 10:19:50.267" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-31 10:19:50.27" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-31 10:19:50.27" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-31 10:19:50.27" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-31 10:19:50.27" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "1925-02-09 10:19:50.273" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "1925-02-09 10:19:50.273" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "1925-02-09 10:19:50.273" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 10:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "1925-02-09 10:19:50.273" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 11:19:50.286" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 11:19:50.286" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 11:19:50.286" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-16 11:19:50.286" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 11:19:50.295" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 11:19:50.295" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 11:19:50.295" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-01-16 11:19:50.295" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "2024-12-31 11:19:50.299" AND charged_kwh > 0 AND solar_percentage IS NOT NULL 1 <nil>
[... cut ...]
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "1925-02-09 11:19:50.302" AND charged_kwh > 0 AND price_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:19:50 SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0) FROM sessions WHERE finished >= "1925-02-09 11:19:50.302" AND charged_kwh > 0 AND co2_per_kwh IS NOT NULL 1 <nil>
[db ] TRACE 2025/01/15 11:21:34 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created DESC 141 <nil> 

Hope this helps. Or do you need this shortly after the problem appeared?

andig, 2025-01-15:
Looking into this I think we need a log of the charging session that produces the wrong log entry. Once its in the database it's as recorded.

Micha70, 2025-01-15:
Ok, once I reproduce it I will provide new data. 
Is possible to extent the storage time-range of the logs?

Micha70, 2025-01-19:
@andig today again charged in PV mode, several hours, about 5kWh. But is not shown in GUI

[image]
 and not in charging overview:

[image]

Logs are available now:

[evcc-20250119-144810-trace.log](https://github.com/user-attachments/files/18469006/evcc-20250119-144810-trace.log)

[evcc-20250119-144714-debug.log](https://github.com/user-attachments/files/18469007/evcc-20250119-144714-debug.log)

WIll you reopen this ticket or shall I create a new one?

andig, 2025-01-19:
@naltatis willst Du Dir das anschauen?

andig, 2025-01-19:
> Seltsamerweise gleiche Lademenge 35,2kWh wie am 10.1.. Interessanterweise mit 23,2kW bei 100% Sonne geladen, meine PV Anlage hat eine Leistung von 5,7kW ;-)

> today again charged in PV mode, several hours, about 5kWh. But is not shown in GUI

Hier gehts bunt durcheinander: was ist das _konkrete_ Problem mit dem letzten Ladevorgang?

Micha70, 2025-01-19:
ich habe gestern und heute geladen. Gestern 7,8kwh, in der Übersicht vorhanden. 
Heute auch mehrer Stunden geladen.
der Start ist leider nicht im logfile ....
[lp-1 ] INFO 2025/01/19 14:46:20 stop charging <- 
Aber der Ladevorgang erscheint nicht in der Übersicht ....

andig, 2025-01-19:
> Aber der Ladevorgang erscheint nicht in der Übersicht ....

Zu welchem Timestamp im Logfile wird die Übersicht aufgerufen? Ich sehe da nichts.

Micha70, 2025-01-19:
wie meinst du dass? Muss ich die Ladevorgangsübersicht aufrufen und danach das Logfile generieren ?

Habe ich gemacht:

[image]

und Logfile:

[evcc-20250119-160957-trace.log](https://github.com/user-attachments/files/18469252/evcc-20250119-160957-trace.log)

Aber der Ladevorgang von heute fehlt komplett ....

andig, 2025-01-19:
Ich sehe den Fehler:

```
[db    ] TRACE 2025/01/19 14:46:20 UPDATE `sessions` SET `created`="2025-01-19 10:49:50.593",`finished`="2025-01-19 14:46:20.518",`loadpoint`="Garage",`identifier`="",`vehicle`="IONIQ 5",`odometer`=12826.2,`meter_start_kwh`=1661.079,`meter_end_kwh`=1667.689,`charged_kwh`=0,`charge_duration`="3h56m30s",`solar_percentage`=0,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 159 1 <nil>
```

konkret

    `charged_kwh`=0

@Micha70 hast Du dazu bitte noch das vollständige Logfile ab evcc Start?

andig, 2025-01-19:
@naltatis ich verstehe die Logik nicht:

```go
	if chargedEnergy := lp.getChargedEnergy() / 1e3; chargedEnergy > s.ChargedEnergy {
		lp.sessionEnergy.Update(chargedEnergy)
	}
```

aber

```go
func (lp *Loadpoint) getChargedEnergy() float64 {
	lp.RLock()
	defer lp.RUnlock()
	return lp.sessionEnergy.TotalWh()
}
```

ist das nicht ein Zirkelbezug? Ich habe das Gefühl, das würde der Roundtrip über die EnergyMetrics dazu führen, als würden hier Werte verschluckt.

naltatis, 2025-01-19:
> Interessanterweise mit 23,2kW bei 100% Sonne geladen, meine PV Anlage hat eine Leistung von 5,7kW ;-)

@Micha70 Hast du einen Hausspeicher? Ist aus dem geladen worden? In der Config stehen keine Meter.

andig, 2025-01-19:
Lasst uns doch bitte erstmal beim Thema bleiben- fehlender Ladevorgang. Analyse siehe oben.

Micha70, 2025-01-19:
> Ich sehe den Fehler:
>
> ```
> [db    ] TRACE 2025/01/19 14:46:20 UPDATE `sessions` SET `created`="2025-01-19 10:49:50.593",`finished`="2025-01-19 [...]

Logfile ist oben im ersten Kommentar von heute....

Micha70, 2025-01-20:
@andig , hatte heute wieder gleiches Problem. Gestern Abend über Netz auf 80% geladen, erscheint korrekt in der Übersicht der Ladevorgänge. Heute weitere 8% über PV geladen, wurde nicht aufgezeichnet..... kein Eintrag in der Logvorgängen...

Benötigt ihr noch weitere Logdaten? Kann man die Länge der Aufzeichnung des Logfiles konfigurieren? Ich sehe immer nur die letzten 2h ....

andig, 2025-01-20:
> Logfile ist oben im ersten Kommentar von heute....

Wo? Du meinst nicht die 20 Zeilen?!

Micha70, 2025-01-20:
Sorry, mein Fehler. Ich dachte ich hätte das Log in https://github.com/evcc-io/evcc/issues/18230#issuecomment-2600871515 reingehangen.... Aber scheint nicht hochgeladen worden zu sein. Bin diese Woche auch unterwegs und habe keinen Zugriff auf das Logfile.

Heute habe ich beim Laden folgendes bemerkt: 

[image]

Es wird mit 2,0kW geladen über PV Überschuss, aber die 22,2kWh und die 6,02€ sind vom Laden gestern über Netz.... Ist das so korrekt?

Werden die Logfiles irgendwo abgelegt? EVCC läuft im Container. Und in der Logfiles sieht man immer nur die letzten ca. 2h....

Micha70, 2025-01-24:
Hier nochmal einen Screenshot von heute:

[image]

[image]

und das dazugehörige Logfile, hoffe es funktioniert:
[evcc-20250124-145316-trace.log](https://github.com/user-attachments/files/18536565/evcc-20250124-145316-trace.log)

Ich kann mir folgendes vorstellen. Ich hatte am 20.01 eine Ladevorgang über PV, ca 7kW. Der wurde nicht aufgezeichnet.
Den Ladevorgang von heute hatte ich beendet nach ca. 1Min, da mit 3 anstatt 1 Phase geladen worden ist. Scheinbar hat die Wallbox nicht reagiert. 
Wurden dann die 7kW vom 20.1 mit eingerechnet und man bekommt durch die Dauer der Ladesession von 1min auf die stattliche Ladeleistung von 298,2kWh? (Ich hatte wirklich kein Kraftwerk zum Laden angeschlossen ;-)

Micha70, 2025-01-24:
ok, jetzt wurde er Ladevorgang bendet:

[image]

Die Ladeleistung angepasst:

[image]

Die Daten stimmen aber trotzdem nicht:

[image]

Nach Zählerstand wurden 1,9kWh geladen. Aber lasut Tabelle 7,5kWh.

Hier noch das Logfile dazu: 
[evcc-20250124-155456-trace.log](https://github.com/user-attachments/files/18537807/evcc-20250124-155456-trace.log)

@andig , sollten erstmal genug Daten sein. Ich hoffe ihr findet das Problem.... Lasst mich wissen, wenn noch was benötigt wird.

andig, 2025-01-26:
ping @naltatis

BastiZ0, 2025-01-28:
Moin!

Mir geht es ähnlich. Es scheint, als "Verschluckt" sich EVCC mit dem Beenden/Starten von Ladevorgängen.. 

Wenn ich zum Beispiel geladen habe und das Auto abstecke um es nach einer kurzen Fahrt wieder anzuschließen und zu Laden, gibt er mir auf dem Dashboard zwar an, dass eine gewisse Leistung anliegt, aber der Zähler auf der Übersichtsseite zählt nicht. Normalerweise sollte der Ladevorgang ja "abgeschlossen" werden, wenn das Auto abgesteckt wird. Dies macht er aber nicht. Wenn ich es nach 2h wieder anstecke, gibt er mir die alte geladene Energie wieder im Dashboard an.

Ich nutze seit kurzem eine workaround mit einer "Automation", welche EVCC neu startet vor dem Ladevorgang und nach dem abstecken des Fahrzeugs, damit der Zähler wieder 0 kWh zeigt und der Ladevorgang in der Übersicht erscheint. Dies funktioniert "fast" immer. Es kommt trotzdem vor, dass EVCC nicht "aufzeichnet" und es nicht in die Übersicht schreibt bzw die Statistik füttert oder wie oben beschrieben doppelte Einträge macht.

Leider ist das Auto gerade nicht zum Laden bereit, somit kann ich jetzt gerade keine Logs, wo der Fehler auftritt, anhängen.
Würde aber gerne hier unterstützen. Was bräuchtet ihr von mir, damit ihr es einsehen könnt.

steps56, 2025-02-02:
Bei mir das Gleiche.
Seit 30.1.25 werden n Evcc alle 4 weiteren bisherigen Ladevorgänge zu einem und am 31.1. zusammengefasst.
Die Auflistung zeigt also für Feb. 25 keine Ladevorgänge an, obwohl es alleine im Feb. 2 gab.
Die letzten gezeigten Ladevorgänge sind Alle summiert dann falsch dem 30.1. zugeordnet und zeigen in Summe (38,8 kWh) alle seit 30. 1. gemachten Ladungen.
Ob das erst seit dem 30.1
 und/oder dem letzten Evcc Update so ist, keine Ahnung.
Dürfe bei Allen so sein und verwunderlich, dass es hier dazu nicht spammt betr. Meldungen.
Aber fiel mir eben ja auch erst auf, da ich für solche Anzeigen die Entitäten von der Evcc Integration nutze  da euer Evcc Addon (das top ist !) da leider ohne Tricks keine Entitäten bereit stellt. 🤷🏼‍♂️

steps56, 2025-02-03:
Hier noch 3 Screens, die das falsche Datum/Ladewerte zeigen:

[image]
[image]
[image]

steps56, 2025-02-04:
Erledigt bei mir, war ein Fehler von mir.
Details betr. Zuordnung stimmen, nur der Balken im Diagramm ballert Alles über Tage auf den Anfangszeitpunkt an dem der Stecker das 1. Mal im Auto steckt.
Das ist auf den ersten und 2. Blick verwirrend, wenn man nicht in die Details sieht, weil oben zum neuen Monat steht "kein Ladevorgang", obwohl bis 2.2. z. B. geladen wurde.
Aber egal, wenn man es weiß, Alles okay.
Für mich nun erledigt. 🙋🏼‍♂️

andig, 2025-02-20:
@naltatis wie machen wir hier weiter?

apahlke, 2025-02-26:
Moin, ich hab das gleiche Phänomen:
Beim Laden über Nacht wird am nächsten Tag zu viel Energie angezeigt. Ich vermute, dass einmal der Zähler der Wallbox abgefragt wird, der den gesamten Ladevorgang zurück meldet, während die Auswertung die Tagesgrenze einbezieht.

[image]

apahlke, 2025-02-26:
Das ist aus den Ladevorgängen der Punkt: Download csv

[image]

apahlke, 2025-02-27:
@andig: Aber ist das nicht genau das Thema? Ich hab das Auto am 20. geladen, aber nicht vollständig. Am nächsten Tag lädt dann noch etwas über die PV dazu, aber die Lademenge des Vortags wird dazu gerechnet. Anzeige am 21.: geladen über 20 kw statt tatsächlich 1,9

andig, 2025-02-27:
Hab ich anders verstanden. Vllt. Kann @Micha70 das Problem nochmal präzisieren.

steps56, 2025-02-27:
> Hab ich anders verstanden. Vllt. Kann [@Micha70](https://github.com/Micha70) das Problem nochmal präzisieren.

Die Evcc Programmierer nuten doch auch Evcc, oder nicht?
Wenn ja, sieht das doch jeder, denn das dürfte generell so sein.
Gerade das Angesteckt lassen an der Wallbox ist ja normaler use case... und dabei tritt das Addieren am Vortag auf.
Natürlich übrrhaupt erst, wenn Abgesteckt wird.
Solange Angeschlosden = 1 Ladevorgang am Anstecktag.
Warum das erklärt werden muss....?.... sieht doch jeder schon ewig so.
Darum dies ggf. zu korrigieren, geht es hier, ist null O. T. !

andig, 2025-02-27:
> Hab ich anders verstanden.

steps56, 2025-02-27:
> > Hab ich anders verstanden.

Ist ja nicht schlimm, ketzt aber klar, oder?

Ob dies so gewollt ist bzw. gar nicht anders geht (in Evcc), weiß ich nicht.
Spontan fällt mir dazu ein (bin kein Programmierer), dass 2 Parameter zum Erkennen des Tages/Ladevorgangs dienen könnten:
1.
Datum
2.
Wird um Mitternacht noch geladen = fließt Strom.

Wenn 2. = nein, dann Loggen als abgeschlossener Ladevorgang, auch wenn Ladekabel noch angesteckt ist.
Momentan wird Alles in Evcc zu einem LDevorgang adfiert und wenn z. B. am 30. begonnen und am 1. abgesteckt, gibt es im neuen Monat null Ladevorgänge, da Alles am 30. summiert gelistdt wird, aber auch erst NACH dem Abstecken des Kabels.

Ich denke es gut verständlich und  deutlich erläutert zu haben.
Bitte einfach mal ausprobieren, verwunderlich dass dies nicht jedem als komisch/störend auffiel, gerade Leuten, die da auf den Tag genau abrechnen müssen mit Fa. o. Ä.
(Ist Alles konstruktiv gemeint, aber Vieles wird gerade auf Github leider oft schlicht weggebügelt aus Erfahrung.)

andig, 2025-02-27:
>  ketzt aber klar, oder?

Ja, gehört hier aber nicht her. Das oben beschriebene Problem ist:

> Seltsamerweise gleiche Lademenge 35,2kWh wie am 10.1.

Also falsche Mengen. Dein Problem (Zuordnung zum falschen Tag ist im Moment so wie es ist, gerne separates Thema dazu). Hier geht es um richtige Mengen.

naltatis, 2025-02-27:
@steps56 Ladevorgänge werden aktuell bewusst nicht künstlich für die Aufzeichnung unterbrochen. Wir bilden hier den physischen Ladevorgang (Kabel einstecken bis Kabel abstecken) ab und bilden auch alle Metriken dementsprechend.

Bei der grafischen Auswertung wählen wir die Startzeit für für die Tag/Monat/Jahr Zuordnung. Mit der Unschärfe muss man, zumindest in dieser Ansicht, leben. Eine künstliche "Verdopplung"/Aufsplittung aller Ladevorgänge führt auch zu Verzerrungen.  Rein zeitbasierte Metriken stehen auf der Roadmap: https://github.com/evcc-io/evcc/issues/13050

apahlke, 2025-02-27:
@naltatis Hmm, wenn ich die csv für Abrechnungen nutzen müsste, wäre der Preis falsch ausgewiesen. Die Energie vom Vortag wird mit eingerechnet.

andig, 2025-02-27:
...und genau das ist der Fehler um den es heir geht!

apahlke, 2025-02-27:
Dann hier nochmal mein Hinweis ;)
Beim Laden über Nacht wird am nächsten Tag zu viel Energie angezeigt. Ich vermute, dass einmal der Zähler der Wallbox abgefragt wird, der den gesamten Ladevorgang zurück meldet, während die Auswertung die Tagesgrenze einbezieht.

Micha70, 2025-02-27:
Ich habe jetzt nochmal nachgeschaut. Ladevorgang heute tagsüber ( Netzanteil 93%), Aufzeichnung korrekt (Zählerstand passt zu Ladeleistung):

[image]

Ladevorgang gestern Nachmittag gestartet, heute Morgen beendet. 100% Sonne kann nicht sein. Laut Zählerstand 15,2kWh geladen ... laut Summe oben 36kWh, passt nicht. Ladedauer 2:19h mit 15,7kW ... Ich habe nur ein 11kw Wallbox. Und wenn angeblich 100% Sonne dann müsste ich ja ein PV Anlage mit 15,7kWh  Leistung haben. Habe aber bis jetzt nur 5,7kWh auf dem Dach ...

[image]

An anderen Tagen funktioniert es wiederum, auch wenn über den Tageswechsel geladen wird. Bsp:

[image]

BastiZ0, 2025-02-28:
Um mich auch noch mal einzuklinken. Habe eben den Ladevorgang nochmal gestartet, um das Phänomen zu reproduzieren. Auto hängt seit gestern Abend an der Leitung und wurde nur kurz für 400Wh geladen. Jetzt wieder gestartet. Poweranzeige zeigt direkt die Ladeleistung. Lademenge wird aber wieder nicht aktualisiert. Im Protokoll steht auch, dass der Vorgang anscheinend um 7:12 Uhr beendet ist. Es ist allerdings auch schon 7:15Uhr gewesen. Ich hatte das Anfang Februar schon einmal, aber keine Zeit mich darum zu kümmern. Eine halbe Stunde nach Start des Ladevorgangs ist dann auch irgendwann die Anzeige des Ladeverbrauchs und damit alle anderen Werte wie Preis, Sonne etc erst aktualisiert worden.

[image]

[image]

[image]

Jetzt nach 10 Minuten Laden fängt er an auch die Lademenge weiterzuzählen.

[image]

andig, 2025-02-28:
Ich wiederhole mich: Screenshots sind nutzlos. Es braucht Logfiles.

naltatis, 2025-02-28:
Für die Eingrenzung des Problems wäre ein größere Datenbasis hilfreich. Können mir mal alle, die das Problem beobachten einen CSV Export der Ladevorgänge (gerne Gesamt) schicken. Bitte **mit Angabe welcher Charger** hier zum Einsatz kommt.

Ich kann in meinem Ladelog auch erkennen, dass es teilweise Abweichungen zwischen Start/Endzählerstand und der geladenen Menge gibt. Sofern die Wallbox das anbietet nehmen wir für die Berechnung immer das Geladene Energie-Feld der Wallbox als Datengrundlage. Bei einigen Wallboxen (z.b. meiner Easee Home) wird der Zählerstand und die Energie nur sehr unregelmäßig und nicht synchron (teilweise 30min Versatz) von der Box (bzw. Cloud) aktualisiert. Das ist keine Erklärung für alle Effekte, die hier beschrieben werden, aber ggf. für einige.

Daher der Wunsch nach einer breiteren Datenbasis.

steps56, 2025-02-28:
Hast gerade eine Email von mir mit session und Feb. csv Dateien.
Daheimladen smart 11kW Wallbox

Micha70, 2025-02-28:
@naltatis , habe dir auch gerade eine Nachricht gesendet, mit den Ladedaten von Januar und Februar. Ich benutze als Wallox Sungrow AC011E-01 angebunden über Modbus. 

@andig , da das Problem ja scheinbar über Nacht auftritt, wie kann man denn ein längeres Log ziehen? Im Log ist doch immer nur die letzte Stunde sichtbar? Gibt es da eine Möglichkeit?

naltatis, 2025-02-28:
Wir evcc bei euch nachts ggf. neu gestartet (bspw. software update)?

Micha70, 2025-02-28:
Nein, ich habe keinen automatischen Restart....

Micha70, 2025-02-28:
Weiss aber nicht, ob ich eventuell EVCC upgedatet hab. Werd drauf achten. Aber während Laden aktiv ist, führe ich kein Update durch. Eventuell war das Fahrzeug noch angesteckt...

steps56, 2025-02-28:
> Wird evcc bei euch nachts ggf. neu gestartet (bspw. software update)?

Nein.
Alle Updates starte ich nachdem Info dazu kommt, immer manuell, denn sehr oft startet danach Evcc nicht selbst, sondern muss unter Addons manuell neu gestartet werden. (bin ich schon lange so gewohnt, daher für mich normal).

steps56, 2025-02-28:
> [@andig](https://github.com/andig) , da das Problem ja scheinbar über Nacht auftritt, wie kann man denn ein längeres Log ziehen? Im Log ist doch immer nur die letzte Stunde sichtbar? Gibt es da eine Möglichkeit?

Ein vorherigen Log kann man siehe Screen (@andig bitte weg gucken 😘) ansehen und ggf. kopieren/speichern:

[image]

BastiZ0, 2025-02-28:
@naltatis : Mail ist auch jetzt raus. Ebenfalls Sungrow AC011E-01 über Modbus an HA. 

@andig : sorry, wollte erstmal das Problem darstellen und ob das hier überhaupt reinpasst.

Protokoll mach ich gern, leider geht anscheinend nicht mehr als 10k Einträge, also etwa 2h.   Hätte ich heute Morgen mal dran denken sollen....
Werde versuchen es zeitnahe zu reproduzieren und dann direkt die Logs zu ziehen. 

@steps56 : irgendwie bekomme ich diese Anzeige wie bei dir mit dem "vorheriger" nicht oder ich bin völlig falsch. Bei Protokolle und EVCC kann ich wie gesagt nur die letzten 10.000 Einträge herunterladen. Falls es noch einen anderen Weg gibt, bitte helfen. :)

andig, 2025-02-28:
@BastiZ0 bitte lass das. Wir brauchen Logfiles und Fehlerbeschreibungen. Sonst absolut gar nichts es sei denn es wäre angefragt. Danke.

BastiZ0, 2025-02-28:
Gern! Sagst du mir noch kurz, ob das richtig ist mit den max 10.000 Einträgen oder bin ich da auf dem Holzweg...sonst muss ich mich hier leider ausklinken, weil ich da dann nicht helfen kann und mit dem Problem leben....

Danke! :)

steps56, 2025-02-28:
Genau !
Denn es wäre weit konstruktiver, wen  die Wissenden (@Andig mainly) die im Dunklen tastenden mal an die Hand nehmen würden, um ihnen zu erklären (mich eingeschlossen), wo man wie und welche der gewünschten Logs denn überhaupt habhaft werden kann.
Ist das so scher zu kapieren ???
Destruktiv ist, nach Logs zu fragen, die niemand liefern kann, mangels gewußt wo, wie...
Mann, Mann, Mann... kotzt mich deftig an so ein Gehabe.
Hatten wir doch schon mal, also mal runter vom Turm.
Nun kann mich hier sperren wer will, aber der Tritt war bitter nötig und sitzt hoffentlich 😎☝️.
Alle Anderen sind da konstruktiver, aber @Andig sollte sich da optimieren, deutlichst ☝️.

BastiZ0, 2025-03-01:
@andig heute morgen kurz nochmal getestet. Fing wieder später an zu zählen. Dieses mal allerdings schon nach wenigen Minuten. 

Ich habe jetzt ein DEBUG und ein TRACE log. Welches brauchst du und wie kann ich es dir zur Verfügung stellen?

andig, 2025-03-01:
> Fing wieder später an zu zählen

Nochmal: hier geht es darum, dass die Energie des Vortags am nächsten Tag _wieder_ mit gezählt wird. Wenn Du ein anderes Problem hast bitte neues Issue. Wenn es genau das Problem ist dann bitte debug log mit DB auf trace und Beschreibung was konkret falsch ist.

[23 more replies]

closed 2025-04-12 by andig