evcc-io/evcc #18243: EVCC sometimes can't start charging Tesla Car

martin-eberle, 2025-01-16:
### Describe the bug

Sometimes EVCC cannot start the charging process for a Tesla vehicle. Unfortunately, I have not yet recognized a pattern that could be related to the problem. It occurs quite often and without changing anything in the configuration it suddenly works again the next day.

EVCC then displays the message “Ready to charge. Waiting for vehicle ...” („Ladebereit. Warte auf Fahrzeug …“).
The house battery is full, the PV provides more than enough power, but everything is fed into the grid for hours without the Tesla connected to the wallbox starting the charging process.

I am using the latest version of EVCC (v0.132.1) with two TWC3 and three Tesla Cars. This Issue occurs with all cars randomly.

But since it sometimes works again without any problems, it is obviously not due to a wrong token or something similar.

Haven't tested with latest nightly build since I didn't find out how to install it with docker.

### Steps to reproduce

1. Connect car to wallbox
2. House battery is full, the PV provides more than enough power
3. Charging doesn't start, all power is fed into the grid
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
- type: template
  template: rct-power
  usage: grid
  host: 192.168.2.24
  name: grid1
- type: template
  template: rct-power
  usage: pv
  host: 192.168.2.24
  name: pv2
- type: template
  template: rct-power
  usage: battery
  host: 192.168.2.24
  name: battery3

chargers:
- type: template
  template: twc3
  host: 192.168.2.88
  name: wallbox4
- type: template
  template: twc3
  host: 192.168.2.51
  name: wallbox5

loadpoints:
- title: Wallbox-Garage
  charger: wallbox4
  mode: pv
- title: Wallbox-Haus
[... 185 lines cut ...]
[lp-1 ] DEBUG 2025/01/16 11:09:34 charge currents: [0 0 0.1]A
[twc3 ] TRACE 2025/01/16 11:09:34 {"contactor_closed":false,"vehicle_connected":false,"session_s":0,"grid_v":232.2,"grid_hz":50.106,"vehicle_current_a":0.2,"currentA_a":0.0,"currentB_a":0.2,"currentC_a":0.0,"currentN_a":0.2,"voltageA_v":0.0,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.9,"pcba_temp_c":9.6,"handle_temp_c":4.4,"mcu_temp_c":15.1,"uptime_s":240219,"input_thermopile_uv":-122,"prox_v":0.0,"pilot_high_v":11.8,"pilot_low_v":11.8,"session_energy_wh":34134.602,"config_status":5,"evse_state":1,"current_alerts":[49],"evse_not_ready_reasons":[4,8]}
[lp-2 ] DEBUG 2025/01/16 11:09:34 charge power: 0W
[lp-2 ] DEBUG 2025/01/16 11:09:34 charge currents: [0 0.2 0]A
[site ] DEBUG 2025/01/16 11:09:34 battery 1 power: -582W
[site ] DEBUG 2025/01/16 11:09:35 grid power: -5581W
[site ] DEBUG 2025/01/16 11:09:35 battery 1 soc: 99%
[site ] DEBUG 2025/01/16 11:09:35 pv 1 power: 8080W
[site ] DEBUG 2025/01/16 11:09:35 site power: -6163W
[lp-1 ] DEBUG 2025/01/16 11:09:35 charge voltages: [1.7 0 0]V
[lp-1 ] DEBUG 2025/01/16 11:09:35 charger status: B
[lp-1 ] DEBUG 2025/01/16 11:09:35 pv charge current: 8.93A = 0A + 8.93A (-6163W @ 3p)
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.1

andig, 2025-01-16:
Zu welchem Zeitpunkt im Log passiert das?

martin-eberle, 2025-01-17:
> Zu welchem Zeitpunkt im Log passiert das?

Wie ist das gemeint? Der Zustand bleibt so bestehen, irgendwann geht es dann plötzlich wieder.

Aber, da ich nun gestern noch auf die nightly Version gewechselt bin, muss ich es nun wohl eh so einrichten wie hier (https://github.com/evcc-io/evcc/discussions/17501#discussioncomment-11620698) beschrieben?

Das versuche ich nun zuerst mal, ob das an der Sache etwas ändert.

closed 2025-01-17 by andig