evcc-io/evcc #18250: evcc hängt sich auf, wenn Ladeplan existiert

naltatis, 2025-01-16:
### Describe the bug

Vmtl. eine Regression von https://github.com/evcc-io/evcc/pull/18211

Auszug aus `/debug/pprof`
```
goroutine 998 [sync.RWMutex.Lock, 1 minutes]:
sync.runtime_SemacquireRWMutex(0xc001a4b401?, 0x30?, 0x23369e2?)
	runtime/sema.go:105 +0x25
sync.(*RWMutex).Lock(0x5935560?)
	sync/rwmutex.go:153 +0x6a
github.com/evcc-io/evcc/core.(*Loadpoint).maxActivePhases(0xc0018cb008)
	github.com/evcc-io/evcc/core/loadpoint_phases.go:105 +0xf3
github.com/evcc-io/evcc/core.(*Loadpoint).EffectiveMaxPower(0xc0018cb008)
	github.com/evcc-io/evcc/core/loadpoint_effective.go:205 +0x28
github.com/evcc-io/evcc/core.(*Loadpoint).nextVehiclePlan(0xc0018cb008)
	github.com/evcc-io/evcc/core/loadpoint_effective.go:85 +0x5a9
```

### Steps to reproduce

1. Ladeplan (statisch oder wiederholend) an einem Fahrzeug anlegen (getestet mit Tesla)
2. Keine Updates mehr im Log

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-16:
EffectiveMaxPower früher abfragen?

naltatis, 2025-01-17:
@andig ist dieser Lock überhaupt notwendig?
https://github.com/evcc-io/evcc/blob/fix/gok/core/loadpoint_plan.go#L65-L66

andig, 2025-01-17:
Fehler gefunden. Da muss ein lock in rlock verwandelt werden.

closed 2025-01-17 by andig