evcc-io/evcc #18259: EVBox ELVI : charger out of sync: expected disabled, got enabled

jassi0001, 2025-01-17:
### Describe the bug

I have an EVBox which works over OCPP as expectet. I often get the Info in the Log and as Error "charger out of sync: expected disabled, got enabled" No Car is connected at that moment. When i Connect the car, everything works as expectet...

### Steps to reproduce

Starting EVCC, 
Waiting for OCPP Connection
doing nothing and looking at the Log

### Configuration details

```yaml
loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    vehicle: my_car # default vehicle
    meter: my_charger1
    mode: minpv # charge mode (off, now, minpv, pv)
meters:
  - name: my_charger1
    type: template
    template: shelly-pro-3em
    usage: charge
    host: 192.168.176.88 # IP-Adresse oder Hostname

chargers:
  - name: my_charger
    type: template
    template: elvi
    stationid: EVB-P21XXXX
    meter: false # optional
    connecttimeout: 5m # optional
```

### Log details

```text
[lp-1 ] WARN 2025/01/17 16:06:20 charger out of sync: expected disabled, got enabled
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.1

andig, 2025-01-17:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-01-17 by andig

jassi0001, 2025-01-17:
attached the Log...

andig, 2025-01-17:
Impossible to diagnose without the trace log messages for the evbox during that time.

jassi0001, 2025-01-17:
shure.... can you tell how i can get the ocpp logs?

mfuchs1984, 2025-01-17:
Maybe have a look at this discussion 
https://github.com/evcc-io/evcc/discussions/17208

jassi0001, 2025-01-18:
dann nehme ich es mal so hin. Es hat, wie im anderen Artikel ja keine Auswirkung auf die Funktion....

mfuchs1984, 2025-01-18:
Dem Fehler bzgl. 90 Ampere Current.Offered am besten auch nochmal an den Hersteller melden.