evcc-io/evcc #18263: Load management stopping charge instead of throttling down

brainstormi, 2025-01-18:
### Describe the bug

Load management seems not to work properly when initiating a new charge. Multiple times it stops the charge instead of throttling it down although there is capacity available in the circuit to charge the cars. The process enters in a loop where it start and stops the charge during some tries until one of them it allows it to continue charging. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
site:
  title: Home
  circuit: main 
meters:
    grid: grid
    pv:
      - pv

circuits: # load management  peakshaving
  - name: main
    title: "main circuit"
    maxPower: 5700 # W  mx watts
    meter: grid

loadpoints:
  - title: Garage # display name for UI
    charger: TeslaBLE # charger reference
    vehicle: Kitt # reference to standard vehicle
    mode: pv # charge mode (off, now, minpv, pv)
    circuit: main
```

### Log details

```text
[lp-1 ] INFO 2025/01/18 09:12:20 car connected
[lp-1 ] DEBUG 2025/01/18 09:12:20 pv timer elapse
[lp-1 ] DEBUG 2025/01/18 09:12:20 pv timer inactive
[lp-1 ] DEBUG 2025/01/18 09:12:20 vehicle soc: 59%
[lp-1 ] DEBUG 2025/01/18 09:12:20 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/01/18 09:12:20 vehicle range: 238km
[circuit-main] DEBUG 2025/01/18 09:12:20 validate power: 2944.1W + (0W -> 2990W) > 5700W capped at 2755.9W
[lp-1 ] DEBUG 2025/01/18 09:12:20 max charge current: 11A
[lp-1 ] DEBUG 2025/01/18 09:12:20 charger enable
[lp-1 ] DEBUG 2025/01/18 09:12:20 wake-up timer: start
[site ] DEBUG 2025/01/18 09:12:35 ----
[lp-1 ] DEBUG 2025/01/18 09:12:35 charge power: 0W
[circuit-main] DEBUG 2025/01/18 09:12:35 power: 4798.6W
[circuit-main] DEBUG 2025/01/18 09:12:35 current: 0A
[site ] DEBUG 2025/01/18 09:12:35 grid power: 4799W
[site ] DEBUG 2025/01/18 09:12:35 pv 1 power: 154W
[site ] DEBUG 2025/01/18 09:12:35 site power: 4799W
[lp-1 ] DEBUG 2025/01/18 09:12:35 vehicle odometer: 17218km
[lp-1 ] DEBUG 2025/01/18 09:12:35 charger status: B
[circuit-main] DEBUG 2025/01/18 09:12:35 validate power: 4798.6W + (0W -> 2990W) > 5700W capped at 901.4W
[lp-1 ] DEBUG 2025/01/18 09:12:35 charger disable
[lp-1 ] DEBUG 2025/01/18 09:12:35 wake-up timer: stop
[site ] DEBUG 2025/01/18 09:12:50 ----
[lp-1 ] DEBUG 2025/01/18 09:12:50 charge power: 0W
[circuit-main] DEBUG 2025/01/18 09:12:50 power: 2143W
[circuit-main] DEBUG 2025/01/18 09:12:50 current: 0A
[site ] DEBUG 2025/01/18 09:12:50 grid power: 2143W
[site ] DEBUG 2025/01/18 09:12:50 pv 1 power: 144W
[site ] DEBUG 2025/01/18 09:12:50 site power: 2143W
[lp-1 ] DEBUG 2025/01/18 09:12:50 charger status: B
[lp-1 ] DEBUG 2025/01/18 09:12:50 max charge current: 13A
[... 233 lines cut ...]
[lp-1 ] DEBUG 2025/01/18 09:17:35 charge power: 0W
[circuit-main] DEBUG 2025/01/18 09:17:35 power: 2545W
[circuit-main] DEBUG 2025/01/18 09:17:35 current: 0A
[site ] DEBUG 2025/01/18 09:17:35 grid power: 2545W
[site ] DEBUG 2025/01/18 09:17:35 pv 1 power: 168W
[site ] DEBUG 2025/01/18 09:17:35 site power: 2545W
[lp-1 ] DEBUG 2025/01/18 09:17:35 charger status: B
[site ] DEBUG 2025/01/18 09:17:50 ----
[lp-1 ] DEBUG 2025/01/18 09:17:50 charge power: 0W
[circuit-main] DEBUG 2025/01/18 09:17:50 power: 5194W
[circuit-main] DEBUG 2025/01/18 09:17:50 current: 0A
[site ] DEBUG 2025/01/18 09:17:50 grid power: 5194W
[site ] DEBUG 2025/01/18 09:17:50 pv 1 power: 168W
[site ] DEBUG 2025/01/18 09:17:50 site power: 5194W
[lp-1 ] DEBUG 2025/01/18 09:17:50 charger status: B
[circuit-main] DEBUG 2025/01/18 09:17:50 validate power: 5194W + (0W -> 2990W) > 5700W capped at 506W
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

VolkerK62, 2025-01-18:
not sure, if this causing the behaviour, but `circuit: main` within the `site` is not necessary

mfuchs1984, 2025-01-18:
Your log always says `charge power: 0W` although the charger seems to be charging. Maybe this leads to the stop, because evcc just sees circuit power going up, so there is nothing remaining for charging?

andig, 2025-01-18:
> capped at 477.8W

Charging starts at 1.4kW- 477W thorttling simply isn‘t possible.

closed 2025-01-18 by andig

brainstormi, 2025-01-18:
> > capped at 477.8W
> 
> Charging starts at 1.4kW- 477W thorttling simply isn‘t possible.

That is not right... charging point capacity is 5.7kw, before starting charging current load is 3.1kw in the worst case, so there is buffer available to start the charge without stopping it. Indeed the maximum load with car charger power included it's 5.2kw. 
It's the calculus that seems wrong. I have 2 power meters, 1 for the grid and 1 for pv. It seems that the car charging power it's calculated as part of the grid load and that's why it's stopping the charge... but this is only at the beginning of the charge, after some tries starting and stopping it, it runs successfully... 
Could you reopen the issue or do you prefer I open a new one?.

brainstormi, 2025-01-18:
> not sure, if this causing the behaviour, but `circuit: main` within the `site` is not necessary

With and without it, the issue is the same...

brainstormi, 2025-01-18:
> Your log always says `charge power: 0W` although the charger seems to be charging. Maybe this leads to the stop, because evcc just sees circuit power going up, so there is nothing remaining for charging?

I only have 2 power meters, 1 for the grid and 1 for pv matching use case in documentation, so not sure why it logs 0w.

mfuchs1984, 2025-01-18:
`[lp-1 ] DEBUG 2025/01/18 09:15:35 charge power: 0W`
`[circuit-main] DEBUG 2025/01/18 09:15:35 validate power: 4990.7W + (0W -> 2990W) > 5700W capped at 709.3W`

Available power for charging is 5700 - grid power. When charge power is 0 but grid power 4990.7, then power remaining for charging is 709,3. This is not enough to start (or continue) charging.

I don't know anything about the Tesla ble an what charger you use, so I don't know where charge power should come from in your setup.

andig, 2025-01-18:
We don‘t need an issue. Your mincurrent is 6A hence 1.4kW

brainstormi, 2025-01-18:
> We don‘t need an issue. Your mincurrent is 6A hence 1.4kW

Please, at least let me give you a short log snippet from the log attached, that maybe helps to clarify why that statement is not totally true and progress with this conversation:

[circuit-main] DEBUG 2025/01/18 09:16:35 power: 2094.5W
[circuit-main] DEBUG 2025/01/18 09:16:35 current: 0A
**[site ] DEBUG 2025/01/18 09:16:35 grid power: 2094W**
[site ] DEBUG 2025/01/18 09:16:35 pv 1 power: 158W
[site ] DEBUG 2025/01/18 09:16:35 site power: 2094W
[lp-1 ] DEBUG 2025/01/18 09:16:35 charger status: B
[lp-1 ] DEBUG 2025/01/18 09:16:35 max charge current: 13A
[lp-1 ] DEBUG 2025/01/18 09:16:35 charger enable
**[lp-1 ] DEBUG 2025/01/18 09:16:35 wake-up timer: start**
[site ] DEBUG 2025/01/18 09:16:50 ----
[lp-1 ] DEBUG 2025/01/18 09:16:50 charge power: 0W
[circuit-main] DEBUG 2025/01/18 09:16:50 power: 2094.5W
[circuit-main] DEBUG 2025/01/18 09:16:50 current: 0A
[site ] DEBUG 2025/01/18 09:16:50 grid power: 2094W
[site ] DEBUG 2025/01/18 09:16:50 pv 1 power: 180W
[site ] DEBUG 2025/01/18 09:16:50 site power: 2094W
[lp-1 ] DEBUG 2025/01/18 09:16:50 charger status: B
[site ] DEBUG 2025/01/18 09:17:05 ----
[lp-1 ] DEBUG 2025/01/18 09:17:05 charge power: 0W
**[circuit-main] DEBUG 2025/01/18 09:17:05 power: 5202.7W**
[circuit-main] DEBUG 2025/01/18 09:17:05 current: 0A
[site ] DEBUG 2025/01/18 09:17:05 grid power: 5203W
[site ] DEBUG 2025/01/18 09:17:05 pv 1 power: 162W
[site ] DEBUG 2025/01/18 09:17:05 site power: 5203W
[lp-1 ] DEBUG 2025/01/18 09:17:05 charger status: C
[lp-1 ] INFO 2025/01/18 09:17:05 start charging ->
**[lp-1 ] DEBUG 2025/01/18 09:17:05 wake-up timer: stop**
[lp-1 ] DEBUG 2025/01/18 09:17:05 soc estimated: 59.02% (vehicle: 59.00%)
[lp-1 ] DEBUG 2025/01/18 09:17:05 vehicle soc: 59%
[lp-1 ] DEBUG 2025/01/18 09:17:05 vehicle soc limit: 70%
[lp-1 ] DEBUG 2025/01/18 09:17:05 vehicle range: 238km
**[circuit-main] DEBUG 2025/01/18 09:17:05 validate power: 5202.7W + (0W -> 2990W) > 5700W capped at 497.3W**

The grid power consumption just before start charging it's 2KW
`[site ] DEBUG 2025/01/18 09:16:35 grid power: 2094W`
`[lp-1 ] DEBUG 2025/01/18 09:16:35 max charge current: 13A`
Charge is started:
`[lp-1 ] DEBUG 2025/01/18 09:16:35 wake-up timer: start`
Grid power consumption after charge started: 5.2KW     =  2KW grid + 3KW car power consumption+0.2KW from varying grid power conditions just at that second.
`[circuit-main] DEBUG 2025/01/18 09:17:05 power: 5202.7W`
Charge is stopped because in those 5.2KW, evcc doesn't account that it's already included the 3KW of the car power consumption:
`[lp-1 ] DEBUG 2025/01/18 09:17:05 wake-up timer: stop`
`[circuit-main] DEBUG 2025/01/18 09:17:05 validate power: 5202.7W + (0W -> 2990W) > 5700W capped at 497.3W`
So in reality in the main circuit remains available 497.3W after counting grid+car power consumption... but the charge is stopped.

This is the loop, until suddenly it's able to start charging without stopping it and from that point it automatically adjust the charging amps without no issue, after random number of initial tries.

I'm not pointing that this could be a bug, probably could be an issue with my configuration... but only happens when starting the charge, once it has started stable, I can increase the grid power consumption from that point, for example powering on the oven and the load management works totally fine, adjusting charging amps or stopping when it's really needed. This is why it's so weird. 

If your prefer not to reopen and that I open a discussion topic better, let me know it.

Regarding why EVCC at that time reports charging power is 0W... frankly I don't know how it's calculated by evcc:
`[lp-1 ] DEBUG 2025/01/18 09:15:35 charge power: 0W`
I only have two 'physical' power meters, 1 for grid and 1 for pv.
TeslaBLE it's a custom charger based in esphome-teslable, a library that allows you with ESP32 board via bluetooth to adjust charging amps, wake-up and start/stop the charge. In custom charger/vehicle definition I can't find a way in the doc to define/get charging power, only maxcurrent:
`
    maxcurrent: # 
      source: http
      uri: "http://localhost:8123/api/services/number/set_value"
      method: POST
      body: '{"value":"{{.maxcurrent}}","entity_id":"number.teslable_charging_amps"}'
      headers:
        - content-type: application/json
`

andig, 2025-01-18:
Logs are useless without context. Can you add the last 2 cycles before log starts?

brainstormi, 2025-01-18:
> Logs are useless without context. Can you add the last 2 cycles before log starts? It shouldn‘t enable at 2kW grid power.

But it does...
Example, full log, disconnecting the cable and connecting back... This time it only does it once, as said the number of tries it's random... from 0 to a LOT... what it's clear it's hard for it to identify the current charge power and differentiate it from the grid power.

This time we start with a 2.4KW grid load and although that it throtles down to 10A although it's not needed, then stops the charge, because it thinks there is no capacity available until 5.7KW (min charging amps 5A - max 13A). It's clear it doesn't differentiate properly until this point the grid load from the car charger power. 
After automatically resuming it in the next cycle, charging throttles down to 5A, then 9A, again 5A (None of these are needed)... and suddenly 13A. Here everything becomes "stable" and from this point load managment does any needed adjustment properly without issues.

[lp-1 ] INFO 2025/01/18 21:08:20 car disconnected
[lp-1 ] DEBUG 2025/01/18 21:08:20 set battery boost: false
[lp-1 ] DEBUG 2025/01/18 21:08:20 set session soc limit: 0
[lp-1 ] DEBUG 2025/01/18 21:08:20 set session energy limit: 0
[lp-1 ] DEBUG 2025/01/18 21:08:20 charger disable
[lp-1 ] DEBUG 2025/01/18 21:08:20 wake-up timer: stop
[site ] DEBUG 2025/01/18 21:08:35 ----
[lp-1 ] DEBUG 2025/01/18 21:08:35 charge power: 0W
[circuit-main] DEBUG 2025/01/18 21:08:35 power: 2335.2W
[circuit-main] DEBUG 2025/01/18 21:08:35 current: 0A
[site ] DEBUG 2025/01/18 21:08:35 grid power: 2335W
[site ] DEBUG 2025/01/18 21:08:35 pv 1 power: 6W
[site ] DEBUG 2025/01/18 21:08:35 site power: 2335W
[lp-1 ] DEBUG 2025/01/18 21:08:35 vehicle odometer: 17320km
[lp-1 ] DEBUG 2025/01/18 21:08:35 charger status: A
[site ] DEBUG 2025/01/18 21:08:50 ----
[lp-1 ] DEBUG 2025/01/18 21:08:50 charge power: 0W
[circuit-main] DEBUG 2025/01/18 21:08:50 power: 2322.3W
[circuit-main] DEBUG 2025/01/18 21:08:50 current: 0A
[site ] DEBUG 2025/01/18 21:08:50 grid power: 2322W
[site ] DEBUG 2025/01/18 21:08:50 pv 1 power: 6W
[site ] DEBUG 2025/01/18 21:08:50 site power: 2322W
[lp-1 ] DEBUG 2025/01/18 21:08:50 charger status: B
[lp-1 ] INFO 2025/01/18 21:08:50 car connected
[lp-1 ] DEBUG 2025/01/18 21:08:50 vehicle soc: 37%
[lp-1 ] DEBUG 2025/01/18 21:08:50 vehicle soc limit: 70%
[lp-1 ] DEBUG 2025/01/18 21:08:50 vehicle range: 161km
[lp-1 ] DEBUG 2025/01/18 21:08:50 max charge current: 13A
[lp-1 ] DEBUG 2025/01/18 21:08:50 charger enable
[lp-1 ] DEBUG 2025/01/18 21:08:50 wake-up timer: start
[site ] DEBUG 2025/01/18 21:09:05 ----
[lp-1 ] DEBUG 2025/01/18 21:09:05 charge power: 0W
[circuit-main] DEBUG 2025/01/18 21:09:05 power: 3198.2W
[circuit-main] DEBUG 2025/01/18 21:09:05 current: 0A
[site ] DEBUG 2025/01/18 21:09:05 grid power: 3198W
[site ] DEBUG 2025/01/18 21:09:05 pv 1 power: 6W
[site ] DEBUG 2025/01/18 21:09:05 site power: 3198W
[lp-1 ] DEBUG 2025/01/18 21:09:05 vehicle odometer: 17320km
[lp-1 ] DEBUG 2025/01/18 21:09:05 charger status: B
[circuit-main] DEBUG 2025/01/18 21:09:05 validate power: 3198.2W + (0W -> 2990W) > 5700W capped at 2501.8W
[lp-1 ] DEBUG 2025/01/18 21:09:05 max charge current: 10A
[site ] DEBUG 2025/01/18 21:09:20 ----
[lp-1 ] DEBUG 2025/01/18 21:09:20 charge power: 0W
[circuit-main] DEBUG 2025/01/18 21:09:20 power: 4661.6W
[circuit-main] DEBUG 2025/01/18 21:09:20 current: 0A
[site ] DEBUG 2025/01/18 21:09:20 pv 1 power: 6W
[site ] DEBUG 2025/01/18 21:09:20 grid power: 4662W
[site ] DEBUG 2025/01/18 21:09:20 site power: 4662W
[lp-1 ] DEBUG 2025/01/18 21:09:20 charger status: C
[lp-1 ] INFO 2025/01/18 21:09:20 start charging ->
[lp-1 ] DEBUG 2025/01/18 21:09:20 wake-up timer: stop
[lp-1 ] DEBUG 2025/01/18 21:09:20 soc estimated: 37.00% (vehicle: 37.00%)
[lp-1 ] DEBUG 2025/01/18 21:09:20 vehicle soc: 37%
[lp-1 ] DEBUG 2025/01/18 21:09:20 vehicle soc limit: 70%
[lp-1 ] DEBUG 2025/01/18 21:09:20 vehicle range: 161km
[circuit-main] DEBUG 2025/01/18 21:09:20 validate power: 4661.6W + (0W -> 2990W) > 5700W capped at 1038.4W
[lp-1 ] DEBUG 2025/01/18 21:09:20 charger disable
[lp-1 ] DEBUG 2025/01/18 21:09:20 wake-up timer: stop
[site ] DEBUG 2025/01/18 21:09:35 ----
[lp-1 ] DEBUG 2025/01/18 21:09:35 charge power: 0W
[circuit-main] DEBUG 2025/01/18 21:09:35 power: 2311.5W
[circuit-main] DEBUG 2025/01/18 21:09:35 current: 0A
[... 61 lines cut ...]
[site ] DEBUG 2025/01/18 21:10:50 ----
[lp-1 ] DEBUG 2025/01/18 21:10:50 charge power: 2990W
[circuit-main] DEBUG 2025/01/18 21:10:50 power: 5379.8W
[circuit-main] DEBUG 2025/01/18 21:10:50 current: 0A
[site ] DEBUG 2025/01/18 21:10:50 pv 1 power: 6W
[site ] DEBUG 2025/01/18 21:10:50 grid power: 5380W
[site ] DEBUG 2025/01/18 21:10:50 site power: 5380W
[lp-1 ] DEBUG 2025/01/18 21:10:50 charger status: C
[lp-1 ] DEBUG 2025/01/18 21:10:50 soc estimated: 37.03% (vehicle: 37.00%)
[lp-1 ] DEBUG 2025/01/18 21:10:50 vehicle soc: 37%
[lp-1 ] DEBUG 2025/01/18 21:10:50 vehicle soc limit: 70%
[lp-1 ] DEBUG 2025/01/18 21:10:50 vehicle range: 161km
[site ] DEBUG 2025/01/18 21:11:05 ----
[lp-1 ] DEBUG 2025/01/18 21:11:05 charge power: 2990W
[circuit-main] DEBUG 2025/01/18 21:11:05 power: 5404.8W
[circuit-main] DEBUG 2025/01/18 21:11:05 current: 0A
[site ] DEBUG 2025/01/18 21:11:05 grid power: 5405W
[site ] DEBUG 2025/01/18 21:11:05 pv 1 power: 6W
[site ] DEBUG 2025/01/18 21:11:05 site power: 5405W
[lp-1 ] DEBUG 2025/01/18 21:11:05 charger status: C
[lp-1 ] DEBUG 2025/01/18 21:11:05 soc estimated: 37.04% (vehicle: 37.00%)
[lp-1 ] DEBUG 2025/01/18 21:11:05 vehicle soc: 37%
[lp-1 ] DEBUG 2025/01/18 21:11:05 vehicle soc limit: 70%
[lp-1 ] DEBUG 2025/01/18 21:11:05 vehicle range: 161km
[site ] DEBUG 2025/01/18 21:11:20 ----

andig, 2025-01-19:
> [lp-1 ] INFO 2025/01/18 21:08:20 car disconnected

@brainstormi das mypv hat nirgendwo einen "disconnected" Status. Es ist (für mich) nicht nachvollziehbar wie das hier zustande kommen soll.

andig, 2025-01-19:
@brainstormi für Lesbarkeit formatiert

andig, 2025-01-19:
@brainstormi 

```
[site ] DEBUG 2025/01/18 21:08:50 site power: 2322W
[lp-1 ] DEBUG 2025/01/18 21:08:50 charger status: B
[lp-1 ] INFO 2025/01/18 21:08:50 car connected
[lp-1 ] DEBUG 2025/01/18 21:08:50 vehicle soc: 37%
[lp-1 ] DEBUG 2025/01/18 21:08:50 charger enable
```

Kann ich nicht nachvollziehen es sei denn Du hättest irgendwelche merkwürdigen Thresholds oder minsoc oder... eingetragen. In jedem Fall ein lokaler Konfigurationsfehler. Im Zweifel yaml/db löschen und von vorne anfangen. Das hat mit Lastmangement nichts zu tun.

brainstormi, 2025-01-19:
> [@brainstormi](https://github.com/brainstormi)
>
> ```
> [site ] DEBUG 2025/01/18 21:08:50 site power: 2322W
> [lp-1 ] DEBUG 2025/01/18 21:08:50 charger status: B
> [lp-1 ] INFO 2025/01/18 21:08:50 car [...]

That it's the question... if it's config, what is wrong with it? I double checked it multiple times and can't find anything weird compared to similar setups discussed here. And what it's more weird is that it only happens when starting the charge randomly, once it settle down it works properly.
How EVCC calculates car charger power with the advised configuration of 2 meters. 1 for grid and 1 for pv? Is there any way to feed the system with car charging power explicitly in the configuration? or is automatically calculated based on the charging amps EVCC requests? because this last point seems to be the issue, not being able to different from grid load what it's the car charging power piece, miscalculating available capacity.

premultiply, 2025-01-19:
1. Looks like your configuration is missing a load counter? Estimating charging power can be very inaccurate.
2. The configured interval may be too low for your devices.

brainstormi, 2025-01-19:
> 1. Looks like your configuration is missing a load counter? Estimating charging power can be very inaccurate.
> 2. The configured interval may be too low for your devices.

1. Not sure how to implement a "load counter", I mean in the doc for custom chargers and vehicles I don't see a way to setup something similar. Only available attributes for them it's maxcurrent and seems to be used as a setter only... or maybe I'm missing something else. How can I feed evcc with current charger power instead to relay on it to estimate it?
2. Initially I had configured an interval of 30s but with similar issues, I can't recall/confirm if better or worse. I decreased to 15s seeing no warnings in the logs about issues with it and validating that changes done by load management are usually immediate to be "executed" in reality. Anyway I'll setup 30s back and check better how it performs with it.

brainstormi, 2025-01-20:
Regarding previous point 1... I just created a "fake" meter linked to the loadpoint, extracting the info from Teslamate. I'll update how it works

premultiply, 2025-01-20:
Ok, you are using Tesla cloud stuff with unknown behavior.
Load management should only be used with real local hardware and local connection.

andig, 2025-01-20:
Imho this discussion does not address the point. The question here is why charging starts although site power is positive.

brainstormi, 2025-01-20:
> Ok, you are using Tesla cloud stuff with unknown behavior. Load management should only be used with real local hardware and local connection.

I'm not using Tesla cloud stuff... only for creating a loadpoint meter with Teslamate data. For vehicle wake-up, setting charging-amps... I'm using Esphome-TeslaBLE running on an ESP32 board that communicates via Bluetotth directly with the car, so totally local and seems to work quite well, at least regarding timing to react to EVCC commands.

brainstormi, 2025-01-20:
> The question here is why charging starts although site power is positive: [#18263 (comment)](https://github.com/evcc-io/evcc/issues/18263#issuecomment-2600818222). If cloud or not, evcc has metering data. The decision does not fit that data. Question of configured thresholds is still open, so is

Not sure what you mean with charging shouldn't start with site power positive. Meanwhile there is grid load in the house, and always be, there is going to some value reporting this power consumption from the grid meter. Or do you mean this power consumption should be reported as negative value?. Could you explain better your point?
And about threshold and I don't have any configured.

brainstormi, 2025-02-04:
Just in case of anybody find this conversation and could be useful, after implementing a loadpoint meter (getting the data in my case from Teslamate) the issue didn't replicate anymore in any of the charges, working perfectly fine. So IMHO it seems that loadmanagement calculus based only in two meters, grid and pv, are buggy when differentiating charging power vs grid power when the charge process starts until it settle down.