evcc-io/evcc #18271: It seems the SMA values are wrong

supernova1963, 2025-01-18:
### Describe the bug

It seems the battery discharge power value is not correct (= power pv).

[image]

### Steps to reproduce

start evcc

```
root@evcc:/etc# service evcc restart
root@evcc:/etc# service evcc status 
* evcc.service - evcc
     Loaded: loaded (/usr/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Sat 2025-01-18 15:06:18 UTC; 11s ago
   Main PID: 772 (evcc)
      Tasks: 6 (limit: 18886)
     Memory: 35.3M (peak: 35.6M)
        CPU: 241ms
     CGroup: /system.slice/evcc.service
             `-772 /usr/bin/evcc

Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] INFO 2025/01/18 15:06:25 car connected
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 pv timer elapse
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 pv timer inactive
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 charge total import: 2378.928kWh
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 vehicle status: B
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 vehicle soc: 25%
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 vehicle range: 101km
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 pv charge current: 0A = 0A + -0.601A (415W @ 3p)
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 pv enable timer reset
Jan 18 15:06:25 evcc evcc[772]: [lp-1  ] DEBUG 2025/01/18 15:06:25 pv timer inactive
root@evcc:/etc# 
```

```
root@evcc:/etc# evcc -l debug meter
[main  ] INFO 2025/01/18 15:04:23 evcc 0.132.1
[main  ] INFO 2025/01/18 15:04:23 using config file: /etc/evcc.yaml
[db    ] INFO 2025/01/18 15:04:23 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/01/18 15:04:23 connecting evcc-640079682 at tcp://10.100.1.22:1883
[mqtt  ] DEBUG 2025/01/18 15:04:23 tcp://10.100.1.22:1883 connected
battery
-------
Power:          3745W
Energy:         18538.3kWh
Current L1..L3: 5.42A 5.44A 5.4A
Voltage L1..L3: 231V 232V 228V
Power L1..L3:   1253W 1260W 1233W
Soc:            100%

pv
--
Power:          3745W
Energy:         18538.3kWh
Current L1..L3: 5.42A 5.44A 5.4A
Voltage L1..L3: 231V 232V 228V
Power L1..L3:   1253W 1260W 1233W
Soc:            100%

grid
----
Power:          -3346W
Energy:         2786.6kWh
Current L1..L3: -4.57A -5.09A -5.08A
Voltage L1..L3: 229V 234V 232V
Power L1..L3:   -1002W -1176W -1169W

```

### Configuration details

```yaml
gernot@evcc:~$ cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 463c44df7778400b5eaaf1f490fdb34f140326a27ed3cf785923f6c6a14520cd

interval: 30s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 10.100.1.17
  name: grid
- type: template
  template: sma-inverter-speedwire
  usage: pv
  host: 10.100.1.16
  password: ***
  name: pv
- type: template
  template: sma-inverter-speedwire
  usage: battery
  host: 10.100.1.16
  password: ***
  name: battery

chargers:
- type: template
  template: smaevcharger
  host: 10.100.1.18
  user: installer
  password: ***
  name: wallbox

vehicles:
- type: template
  template: smart-hello
  user: gernot.rau@xxx.xx
  password: ***
  vin: XXXX
  title: 'Smart #1'  
  capacity: 66
  name: ev

loadpoints:
- title: Carport
  charger: wallbox
  vehicle: ev
  mode: pv

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery

# hems:
  # type: sma
  # AllowControl: false

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.34 # EUR/kWh
  # co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
    # type: grünstromindex # GrünStromIndex (Germany only)
    # zip: 51588
mqtt:
  broker: 10.100.1.22:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: xxx
  password: ***

# influx database
influx:
  url: http://10.100.1.20:8086
  database: evcc
  org: raunet
  token: XXX

messaging:
  events:
    start: # charge start event
      title: Laden gestartet
      msg: Ladevorgang im Modus "${mode}" gestartet
    stop: # charge stop event
      title: Laden beendet
      msg: Ladevorgang nach ${chargedEnergy:%.1fk}kWh in ${chargeDuration} beendet.
    connect: # vehicle connect event
      title: Auto verbunden
      msg: "Auto mit ${pvPower:%.1fk}kW PV verbunden"
    disconnect: # vehicle connected event
      title: Auto getrennt
      msg: Auto nach ${connectedDuration} getrennt
    soc: # vehicle soc update event
      title: Soc aktualisiert
      msg: Batterie wurde auf ${vehicleSoc:%.0f}% geladen
    guest: # vehicle could not be identified
      title: Unbekanntes Auto
      msg: Unbekanntes Auto erkannt, Gast verbunden?
```

### Log details

```text
[main ] INFO 2025/01/18 16:23:40 evcc 0.132.1
[main ] INFO 2025/01/18 16:23:40 using config file: /etc/evcc.yaml
[db ] INFO 2025/01/18 16:23:40 using sqlite database: /var/lib/evcc/evcc.db
[mqtt ] INFO 2025/01/18 16:23:40 connecting evcc-1154917350 at tcp://10.100.1.22:1883
[main ] INFO 2025/01/18 16:23:40 listening at :7070
[mqtt ] DEBUG 2025/01/18 16:23:40 tcp://10.100.1.22:1883 connected
[lp-1 ] DEBUG 2025/01/18 16:23:45 charge total import: 2378.928kWh
[site ] INFO 2025/01/18 16:23:46 site config:
[site ] INFO 2025/01/18 16:23:46 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/01/18 16:23:46 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2025/01/18 16:23:46 pv 1: power ✓ energy ✓ currents ✓
[site ] INFO 2025/01/18 16:23:46 battery 1: power ✓ energy ✓ currents ✓ soc ✓ capacity ✗
[site ] INFO 2025/01/18 16:23:46 vehicles:
[site ] INFO 2025/01/18 16:23:46 vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
[lp-1 ] INFO 2025/01/18 16:23:46 loadpoint 1:
[lp-1 ] INFO 2025/01/18 16:23:46 mode: pv
[lp-1 ] INFO 2025/01/18 16:23:46 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/01/18 16:23:46 meters: charge ✓
[lp-1 ] INFO 2025/01/18 16:23:46 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/01/18 16:23:46 set buffer soc: 100
[site ] DEBUG 2025/01/18 16:23:46 set priority soc: 60
[lp-1 ] DEBUG 2025/01/18 16:23:46 phase timer inactive
[lp-1 ] DEBUG 2025/01/18 16:23:46 pv timer inactive
[lp-1 ] INFO 2025/01/18 16:23:46 vehicle updated: unknown -> Smart #1
[site ] DEBUG 2025/01/18 16:23:46 ----
[lp-1 ] DEBUG 2025/01/18 16:23:46 charge power: 0W
[lp-1 ] DEBUG 2025/01/18 16:23:46 charge currents: [-0 -0 -0]A
[site ] DEBUG 2025/01/18 16:23:46 grid power: -2011W
[site ] DEBUG 2025/01/18 16:23:46 grid powers: [-583 -670 -758]W
[site ] DEBUG 2025/01/18 16:23:46 grid currents: [-2.88 -2.98 -3.34]A
[site ] DEBUG 2025/01/18 16:23:46 pv 1 power: 2468W
[site ] DEBUG 2025/01/18 16:23:46 battery 1 power: 2468W
[site ] DEBUG 2025/01/18 16:23:46 battery 1 soc: 100%
[site ] DEBUG 2025/01/18 16:23:46 site power: 457W
[lp-1 ] DEBUG 2025/01/18 16:23:46 vehicle odometer: 9425km
[lp-1 ] DEBUG 2025/01/18 16:23:46 charge total import: 2378.928kWh
[lp-1 ] DEBUG 2025/01/18 16:23:46 charger status: B
[lp-1 ] INFO 2025/01/18 16:23:46 car connected
[lp-1 ] DEBUG 2025/01/18 16:23:46 pv timer elapse
[lp-1 ] DEBUG 2025/01/18 16:23:46 pv timer inactive
[lp-1 ] DEBUG 2025/01/18 16:23:46 charge total import: 2378.928kWh
[lp-1 ] DEBUG 2025/01/18 16:23:46 vehicle status: B
[lp-1 ] DEBUG 2025/01/18 16:23:46 vehicle soc: 25%
[lp-1 ] DEBUG 2025/01/18 16:23:46 vehicle range: 101km
[lp-1 ] DEBUG 2025/01/18 16:23:46 pv charge current: 0A = 0A + -0.662A (457W @ 3p)
[lp-1 ] DEBUG 2025/01/18 16:23:46 pv enable timer reset
[lp-1 ] DEBUG 2025/01/18 16:23:46 pv timer inactive
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.132.1

VolkerK62, 2025-01-18:
I guess `sma-inverter-speedwire` ist NOT a hybrid device, so you cannot use it for both (pv and battery)

supernova1963, 2025-01-18:
It's a Sunny Tripower 10.0 Smart Energy with BYD HVS 12.8. 
I've used it nearly 1 year without problems.
What's the recommended configuration  now?

VolkerK62, 2025-01-18:
`sma-hybrid´ should be the correct template.
https://docs.evcc.io/docs/devices/meters#sma