evcc-io/evcc #18279: SUNGROW RH8.0RT / BYD HVS - Fehler bei Akku-Kommunikation /  beim Laden in Schnellmodus

Jakob-Gliwa, 2025-01-19:
### Describe the bug

Kriege "neuerdings" (hatte den Schnellmodus zugegebenermaßen schon ein paar Wochen nicht mehr an) folgende Fehlermeldungen im Log, wenn ich den Schnellmodus anschalte:
[site  ] ERROR 2025/01/19 11:22:14 grid power: read failed: write tcp 192.168.x.x:57092->192.168.x.x:502: write: broken pipe
[site  ] ERROR 2025/01/19 11:23:44 battery mode: write tcp 192.168.x.x:59344->192.168.x.x:502: write: broken pipe
(linker Wert die IP von evcc, rechter wert der meines Wechselrichters)

EVCC bricht den Schnellmodus ab und stellt sich auf "Aus"

Nutze folgendes Setup:
SUNGROW RH8.0RT (V112) / BYD HVS 5.1 / Elli Wallbox
Akku und WR-Neustart bereits ausprobiert
ETH ist über den LAN port hinten am gerät angebunden

EVCC läuft im Docker. Fehler tritt nach meinem Testen in folgenden Versionen auf:
0.132.1 (latest und nightly)
0.132.0
0.131.0
0.130.0

### Steps to reproduce

1. Auf "Schnell" schalten
2. Warten
3. EVCC setzt sich auf "Aus"

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  eebus: trace

# unique installation id
plant: x

interval: 10s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.b 
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.b 
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.b 
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
[... 390 lines cut ...]
[eebus ] DEBUG 2025/01/19 11:51:45 Send: d:_i:XXXXX_Elli-Wallbox-XXXXX notify 81 DeviceDiagnosisHeartbeatData
[site ] DEBUG 2025/01/19 11:51:46 ----
[lp-1 ] DEBUG 2025/01/19 11:51:46 charge power: 3680W
[site ] ERROR 2025/01/19 11:51:46 battery mode: write tcp <EVCCIP>:49358-><WRMODBUSIP>:502: write: broken pipe
[site ] DEBUG 2025/01/19 11:51:46 grid power: 6W
[site ] DEBUG 2025/01/19 11:51:46 pv 1 power: 1453W
[site ] DEBUG 2025/01/19 11:51:46 battery 1 power: 2465W
[site ] DEBUG 2025/01/19 11:51:46 grid currents: [10.2 -4.93 -5.17]A
[site ] DEBUG 2025/01/19 11:51:46 battery 1 soc: 9%
[site ] DEBUG 2025/01/19 11:51:46 site power: 2621W
[lp-1 ] DEBUG 2025/01/19 11:51:46 charger status: C
[eebus ] DEBUG 2025/01/19 11:51:47 Send: d:_i:XXXXX_Elli-Wallbox-XXXXX notify 82 DeviceDiagnosisHeartbeatData
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.132.1 (81b3d445). Auch gestestet: 0.132.1 (latest und nightly) 0.132.0 0.131.0 0.130.0

andig, 2025-01-19:
Bitte ohne Docker reproduzieren.

closed 2025-01-19 by andig

Jakob-Gliwa, 2025-01-19:
Hi, lag nicht an Docker.

Konnte es zwischenzeitlich lösen - es waren zwei Fehler:
**1. Fehler "Laden wird auf AUS gesetzt":** 
Eine Race-Condition aus einem anderen Skript, welches schlichtweg das Laden ausgeschaltet hat
=> In diesem Fall war das eine Steuerung über MQTT. Was mir geholfen hätte wäre, wenn min. Steuerbefehle von Außen auch in den Logs enhalten wären

**2. Fehler "[Broken:](write: broken pipe):**
Neben EVCC griff meine Heimautomatisierung auch auf die gleichen Modbusschnittstelle zu. Modbusproxy schuf hier abhilfe. Kam hierauf ehrlich gesagt nur durchs durchforsten von anderen Fällen hier.
=> Wäre es ggf. sinnvoll den Hinweis auf diese Konstellation in die Doku zu packen? Glaube für manche ist das selbstverständlich, dass Modbus solche Einschränkungen hat, aber für viele bestimmt nicht. Und dazu hat EVCC ja sogar noch eine eigene Lösung im Bauch.

andig, 2025-01-19:
> Was mir geholfen hätte wäre, wenn min. Steuerbefehle von Außen auch in den Logs enhalten wären

Was für Befehle in welchem Log? Woher soll evcc die kennen?

Jakob-Gliwa, 2025-01-19:
Ich meine Steuerbefehle, die über die offizielle EVCC MQTT API kommen (https://docs.evcc.io/docs/integrations/mqtt-api)

Entweder im Docker Log oder im Logviewer, welcher über die EVCC WebUI verfügbar ist.

Wenn im Log bspw. was gestanden wäre wie "[mqtt ] DEBUG 2025/01/19 11:51:26 received command (evcc/loadpoints/1/mode) : charge mode off" hätte mir das geholfen.

Sorry, wenn das wie Kritik rüberkam. War nur ein Gedanke. 
Finde eure Arbeit super und will nur das beste!

andig, 2025-01-19:
Auch inbound mqtt wird geloggt. Im loadpoint log steht dann

    set mode: off

das sollte also sichtbar sein.

Jakob-Gliwa, 2025-01-19:
Genau - das hier sehe ich, wenn ich das bei mir mit MQTT umschalte:

> [lp-1 ] DEBUG 2025/01/19 21:39:47 set charge mode: off

Mein Problem war, dass ich auch in der UI sehen konnte, dass der LP auf off geschaltet wurde.
Was ich nicht daraus sehen konnte ist, dass davor ein externer Steuerbefehl gekommen ist.
Die Nachricht so wie sie ist sagt mir nur, dass der Modus des LPs gesetzt wurde - aber ob das EVCC selbst aufgrund einer Logik intern war oder was anderes wurde mir daraus nicht direkt klar.

Da es scheinbar nicht so oft vorkommt, dass sich jemand quasi selbst sabotiert, ist das vll auch nur Schmuck am Nachthemd...