evcc-io/evcc #18287: Tesla Doku outdated

andig, 2025-01-19:
@naltatis ich hab beim Release übersehen, dass die Tesla Doku noch nicht passt: https://docs.evcc.io/docs/devices/vehicles#tesla

RikMe, 2025-01-19:
Will the current authentication method for Tesla continue to work in the latest release? Or is the 'new' way with the USD10 'credit' the only way as from 0.132.2 ?

andig, 2025-01-19:
See https://github.com/evcc-io/evcc/discussions/17501

RikMe, 2025-01-19:
Yup, saw that already. Should definitely move towards that in the future as the new way to go. But will the 'current' way continue to work with 0.132.2? I'm not the owner of te vehicle/chager. Can I safely update EVCC to 0.132.2 or do I wait till I'm physically there to make the switch ?

andig, 2025-01-19:
It will no longer work with 0.132.2.

closed 2025-01-19 by naltatis

RikMe, 2025-01-19:
ClientID is no longer optional now, yet it's hidden under the 'advanced options' setting.  Might be a good idea moving clientID to the 'non advanced' section?

naltatis, 2025-01-19:
@RikMe already changed in https://github.com/evcc-io/evcc/commit/2e9cebe254390eff5b2f8191ef59da48df828f22#diff-11b3f298c0c67fc01486d891465093538b9b474cc5829b893b077bbf76b41901L30

RikMe, 2025-01-19:
Here it's not being displayed. Of clientID really is a required parameter I would recommend including it in the standard options so it also gets copied when using the copy
Button.

[image]

naltatis, 2025-01-19:
It will be updated once we release 0.133. We've retracted 0.132.2 due to incomplete upgrade instructions.

mucki12, 2025-01-19:
0.132.2 wurde aber nur wegen der Doku zurückgezogen oder? Bin jetzt auf 0.132.2 und würde ungern downgraden.

naltatis, 2025-01-20:
v0.133 and the corresponding blog post are online now https://docs.evcc.io/en/blog/2025/01/20/tesla-api-update