evcc-io/evcc #18292: Nutzung aller 1-3 Temperatursensoren des AC Thor 9s zur Ermittlung des SOC

walburgf, 2025-01-19:
**Is your feature request related to a problem? Please describe.**
Der AC Thor 9s kann bis zu 3 Heizstäbe mit bis zu 3 Temperaturfühler im Speicher haben (oben, mitte, unten). Die derzeitige Implementierung unterstütut nur einen Temperatursensor und stoppt das Laden,  wenn dieser Temperatursensor das Limit erreicht hat.  jedoch die unteren Heizstäbe noch heizen könnten, weil das Limit der anderen Sensoren noch nicht errecht ist. Deshalb sollten alle Temperaturen im SOC von evcc berücksichtigt werden.

**Describe the solution you'd like**
Der SOC sollte der Mittelwert der Temperatursensoren sein.

**Describe alternatives you've considered**
Alternative wäre, wenn evcc alle bis zu 3 Temperatursensoren einzeln berücksichtigen würde. Das ist jedoch nicht notwendig, da die Berücksichtigung des Mittelwertes ausreichend ist.

**Additional context**
Ich  habe mal versucht die Lösung in Go zu schreiben (auch wenn ich kein ausgewiesener Go Spezialist bin. Man müsste die Datei mypv.go in etwa wie folgt ergänzen:

```
type MyPv struct {
	log     *util.Logger
	conn    *modbus.Connection
	power   uint32
	statusC uint16
	enabled bool
}

const (
	elwaRegSetPower  = 1000
	elwaRegTemp1     = 1001
	thorRegTemp2     = 1030
	thorRegTemp3     = 1031
	elwaRegTempLimit = 1002
	elwaRegStatus    = 1003
	elwaRegPower     = 1000 // https://github.com/evcc-io/evcc/issues/18020#issuecomment-2585300804
)

// CurrentPower implements the api.Meter interface
func (wb *MyPv) Soc() (float64, error) {

    temps := []uint16{}
    registers := []uint16{elwaRegTemp1, thorRegTemp2, thorRegTemp3}

    for _, reg := range registers {
        b, err := wb.conn.ReadHoldingRegisters(reg, 1)
        if err != nil {
            return 0, err
        }
        temp := binary.BigEndian.Uint16(b)
        if temp != 0 {
            temps = append(temps, temp)
        }
    }

    if len(temps) == 0 {
        return 0, fmt.Errorf("no valid temperature readings")
    }

    sum := 0
    for _, temp := range temps {
        sum += int(temp)
    }
    average := float64(sum) / float64(len(temps))

    return average / 10, nil
}

```

andig, 2025-01-19:
Es ist nicht vorgesehen, das umzusetzen. Die Abschaltung kann auch über die Steuerung des Thor erfolgen?

walburgf, 2025-01-19:
da die Steuerung durch evcc erfolgt, wird die Heizung abgeschaltet, sobald evcc keine Leistung mehr vorgibt. Da evcc den 1. Temperatursensor (das ist der Sensor im oberen Teil des Speichers) als Steuerungsgröße nimmt, schaltet der Sensor ab, sobald die maximale eingestellt Temperatur des Sensors erreicht ist (z.B. 60°C). Wenn der 2. Sensor aber noch bei 50°C ist, würde der AC Thor den 2. Heizstab noch heizen, wenn er von evcc eine Leistungsvorgabe hätte. Da evcc aber dann 0W vorgibt, heizt auch der AC Thor nicht mehr.
Insofern wird das ohne evcc Unterstützung nicht funktionieren...

Was spricht denn gegen die Implementierung? Es ist ja nur eine kleine Änderung? Und sie ist auch nicht "schädlich" für alle diejenigen, die "nur" einen Sensor haben.....

andig, 2025-01-19:
Ja- aber es zwingt dich ja niemand, 60° einzustellen. Warum nicht bei 100° lassen? Gerade bei den SG Ready Chargern macht das auch keinen Sinn da die Heizung per Hysterese ohnehin über den Wert drüber geht.

> Und sie ist auch nicht "schädlich" für alle diejenigen, die "nur" einen Sensor haben.....

Es ist m.E. physikalisch falsch und auch für mehrere Speicher- das Argument gabs schonmal ;).

andig, 2025-01-19:
> Wenn man die max. Temperatur auf z.B. 100°C stellt, dauert es viel länger bis der 2. und 3. Heizstab verwendet wird, da der 1. Heizstab ja damit beschäftigt ist, die 100°C zu erreichen.

Das mag sein hat aber mit evcc nichts zu tun- denn das steuert der Thor.

andig, 2025-01-19:
> und die max. Temperatur wird durch den Benutzer manuell eingestellt.....

...und auf die reagiert der Thor.

> Aber selbst wenn man die max Temp. auf 100°C einstellt, schaltet evcc die Beheizung beim Erreichen der Max. Temp die

...aber nur weil 100°C der Höchstwert der Skala ist. Dann würde ich vorschlagen den Sensor auswählbar zu machen.

walburgf, 2025-01-19:
den Sensor auswählbar machen wäre auch ok, dann könnte ich z.B. den 2. Sensor auswählen (in meinen Fall habe ich ja "nur" 2 Sensoren...

Oder: Wäre es auch möglich den SOC in der evcc.yaml zu definieren? Dann könnte man eine eigene Definition des SOC machen.
```

  soc:
    source: http
    uri: http://<ip des Thor>/data.jsn
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" # parse response json, get the median temperature
    scale: 0.1 # floating point factor applied to result, e.g. for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration

```
Wenn es keine Definition in evcc.yaml gibt, nimmst DU dann die Default Einstellung?

andig, 2025-01-19:
Kannst du als custom Charger machen.

walburgf, 2025-01-20:
ok, dann wäre es auf jeden Fall gut, wenn man den Temperatursensor auswählen könnte, der als SOC verwendet wird....

walburgf, 2025-01-20:
was muss ich da tun? hab ich noch nie gemacht?

docolli, 2025-01-20:
Also die aktuelle Lösung in evcc ist NUR für diese Betriebsart gedacht:
### AC•THOR 9s für 3-phasige Heizstäbe bis 9 kW
[image]

Dann gibt es jedoch auch noch diese Betriebsart:
### AC•THOR 9s auch für drei einphasige Heizstäbe
Das Bild hast du ja oben bereits gezeigt.

Zum Schluss kann der AC Thor aber auch in dieser Betriebsart genutzt werden:
### AC•THOR 9s für Warmwasser und Raumheizung
[image]

Wie @andig bereits ausgeführt hat, kann man die anderen Betriebsarten über einen Custom Charger realisieren. Oder besser gesagt mehrere Custom Charger / Loadpoints, die auf die jeweiligen Register von Leistung / Temperatur vom Heizkreis 1/2 bzw. 3 verweisen.

Siehe https://github.com/evcc-io/evcc/discussions/13398

Eventuell magst du ja für jeden Heizstab einen anderen Sollwert vorgeben, bis zu dem geheizt werden soll. Dann machen getrennte Loadpoints Sinn.

Du kannst aber auch einen Pull Request (PR) einreichen, also am evcc Project gerne aktiv mitarbeiten. Das heißt, mache Dir in Github einen Fork von evcc, programmiere dort die notwendigen Änderungen in Go, beachte aber dass diese Änderungen auch für die anderen Betriebsarten (zumindest der aktuell unterstützen) funktionieren. Kompiliere evcc selbst auf deinem System, teste es und wenn du der Meinung bist, der Code ist reif, dann reiche deine Änderungen als PR ein. Wenn keine Probleme bei automatischen Tests auftreten und auch ansonsten nichts gegen den Code spricht, wird er vermutlich in evcc übernommen.

andig, 2025-01-26:
Gerne PR für die Auswahl des Sensors.

closed 2025-01-26 by andig