evcc-io/evcc #18293: update from 0.132.1 --> 0.132.2 on HomeAssistant // Error

michaelkrause-SAP, 2025-01-19:
### Describe the bug

Error Code: Die Aktion update/install konnte nicht ausgeführt werden. Error updating evcc: Can't install evcc/evcc:0.132.2: 404 Client Error for http+docker://localhost/v1.47/images/create?tag=0.132.2&fromImage=evcc%2Fevcc&platform=linux%2Farm64: Not Found ("manifest for evcc/evcc:0.132.2 not found: manifest unknown: manifest unknown")

### Steps to reproduce

1. klick on "aktualisieren", then Error shows up.

[image]

2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# mqtt message broker
mqtt:
  broker: 192.168.178.246:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: evcc
  password: XxX123/hugo%99XxX

log: debug
levels:
  cache: error

# unique installation id
plant: f6d1b6aac4f57ec494b413cd2a8656ed1351c3d453f4fa903991d51f92881797

interval: 10s # control cycle interval

sponsortoken: XXXYYYZZZ

# changed to # due to configuration in GUI
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

#circuits: ##noch nicht aktiv MK ##
#- name: main
#  title: Hauptsicherung
#  maxCurrent: 63
#  name: garage
#  title: Garage
#  maxCurrent: 63
#  parent: main

meters:
#- name: pv20
#  type: template
#  template: sma-inverter-modbus
#  usage: pv
#  modbus: tcpip
#  id: 3
#  host: 192.168.178.148
#  port: 502

#- name: pv10
#  type: template
#  template: sma-hybrid
#  usage: pv
#  modbus: tcpip
#  id: 3
#  host: 192.168.178.223
#  port: 502

#- name: pv5
#  type: template
#  template: sma-inverter-modbus
#  usage: pv
#  modbus: tcpip
#  id: 3
#  host: 192.168.178.170
#  port: 502

- name: grid1
  type: template
  template: sma-energy-meter
  usage: grid
  host: 192.168.178.43
  #interface: eth0

#- name: my_battery
#  type: template
#  template: sma-hybrid
#  usage: battery
#  capacity: 13.12 #Batteriekapa
#  # Modbus TCP
#  modbus: tcpip
#  id: 3
#  host: 192.168.178.223 # Hostname
#  port: 502 # Port

chargers:
- type: template
  template: smaevcharger
  host: 192.168.178.28
  user: krause-15344
  password: HR07!ercon
  name: wallbox1

# changed to # due to configuration in GUI
- type: template
  template: smaevcharger
  host: 192.168.178.243
  user: Krause-15344
  password: HR07!ercon
  name: wallbox2

site:
  title: MK Home
  residualPower: 100
  meters:
#     battery: my_battery
     grid: grid1
     pv:
#       - pv20
#       - pv10
#       - pv5

loadpoints:
  - title: Wallbox links
    charger: wallbox2
##    vehicle: EV1
    mode: pv
  - title: Wallbox rechts # display name for UI
    charger: wallbox1 # Wallbox Referenz
##    vehicle: EV1 # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.35

  feedin:
    type: fixed
    price: 0.08
```

### Log details

```text
Die Aktion update/install konnte nicht ausgeführt werden. Error updating evcc: Can't install evcc/evcc:0.132.2: 404 Client Error for http+docker://localhost/v1.47/images/create?tag=0.132.2&fromImage=evcc%2Fevcc&platform=linux%2Farm64: Not Found ("manifest for evcc/evcc:0.132.2 not found: manifest unknown: manifest unknown")
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

thecem, 2025-01-19:
0.132.2 wurde zurückgezogen.

michaelkrause-SAP, 2025-01-19:
danke, dann überspringe ich das Update !
merci! für die schnelle antwort

thecem, 2025-01-19:
Mach dann bitte hier zu.

closed 2025-01-19 by andig