evcc-io/evcc #18297: Stiebel Eltron - Schätzwert der Leistungsaufnahme via Modbus

Tombra1889, 2025-01-19:
> Wichtiger wäre die Leistung- frag doch mal bei Stiebel an. Ansonsten bleibt ja immer ein Zwischenzähler, den kannst Du dann am Ladepunkt zuordnen.

@andig Ich habe in einem anderem Forum die folgende Modbus-Adresse gefunden, welche wohl die Leistungsaufnahme abbilden soll. Hab es mal geprüft: Modbus 33680

[image]

Bei mir weicht der Wert um 200 Watt vom KNX Zähler ab. KNX: 1,1 KW / Modbus 0.9 KW

_Originally posted by @Tombra1889 in https://github.com/evcc-io/evcc/issues/17735#issuecomment-2588342202_

andig, 2025-01-19:
Nice- machst du nen PR? Musst nur hier das Template editieren…

Tombra1889, 2025-01-19:
So ganz firm bin ich leider noch nicht in der Materie.

premultiply, 2025-01-19:
Helfen uns diese Schätzwerte an der Stelle denn überhaupt?
Ein "richtiger" Zähler könnte auch obligatorisch sein?

Tombra1889, 2025-01-19:
Sofern ein richtiger Zähler davor konfiguriert ist, wäre es natürlich gut, wenn die Schätzwerte dann ignoriert werden. Für andere genügt ggf. eine ungefähre Anzeige. Immerhin besser als 0.

wittmannchris, 2025-01-25:
Der Register stimmt schon und die Werte sind auch die der Wärmepumpe.
Allerdings läuft bei Stiebel die Steuerung und auch das Lüfterrad über einen extra Anschluss und werden daher nicht in dem Register geschrieben.
Hab an den beiden kleine Shellys dran und messe an der Steuerungca. 74W(mit Lüftung) und 140W fürden Lüfter. Passt alo dann genau zusammen.

andig, 2025-01-26:
Gerne PR- da muss nur das Register im Template ergänzt werden. Da die Steuerung jedoch as-is nicht funktioniert ist die Stiebel bis auf Weiteres als "deprecated" markiert.

closed 2025-01-26 by andig