evcc-io/evcc #18298: Duplicate TerraAC

andig, 2025-01-19:
We have the device twice with different names. Once modbus, once ocpp. How do we want to present this? Should we really have ocpp given that modbus is easier to configure?

naltatis, 2025-01-21:
> Once modbus, once ocpp. How do we want to present this?

See previous discussion here: https://github.com/evcc-io/evcc/pull/15293
And a proposal for docs here: https://github.com/evcc-io/docs/pull/597

naltatis, 2025-01-21:
> Should we really have ocpp given that modbus is easier to configure?

There might be different reasons why the user would prefer one over the other (e.g. physical connection, available features). If we can support multiple ways we should do that make this and adjust the docs accordingly.

naltatis, 2025-01-21:
Quickfix here would be to rename the second down to "Terra AC (OCPP)".

andig, 2025-01-21:
Ocpp raus? Ist für den Anwender hier nur komplizierter ohne Vorteile?

premultiply, 2025-01-22:
Hat schon beides seine Berechtigung je nach FW und Anbindung.

andig, 2025-01-23:
Closed in e761b3d55346ca84f0b81118b925fe94f7c94c3a

closed 2025-01-23 by andig