evcc-io/evcc #18309: Fehlermeldung nach Umstellung Tesla API

Dragonkit, 2025-01-20:
### Describe the bug

Hi,
ich bekomme immer folgende beim einbinden meine Autos im EVCC.

[main  ] ERROR 2025/01/20 12:52:20 creating vehicle Tesla_Model_Y failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 403 Forbidden
[main  ] ERROR 2025/01/20 12:52:20 creating vehicle Tesla_Model_3 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 403 Forbidden
[site  ] WARN 2025/01/20 12:52:21 vehicle 'Tesla Model 3' does not support automatic detection
[site  ] WARN 2025/01/20 12:52:21 vehicle 'Tesla Model Y' does not support automatic detection

Mit der alten Konfig geht es ohne Probleme. Hat da jemand eine Lösung ? 
Ich habe alles wie unter https://docs.evcc.io/blog/2025/01/20/tesla-api-update beschrieben eingerichtet. 
Schlüssel habe ich in den Autos auch eingerichtet. Sollte also normal funktionieren.

Was mir auch aufgefallen ist das ich beim Neustarten von EVCC immer diesen Fehler habe. Bekommt man den weg ?
[site ] ERROR 2025/01/20 12:53:08 pv 1 power: API call to https://192.168.178.9/api/login/Basic returned unexpected status code 429 ("{\"code\":429,\"error\":\"Api Limit reached for this endpoint\",\"message\":\"API Limit Reached\"}")

### Steps to reproduce

Ich habe es mehrfach versucht das hinzubekommen. Wenn man nur 1 Auto nimmt scheint es ab und an zu funktionieren. Warum auch immer.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: xxx 
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
[... 313 lines cut ...]
[site  ] DEBUG 2025/01/20 12:52:51 ----
[lp-1  ] DEBUG 2025/01/20 12:52:51 charge power: 0W
[lp-1  ] DEBUG 2025/01/20 12:52:51 charge currents: [0.4 0.1 0.5]A
[lp-2  ] DEBUG 2025/01/20 12:52:51 charge power: 0W
[lp-2  ] DEBUG 2025/01/20 12:52:51 charge currents: [0.2 0 0.2]A
[site  ] DEBUG 2025/01/20 12:52:52 battery 1 power: 10W
[site  ] DEBUG 2025/01/20 12:52:52 pv 1 power: 1963W
[site  ] DEBUG 2025/01/20 12:52:52 battery 1 soc: 100%
[site  ] DEBUG 2025/01/20 12:52:52 grid power: -1428W
[site  ] DEBUG 2025/01/20 12:52:52 site power: -1318W
[lp-2  ] DEBUG 2025/01/20 12:52:52 charge voltages: [2 0 4.6]V
[lp-2  ] DEBUG 2025/01/20 12:52:52 charger status: A
[lp-2  ] INFO 2025/01/20 12:52:52 car disconnected
[lp-2  ] DEBUG 2025/01/20 12:52:52 set battery boost: false
[lp-2  ] DEBUG 2025/01/20 12:52:52 set session soc limit: 0
[lp-2  ] DEBUG 2025/01/20 12:52:52 set session energy limit: 0
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-20:
> API Limit Reached

Die Meldung spricht für sich selbst. Bzgl. Tesla fehlt ein trace log.

closed 2025-01-20 by andig

Dragonkit, 2025-01-20:
Reicht das Trace so aus ?

[tesla ] TRACE 2025/01/20 14:10:52 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2025/01/20 14:10:52 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2025/01/20 14:10:53 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2025/01/20 14:10:53 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2025/01/20 14:10:53 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2025/01/20 14:10:53 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2025/01/20 14:10:53 {"response":null,"error":"Unauthorized missing scopes","error_description":"","txid":"xxx"}
[tesla ] TRACE 2025/01/20 14:10:53 {"response":null,"error":"Unauthorized missing scopes","error_description":"","txid":"xxx"}

andig, 2025-01-20:
Fehlermeldung lesen und verstehen:

> Unauthorized missing scopes

Das ist eindeutig.

teslajoshi, 2025-01-21:
Hallo, ich habe auch die Fehlermeldung Unauthorized missing scopes. Für was ist das eindeutig? Ich habe mich auch an die Anleitung gehalten. Danke für einen Hinweis.

andig, 2025-01-21:
Es fehlen Scopes- da hast du nixht alle notwendigen ausgewählt.

Dragonkit, 2025-01-21:
Ich hab es heute nicht getestet. Aber es sieht so aus also ob es einen Tag dauert bis es sauber synchronisiert ist. Sieht zumindest heute anders aus und ich hab mehr zur Auswahl bei den Funktionen. Ich teste es morgen nochmal.

teslajoshi, 2025-01-22:
So, heute funktioniert's nun. Danke an alle für die hilfreichen Informationen.

Dragonkit, 2025-01-22:
Bei mir gehts heute auch mit den Autos. Muss ich bei der Powerwall auch was umstellen oder kann das so bleiben wie es war ? 
Finde da nichts neues in der Doku. Würde es also erstmal so lassen wie es ist und funktioniert.