evcc-io/evcc #18311: OpenWb mit Software 2.0 ohne Ühasenumschaltung nicht möglich

neupertd, 2025-01-20:
Für die OpenWB mit Software 2.0 ist kein Parameter für die optionale Phasenumschaltung vorhanden.

Falls ich die Konfig so anpasse:

- type: template
  template: openwb-2.0
  id: 1
  host: 192.168.1.72
  port: 1502
  modbus: tcpip
  connector: 1
  name: openwb
  phases1p3p: false

Mit der Option  phases1p3p: false  aus dem Template für die alte Software erhalte ich als Fehlermeldung:

[main  ] FATAL 2025/01/20 13:51:49 cannot create charger 'openwb': cannot create charger type 'template': invalid key: phases1p3p

Es wäre gut, wenn die Phasenumschaltung wieder abschaltbar wäre.
So erhalte ich bei PV geführtem Laden einen Fehler, dass er eine Phase erwartet, aber drei benutzt werden.

andig, 2025-01-20:
Da waren wohl wir/jemand der Meinung, dass eine sw2.0 immer nur für Geräte mit Umschaltung verfügbar ist. Gibts irgendwo eine Übersicht der Gerätematrix?

andig, 2025-01-20:
@snaptec danke- das bedeutet dann also, dass wir für sw20 auch eine Konfiguration (oder besser: Erkennung) der Phasenumschaltung brauchen. Können wir die über Modbus erkennen? Das wäre schöner als das den Anwender konfigurieren zu lassen...

neupertd, 2025-01-20:
Es gibt unter den [OpenWb-Downloads](https://openwb.de/main/?page_id=135) eine Produktmatrix als auch die Registerbeschreibung.
Meine 2.0 Custom hat zwar eine Addon-Platine. Die kann aber keine Phasenumschaltung.

In der Übersicht der Modbus-Register ist aber kein Vermerk zur Erkennung der Phasenumschaltung.
Eine Möglichkeit wäre sonst Testen ob er umschaltet und falls nicht gibt es die Umschaltung nicht.
Den Konfig-Eintrag fände ich aber auch nicht schlecht.

Im MQTT gibt es ein Topic openWB/chargepoint/3/config. Im JSON ist ein Wert auto_phase_switch_hw enthalten, der bei mir korrekter Weise auf false steht.

andig, 2025-01-23:
> Können wir die über Modbus erkennen

ping @snaptec

closed 2025-01-23 by andig