evcc-io/evcc #18313: House battery has discharged despite contrary setting during fast charging

biodiv, 2025-01-20:
### Describe the bug

senec-home battery discharged although the battery was "locked". The battery has been discharged completely during the vehicle fast charge.

[image]

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: ####

interval: 30s

telemetry: true

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.369
  feedin:
    type: fixed
    price: 0.08

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

modbusproxy:

meters:
  - name: my_grid
    type: template
    template: senec-home
    usage: grid
    host: ######
    schema: https # optional 
  - name: my_pv
    type: template
    template: senec-home
    usage: pv
    host: #####
    schema: https # optional
  - name: my_battery
    type: template
    template: senec-home
    usage: battery
    host: #####
    schema: https # optional

chargers:
  - name: warp3
    type: template
    template: tinkerforge-warp3
    host: #####
    port: #####
    timeout: 30s
    topic: warp3

vehicles:
  - name: Polestar2
    type: template
    template: polestar
    title: Polestar
    user: ####
    password: ####
    capacity: 77
    icon: car
    phases: 3
    cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)
    mode: PV
    minCurrent: 6
    maxCurrent: 16
    
# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: my_grid # grid meter
    pv:
      - my_pv # list of pv inverters/ meters
    battery:
      - my_battery # list of battery meters
  residualPower: 0 # additional household usage margin
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    charger: warp3 # charger
    vehicle: Polestar2 # default vehicle
```

### Log details

```text
2025/01/20 16:11:28,stdout,[site  ] DEBUG 2025/01/20 16:11:28 ----

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 vehicle range: 150km

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 vehicle soc: 38%

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 soc estimated: 37.75% (vehicle: 36.00%)

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 charger status: C

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 charge total import: 1553.780kWh

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 detected active phases: 3p

2025/01/20 16:10:58,stdout,[site  ] DEBUG 2025/01/20 16:10:58 site power: 11260W

2025/01/20 16:10:58,stdout,[site  ] DEBUG 2025/01/20 16:10:58 battery 1 soc: 19%

2025/01/20 16:10:58,stdout,[site  ] DEBUG 2025/01/20 16:10:58 pv 1 power: 585W

2025/01/20 16:10:58,stdout,[site  ] DEBUG 2025/01/20 16:10:58 grid power: 7757W

2025/01/20 16:10:58,stdout,[site  ] DEBUG 2025/01/20 16:10:58 battery 1 power: 3403W

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 charge currents: [15.1 15.2 15.3]A

2025/01/20 16:10:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:58 charge power: 10776W

2025/01/20 16:10:58,stdout,[site  ] DEBUG 2025/01/20 16:10:58 ----

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 vehicle range: 150km

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 vehicle soc: 38%

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 soc estimated: 37.65% (vehicle: 36.00%)

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 charger status: C

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 charge total import: 1553.690kWh

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 detected active phases: 3p

2025/01/20 16:10:28,stdout,[site  ] DEBUG 2025/01/20 16:10:28 site power: 11121W

2025/01/20 16:10:28,stdout,[site  ] DEBUG 2025/01/20 16:10:28 battery 1 soc: 20%

2025/01/20 16:10:28,stdout,[site  ] DEBUG 2025/01/20 16:10:28 pv 1 power: 615W

2025/01/20 16:10:28,stdout,[site  ] DEBUG 2025/01/20 16:10:28 battery 1 power: 3403W

2025/01/20 16:10:28,stdout,[site  ] DEBUG 2025/01/20 16:10:28 grid power: 7718W

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 charge currents: [15.1 15.2 15.3]A

2025/01/20 16:10:28,stdout,[lp-1  ] DEBUG 2025/01/20 16:10:28 charge power: 10774W

2025/01/20 16:10:28,stdout,[site  ] DEBUG 2025/01/20 16:10:28 ----

2025/01/20 16:10:25,stdout,[site  ] DEBUG 2025/01/20 16:10:25 set battery discharge control: true

2025/01/20 16:10:24,stdout,[site  ] DEBUG 2025/01/20 16:10:24 set battery discharge control: false

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 vehicle range: 150km

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 vehicle soc: 38%

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 soc estimated: 37.54% (vehicle: 36.00%)

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 charger status: C

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 charge total import: 1553.600kWh

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 detected active phases: 3p

2025/01/20 16:09:58,stdout,[site  ] DEBUG 2025/01/20 16:09:58 site power: 11088W

2025/01/20 16:09:58,stdout,[site  ] DEBUG 2025/01/20 16:09:58 battery 1 soc: 20%

2025/01/20 16:09:58,stdout,[site  ] DEBUG 2025/01/20 16:09:58 pv 1 power: 645W

2025/01/20 16:09:58,stdout,[site  ] DEBUG 2025/01/20 16:09:58 grid power: 7681W

2025/01/20 16:09:58,stdout,[site  ] DEBUG 2025/01/20 16:09:58 battery 1 power: 3407W

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 charge currents: [15.1 15.2 15.3]A

2025/01/20 16:09:58,stdout,[lp-1  ] DEBUG 2025/01/20 16:09:58 charge power: 10772W

2025/01/20 16:09:58,stdout,[site  ] DEBUG 2025/01/20 16:09:58 ----
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-01-20:
This case is not implemented:

    - case: 2 # hold (not implemented)
      set:
        source: error
        error: ErrNotAvailable

Error should be displayed in log/ui.

biodiv, 2025-01-20:
What is the reason for this case not being implemented?

andig, 2025-01-20:
Weil niemand weiss, "wie"...