evcc-io/evcc #18315: Broken carwings API - 404 (INVALID PARAMS)

mschaeuble, 2025-01-20:
### Describe the bug

The SOC value of my Nissan Leaf cannot be retrieved anymore. An error is logged. It looks like the API URL has changed. 

### Steps to reproduce

1. Configure EVCC with a carwings template based vehicle
2. Start EVCC.
3. Observe the error message in the log.

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: carwings
    title: Nissan Leaf
    user: xxx
    password: xxx
    capacity: 39
```

### Log details

```text
[main  ] ERROR 2025/01/20 17:57:44 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'carwings': login failed: received status code 404 (INVALID PARAMS)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-01-20:
Please report at https://github.com/joeshaw/carwings/issues. Nothing we could do right now.

mschaeuble, 2025-01-21:
Reported a bug in the carwings library: https://github.com/joeshaw/carwings/issues/53

MoJo2600, 2025-01-21:
This is an error on Nissan side currently. Login in the app is also not possible at the moment.

mschaeuble, 2025-01-21:
I don't agree. In my case, the official "Nissan EV" app, "My Leaf" as well as homeassistant-carwings is able to access the API.

MoJo2600, 2025-01-21:
> I don't agree. In my case, the official "Nissan EV" app, "My Leaf" as well as homeassistant-carwings is able to access the API.

Okay, that is strange. 

There are two issues. This one is new and I confirm this. The carwings lib, the homeassistant-carwings integration and the app are not working for me and a couple of people here: https://github.com/remuslazar/homeassistant-carwings/issues/84. This is confirmed to be an issue on Nissan side. (Tested it right now again in NE region)

But, there was another issue a couple of days ago https://github.com/remuslazar/homeassistant-carwings/issues/79 which was fixed in version [0.7.12](https://github.com/ev-freaks/pycarwings3/commit/45dd9d0bb3b92ed34c5530023bfb068a187122cc) of the pycarwings3 library.

The 404 error sounds more like the issue that was fixed in 0.7.12 of the lib. Which should be resolved with an update of the lib @andig - but login will still not work until Nissan gets the login api back online.

andig, 2025-01-21:
@MoJo2600 where's the fixing PR for pycarwings3? I can't even find pycarwings3 :/

andig, 2025-01-21:
Siehe https://github.com/joeshaw/carwings/issues/53#issuecomment-2603927208. Der pcw3 Fix ist für uns nicht relevant.

andig, 2025-01-24:
Closed in https://github.com/evcc-io/evcc/commit/ad9b6ad3afc384d70cf26187c58f010643d57f99

closed 2025-01-24 by andig

mschaeuble, 2025-02-03:
@andig When are you planning to create a new evcc release? The missing SOC in the latest EVCC release is quite annoying for Nissan owners.

andig, 2025-02-03:
It‘s fixed in the nightly version as temporary workaround.

mpschaeuble, 2025-02-03:
@andig I couldn't find a way to update EVCC to a nightly build in Home Assistant. How do I do that?

mschaeuble, 2025-02-03:
@andig I couldn't find a way to update EVCC to a nightly build in Home Assistant. How do I do that?

andig, 2025-02-03:
As we've repeated numerous times: you don't. That's one the downsides of using evcc in HA. Unfortnately our release process currently doesn't allow releasing at any specific time for every issue.

MoJo2600, 2025-02-03:
@mschaeuble  I'm using the HA nissan carwings integration and push the data via an automation to MQTT. I added a custom car integration in evcc to load the data from MQTT. This way I'm not dependent on the evcc implementation anymore. But it adds some complexity.