evcc-io/evcc #18318: Falsche Werte auf der GUI bei Ladevorgang

Maxheino, 2025-01-20:
### Describe the bug

Hallo,

seit einigen Versionen zeigt die EVCC-Oberfläche beim Ladevorgang komische Werte (rot markiert) an.

[image]

### Steps to reproduce

1. Fehler tritt nur beim Ladevorgang auf.

### Configuration details

```yaml
meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.72  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.72  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.72  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery1

chargers:
- type: template
  template: versicharge 
  id: 2  
  host: 192.168.178.74
  port: 502
  modbus: tcpip  
  name: wallbox

vehicles:
- title: e-up 
  type: template
  template: vw  
  user: GEHEIM
  password: GEHEIM
  vin: GEHEIM
  capacity: 35
#  targetSoc: 50
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
#  resetOnDisconnect: true

[... 1002 lines cut ...]
[site ] DEBUG 2025/01/20 20:50:48 pv 1 power: 0W
[site ] DEBUG 2025/01/20 20:50:48 battery 1 power: 0W
[site ] DEBUG 2025/01/20 20:50:48 battery 1 soc: 3%
[site ] DEBUG 2025/01/20 20:50:48 grid power: 3667W
[site ] DEBUG 2025/01/20 20:50:48 grid currents: [6.91 7.77 1.34]A
[site ] DEBUG 2025/01/20 20:50:48 site power: 3667W
[lp-1 ] DEBUG 2025/01/20 20:50:48 charge voltages: [227 228 228]V
[lp-1 ] DEBUG 2025/01/20 20:50:48 detected connected phases: 3p
[lp-1 ] DEBUG 2025/01/20 20:50:48 detected active phases: 2p
[lp-1 ] DEBUG 2025/01/20 20:50:49 charge total import: 1673.636kWh
[lp-1 ] DEBUG 2025/01/20 20:50:49 charger status: C
[lp-1 ] DEBUG 2025/01/20 20:50:49 pv charge current: min 6A > 0A (3667W @ 2p, battery: false)
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-01-20:
> charge power: 68117W

Das ist offensichtlich falsch. Bitte beim laden mal

    evcc charger —log trace

Maxheino, 2025-01-20:
[3 Elemente.](https://192.168.178.10:5001/#)
[versicharge] TRACE 2025/01/20 22:03:49 modbus: send 00 02 00 00 00 06 02 03 06 9
c 00 02                                                                          
[versicharge] TRACE 2025/01/20 22:03:49 modbus: recv 00 02 00 00 00 07 02 03 04 0
0 19 90 10                                                                       
[versicharge] TRACE 2025/01/20 22:03:49 modbus: send 00 03 00 00 00 06 02 03 06 6
f 00 03                                                                          
[versicharge] TRACE 2025/01/20 22:03:49 modbus: recv 00 03 00 00 00 09 02 03 06 0
0 06 00 05 00 00                                                                 
[versicharge] TRACE 2025/01/20 22:03:49 modbus: send 00 04 00 00 00 06 02 03 06 7
3 00 03                                                                          
[versicharge] TRACE 2025/01/20 22:03:49 modbus: recv 00 04 00 00 00 09 02 03 06 0
0 e5 00 e5 00 e7                                                                 
[versicharge] TRACE 2025/01/20 22:03:49 modbus: send 00 05 00 00 00 06 02 03 06 3
f 00 01                                                                          
[versicharge] TRACE 2025/01/20 22:03:49 modbus: recv 00 05 00 00 00 05 02 03 02 4
3 32                                                                             
[versicharge] TRACE 2025/01/20 22:03:49 modbus: send 00 06 00 00 00 06 02 03 06 6
1 00 01                                                                          
[versicharge] TRACE 2025/01/20 22:03:49 modbus: recv 00 06 00 00 00 05 02 03 02 0
0 06                                                                             
Power:          68077W                                                           
Energy:         1675.3kWh                                                        
Current L1..L3: 6A 5A 0A                                                         
Voltage L1..L3: 229V 229V 231V                                                   
Charge status:  C                                                                
Enabled:        true

Maxheino, 2025-01-20:
Ein Wallbox Neustart hat keine Änderung bewirkt.

andig, 2025-01-21:
Kannst Du das Log bitte so formatierne, dass man drin suchen kann? Leistung ist egtl. Register `06 7E`, das sehe ich aber nirgendwo.

closed 2025-01-23 by andig

Maxheino, 2025-01-23:
[site  ] DEBUG 2025/01/23 17:51:17 battery 1 power: 0W                           
[battery1] TRACE 2025/01/23 17:51:17 modbus: send 00 86 00 00 00 06 01 04 32 e2 00 02                                                                             
[battery1] TRACE 2025/01/23 17:51:17 modbus: recv 00 86 00 00 00 07 01 04 04 ac 56 00 00                                                                          
[battery1] TRACE 2025/01/23 17:51:17 modbus: send 00 87 00 00 00 06 01 04 32 de 00 01                                                                             
[battery1] TRACE 2025/01/23 17:51:17 modbus: recv 00 87 00 00 00 05 01 04 02 00 00                                                                                
[site  ] DEBUG 2025/01/23 17:51:17 battery 1 soc: 0%                             
[cache ] TRACE 2025/01/23 17:51:17 batteryCapacity: 1                            
[cache ] TRACE 2025/01/23 17:51:17 batterySoc: 0                                 
[cache ] TRACE 2025/01/23 17:51:17 batteryPower: 0                               
[cache ] TRACE 2025/01/23 17:51:17 batteryEnergy: 4411.8                         
[cache ] TRACE 2025/01/23 17:51:17 battery: [{0 4411.8 [] [] 0 0xc000ddc658 0xc000ddc650 0xc000ddc660}]                                                           
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 88 00 00 00 06 01 04 32 d1 00 02                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 88 00 00 00 07 01 04 04 f0 60 ff ff                                                                            
[site  ] DEBUG 2025/01/23 17:51:18 grid power: 4000W                             
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 89 00 00 00 06 01 04 15 e2 00 02                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 89 00 00 00 07 01 04 04 06 58 00 00                                                                             
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 8a 00 00 00 06 01 04 13 9a 00 01                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 8a 00 00 00 05 01 04 02 09 0d 
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 8b 00 00 00 06 01 04 15 e4 00 02                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 8b 00 00 00 07 01 04 04 07 ef 00 00                                                                            
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 8c 00 00 00 06 01 04 13 9b 00 01                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 8c 00 00 00 05 01 04 02 09 02 
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 8d 00 00 00 06 01 04 15 e6 00 02                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 8d 00 00 00 07 01 04 04 01 4a 00 00                                                                            
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 8e 00 00 00 06 01 04 13 9c 00 01                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 8e 00 00 00 05 01 04 02 09 1b 
[site  ] DEBUG 2025/01/23 17:51:18 grid currents: [7.01 8.81 1.42]A              
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: send 00 8f 00 00 00 06 01 04 32 ec 00 02                                                                               
[grid1 ] TRACE 2025/01/23 17:51:18 modbus: recv 00 8f 00 00 00 07 01 04 04 cf 7a 00 00                                                                         
[site  ] DEBUG 2025/01/23 17:51:18 site power: 4000W                             
[versicharge] TRACE 2025/01/23 17:51:18 modbus: send 00 40 00 00 00 06 02 03 06 73 00 03                                                                          
[cache ] TRACE 2025/01/23 17:51:18 grid: {4000 5311.400000000001 [] [7.009063444108761 8.80745880312229 1.4157014157014156] 0 <nil> <nil> <nil>}                  
[cache ] TRACE 2025/01/23 17:51:18 homePower: 0                                  
[cache ] TRACE 2025/01/23 17:51:18 lp-1/smartCostActive: false                   
[cache ] TRACE 2025/01/23 17:51:18 lp-1/smartCostNextStart: 0001-01-01 00:00:00 +0000 UTC                                                                         
[versicharge] TRACE 2025/01/23 17:51:18 modbus: recv 00 40 00 00 00 09 02 03 06 00 e5 00 e5 00 e6                                                                 
[lp-1  ] DEBUG 2025/01/23 17:51:18 charge voltages: [229 229 230]V               
[lp-1  ] DEBUG 2025/01/23 17:51:18 detected connected phases: 3p                 
[lp-1  ] DEBUG 2025/01/23 17:51:18 detected active phases: 2p                    
[versicharge] TRACE 2025/01/23 17:51:18 modbus: send 00 41 00 00 00 06 02 03 06 9c 00 02                                                                          
[cache ] TRACE 2025/01/23 17:51:18 lp-1/chargeVoltages: [229 229 230]            
[cache ] TRACE 2025/01/23 17:51:18 lp-1/phasesActive: 2                          
[cache ] TRACE 2025/01/23 17:51:18 lp-1/chargeCurrent: 6                         
[... cut ...]
[cache ] TRACE 2025/01/23 17:51:38 lp-1/planProjectedStart: 0001-01-01 00:00:00 +0000 UTC                                                                         
[cache ] TRACE 2025/01/23 17:51:38 lp-1/planProjectedEnd: 0001-01-01 00:00:00 +0000 UTC                                                                           
[cache ] TRACE 2025/01/23 17:51:38 lp-1/planOverrun: 0s                          
[cache ] TRACE 2025/01/23 17:51:38 greenShareHome: 0                             
[cache ] TRACE 2025/01/23 17:51:38 greenShareLoadpoints: 0

andig, 2025-01-23:
https://github.com/evcc-io/evcc/issues/18318#issuecomment-2603205282

andig, 2025-01-23:
Aha:

```
[versicharge] TRACE 2025/01/23 17:51:37 modbus: send 00 45 00 00 00 06 02 03 06 7e 00 03
[versicharge] TRACE 2025/01/23 17:51:37 modbus: recv 00 45 00 00 00 09 02 03 06 04 f9 05 04 ff ff
[lp-1 ] DEBUG 2025/01/23 17:51:37 charge power: 68092W
```

Auf einer Phase kommt da `ff ff` was wir als `uint` interpretieren. @premultiply sollen wir einfach `int` draus machen?

andig, 2025-01-23:
Lädst Du ein 2p Fahrzeug?

Maxheino, 2025-01-23:
Ja.