evcc-io/evcc #18328: Fortsetzung Elli/VW Charger 2 / Ladehistorie / 403 API rate limit exceeded

alexraven1970, 2025-01-21:
Moin, 

nachdem ich die installation auf einem separaten Raspi durchlaufen habe lassen, hat es geklappt.
Allerdings ohne die Fahrzeuge. das VW Template mit der Elli resultierte in Fehlermeldungen. 

Punkt 1 - Wir haben in der Zwischenzeit schon geladen. Allerdings zeigt die Historie nichts an? 

[image]

Punkt 2 - diese Fehlermeldung. Ich finde in der Doku nichts, hab auch hier bei den issues gesucht.
_version check failed: GET https://api.github.com/repos/evcc-io/evcc/releases/latest: 403 API rate limit exceeded for 94.31.109.116. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.) [rate reset in 5m56s] (installed: 0.132.1)_

Die Ladehistorie wäre toll, aber die Fehlermeldung nervt - wie bekomm ich die weg? 
interval ist auf 30s 

Danke für Eure Unterstützung
Alexandra

### Discussed in https://github.com/evcc-io/evcc/discussions/18187

CarG70, 2025-01-21:
da gibt es scheinbar noch einen Bug in der Software der Elli2. Siehe #16357

andig, 2025-01-21:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-01-21 by andig

alexraven1970, 2025-01-23:
update erledigt, logfile anbei

 
ok, log kann ich nicht uploaden ...