evcc-io/evcc #18339: SMA_Modbus Batterie nicht gesperrt

MarSlz, 2025-01-21:
### Describe the bug

Hallo,
Folgende Konfiguration:
SMA SHM 2.0 Firmware 2.15.6.R 
STP10.0-3SE-40 Firmware 3.05.26.R
Easee Home Wallbox
ID4

Obwohl Batterie entladen gesperrt sein sollte, wird diese während des Schnelladens immer wieder kurz aktiv und somit entladen.

Getestet mit v0.132.0
Mit v0.133.0 gleuches Verhalten

### Steps to reproduce

1. Batterie im SchnelladeModus entladen aktivieren
2. Schnelladen starten

[image]

[image]

[image]

[image]

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: xxx

# log settings
log: info
levels:
  site: debug
[... 507 lines cut ...]
[lp-1  ] DEBUG 2025/01/21 20:13:29 charge total import: 5229.544kWh
[lp-1  ] DEBUG 2025/01/21 20:13:29 charger status: C
[lp-1  ] DEBUG 2025/01/21 20:13:29 soc estimated: 62.29% (vehicle: 61.00%)
[lp-1  ] DEBUG 2025/01/21 20:13:29 vehicle soc: 62%
[lp-1  ] DEBUG 2025/01/21 20:13:29 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/01/21 20:13:29 vehicle range: 205km
[sma   ] TRACE 2025/01/21 20:13:29 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/01/21 20:13:30 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/01/21 20:13:30 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/01/21 20:13:30 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/01/21 20:13:30 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2025/01/21 20:13:31 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

premultiply, 2025-01-22:
Setz mal bitte `maxacpower` und probier dann nochmal.

Johprit, 2025-01-22:
#18166 
Wie bei mir

MarSlz, 2025-01-23:
> Setz mal bitte `maxacpower` und probier dann nochmal.

Vielen Dank für den Tipp. Zu maxacpower  kann ich leider nichts finden. Wo und mit welchem Wert soll ich das in der Konfig ergänzen? Dann teste ich das gerne... 

Welche AC Power will ich damit limitieren? Das Grid? den Wechselrichter? die Wallbox?
Danke dir!

n0cturne, 2025-01-23:
maxacpower brachte leider keine Besserung:

[image]

[image]

Konfig hatte auf folgendes geändert:
```
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.90  
  port: 502  
  usage: pv  
  modbus: tcpip  
  maxacpower: 10000
  name: pv1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.90  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery1
  chargepower: 7700
  capacity: 7.7
```

MarSlz, 2025-01-26:
Hallo,
Habe das wie vorgeschlagen probiert. 
Aktuell sieht es für mich so aus, wenn die Batterie jetzt gar nicht mehr gesperrt wird und kontinuierlich mit 4kw entladen wird….

Das Logfile dazu im Anhang:
[evcc-20250126-193451-trace.log](https://github.com/user-attachments/files/18551225/evcc-20250126-193451-trace.log)

Weitere Ideen, was ich testen kann?
Vielen Dank für die Unterstützung.

Oggerschummer, 2025-01-28:
Gleiches Problemchen hier (evcc 0.133.0, Audi Q4, SMA mit BYD HVS).
Die Batterie wird in evcc als gesperrt gezeigt, anscheinend wird das aber ignoriert. 
Zwischenzeitlich hatte es ja mal funktioniert, nun ist es dem Wechselrichter wohl wieder Wurst,
die Batterie lädt und entlädt ohne Hemmungen.

Wechselrichter, an dem die Batterie hängt: SMA  STP8.0-3SE-40-000, Firmware: 3.05.26.R, BMS: 3.23.0.R
SMA SHM2 FW:2.15.7.R 
Wallbox: Alfen SL Pro 6.5.0-4217 (der Vollständigket halber, sollte keine EInfluss haben)

Ramada2020, 2025-02-01:
Ich habe das gleiche Problem mit meinem SMA Sunnyboy 3.7. 
Der Fehler tritt ab Version 0.131.8 auf.  Bei der Version 0.131.7 funktioniert die Sperre noch. 

vermute es liegt an dieser Änderung:
[f0aeb9b](https://github.com/evcc-io/evcc/commit/f0aeb9ba3a986e83b9f85e5c09eadce9180928c1) SMA: fix battery control ([#17393](https://github.com/evcc-io/evcc/pull/17393))

Scorgan01, 2025-02-07:
Bei mir zeigt die Kombi keine Probleme: 
- SMA Tripower 8.0 v3.5.26.R
- SMA HomeStorage
- SMA HM 2.0 v2.15.7.R
- evcc v0.133.0
-Die Batteriesperre funktioniert bei schnellem und geplantem Laden.

Ramada2020, 2025-02-09:
> Ich habe das gleiche Problem mit meinem SMA Sunnyboy 3.7.
> Der Fehler tritt ab Version 0.131.8 auf.  Bei der Version 0.131.7 funktioniert die Sperre noch.
>
> vermute es liegt an dieser Änderung: [...]

Nach einem Neustart aller Geräte funktioniert die Sperre nun doch bei mir.

MarSlz, 2025-02-09:
@Ramada2020 : danke dir für deinen Tipp, habe meinen Wechselrichter/HM und EVCC auch grad versucht neuzustarten und werde weiter testen.

Aktuell sieht es nicht so aus als hätte das geholfen…

premultiply, 2025-02-10:
> Kann man den SHM Neustarten?

Ja, via Sunny Portal per Software oder mit dem richtigen Timing(!) per "Hardwarebutton".

kasn24, 2025-02-12:
Habe das gleiche Problem 133.0
Hatte bisher den Verdacht daran, es liegt an dem Switch "Verhindere Entladung im Schnell-Modus und bei geplantem Laden.", der sich immer wieder deaktiviert (ohne Neustart). Habe noch keine Möglichkeit gefunden es in der Config dauerhaft zu aktiveren.

Zurück zum Hauptproblem:
SHM 2.0
STP 8.0 Hybrid
BYD HVS

In der Grafik sieht man gut den Neustart vom Speicher (sowie SHM). Trotzdem wird immer wieder aus dem Akku gezogen. Weiß nicht ob der dauerhafte Lastwechsel so fein ist für den Speicher. 0 bis 6 Uhr hätte die Batterie gesperrt sein sollen.

[image]
[image]

andig, 2025-02-13:
@premultiply I'm confused- what's the resolution here?

MarSlz, 2025-02-17:
Hallo,
Habe bei mir den SHM 2.0 über die Cloud und den WR über den DC Schalter, Batterieschalter und Socherung ebenfalls aus und eingeschaltet.
In EVCC nutze ich inzwischen 0.200.1.

Das Problem besteht bei mir weiterhin.
Irgendwelche Ideen?

MarSlz, 2025-02-17:
Hier nochmal ein Ausschnitt aus dem Log:

[site ] DEBUG 2025/02/17 23:27:34 ----
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 37 00 00 00 0b 03 10 9d 2c 00 02 04 00 00 09 78
[site ] DEBUG 2025/02/17 23:27:34 grid power: 5753W
[site ] DEBUG 2025/02/17 23:27:34 grid powers: [2046 1885 1823]W
[site ] DEBUG 2025/02/17 23:27:34 grid currents: [8.92 8.21 7.92]A
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 37 00 00 00 06 03 10 9d 2c 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 38 00 00 00 0b 03 10 9f 59 00 02 04 00 00 00 00
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 38 00 00 00 06 03 10 9f 59 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 39 00 00 00 06 03 04 7a a3 00 02
[sma ] TRACE 2025/02/17 23:27:34 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 39 00 00 00 07 03 04 04 00 00 0f a7
[site ] DEBUG 2025/02/17 23:27:34 pv 1 power: 0W
[sma_pv] TRACE 2025/02/17 23:27:34 modbus: send 02 3a 00 00 00 06 03 03 78 35 00 02
[sma_pv] TRACE 2025/02/17 23:27:34 modbus: recv 02 3a 00 00 00 07 03 03 04 00 00 00 00
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 3b 00 00 00 0b 03 10 9f 5b 00 02 04 00 00 00 64
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 3b 00 00 00 06 03 10 9f 5b 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 3c 00 00 00 06 03 04 7a a1 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 3c 00 00 00 07 03 04 04 00 00 00 00
[site ] DEBUG 2025/02/17 23:27:34 battery 1 power: 4007W
[sma_pv] TRACE 2025/02/17 23:27:34 modbus: send 02 3d 00 00 00 06 03 03 78 f1 00 02
[sma_pv] TRACE 2025/02/17 23:27:34 modbus: recv 02 3d 00 00 00 07 03 03 04 00 00 00 00
[site ] DEBUG 2025/02/17 23:27:34 pv 2 power: 0W
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 3e 00 00 00 0b 03 10 9f 5d 00 02 04 00 00 00 00
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 3e 00 00 00 06 03 10 9f 5d 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 3f 00 00 00 06 03 03 7a a9 00 04
[sma ] TRACE 2025/02/17 23:27:34 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 3f 00 00 00 0b 03 03 08 00 00 00 00 00 02 7e 1c
[sma_pv] TRACE 2025/02/17 23:27:34 modbus: send 02 40 00 00 00 06 03 03 77 31 00 04
[sma_pv] TRACE 2025/02/17 23:27:34 modbus: recv 02 40 00 00 00 0b 03 03 08 00 00 00 00 00 05 20 ec
[site ] DEBUG 2025/02/17 23:27:34 pv power: 0W
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 41 00 00 00 0b 03 10 9f 5f 00 02 04 00 00 00 00
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 41 00 00 00 06 03 10 9f 5f 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 42 00 00 00 06 03 03 78 7d 00 02
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 42 00 00 00 07 03 03 04 00 00 00 27
[site ] DEBUG 2025/02/17 23:27:34 battery 1 soc: 39%
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: send 02 43 00 00 00 0b 03 10 9f 61 00 02 04 00 00 00 00
[site ] DEBUG 2025/02/17 23:27:34 site power: 9910W
[sma_battery] TRACE 2025/02/17 23:27:34 modbus: recv 02 43 00 00 00 06 03 10 9f 61 00 02
[sma ] TRACE 2025/02/17 23:27:34 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:34 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:34 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:35 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:35 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:35 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:35 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:35 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:36 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:36 recv 192.168.177.78: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2025/02/17 23:27:36 recv 192.168.177.78:

premultiply, 2025-02-19:
Kann jemand die Lösung des Problems mit dem aktuellen Nightly bestätigen bei dem es jetzt zuvor Probleme gemacht hat?

Mikka20, 2025-02-19:
Hi premultiply,
ich habe zwar keine SMA sondern eine Victron Anlage, aber auch mit dem aktuellen Nightly wird die Batterie nicht gesperrt.
Anbei ein Log, vielleicht hilft der weiter.

VG

[evcc-20250219-212915-debug.log](https://github.com/user-attachments/files/18875353/evcc-20250219-212915-debug.log)

andig, 2025-02-19:
Bitte neues Thema. Hier gehts um Sma.

MarSlz, 2025-02-19:
> Kann jemand die Lösung des Problems mit dem aktuellen Nightly bestätigen bei dem es jetzt zuvor Probleme gemacht hat?

Hab das Nightly AddOn installiert, muss jetzt aber noch sehn, wie ich die Konfig da hinbekomme, das Verzeichnis hat sich ja geändert, und ich kämpfe noch mit der Technik :-).
Sobald mir das gelungen ist, gibt es eine Rückmeldung.

kasn24, 2025-02-21:
Kann leider keine Nightly + Logs liefern, das hatte in der Vergangenheit immer wieder Probleme gemacht. 

Aber noch aus den laufenden Betrieb unter 0.200.0 hat man gestern Abend gesehen, die Batterie Sperre durch den Dummy-Loudpoint ab 23 Uhr hat gegriffen. Als die hohen Verbräuche dazu kamen hat die Batterie etwas mit abgenommen. Aber kurz vor 3 Uhr auf einen Schlag war die Batteriesperre anscheinend gar nicht mehr da, trotz geringem Verbrauch. Der Dummy-Loadpoint hat aber noch bis 6 Uhr "schnellgeladen". 

[image]
[image]

Der Schieberegel  in der Bat-Config war gestern Abend aktiviert, heute morgen deaktiviert. Setzt der sich Nachts immer wieder zurück? Einen von mir geplanten Neustart gab es nicht.

[image]

premultiply, 2025-02-21:
Das hilft uns alles nicht weiter.
Die Wahrheit steht immer im Log.

MarSlz, 2025-02-24:
> Kann jemand die Lösung des Problems mit dem aktuellen Nightly bestätigen bei dem es jetzt zuvor Probleme gemacht hat?

Hallo @premultiply 
habe heute mit der aktuellen Nightly Version getestet:
Problem besteht weiterhin, anbei das Logfile:
[evcc-20250224-195323-trace.log](https://github.com/user-attachments/files/18948597/evcc-20250224-195323-trace.log)

123thomasme, 2025-03-01:
Hallo zusammen,

ich habe bei 2 Analgen die ich betreue die gleichen Probleme. Bei beiden ist die Option Schnellladen Speicher sperren aktiviert.

evcc Version: 0.200.5

Die Logs und Konfiguration schicke ich an info@evcc.io, da in den Logs viele Token drin sind und ich nicht weiß in wieweit diese "Missbraucht" werden könnten.

closed 2025-03-14 by github-actions[bot]

123thomasme, 2025-03-21:
Hallo zusammen,

können sich hier mal die melden bei denen dieser Issue auftritt? Ich möchte mal Wissen ob wir Einzelfälle sind oder ob es generell ein Problem ist.

Gibt es Leute mit einem SMA Hybrid Wechselrichter und Batterie bei dennen es nicht im Schnelllademodus nicht zur Entladung kommt?

kasn24, 2025-03-21:
> können sich hier mal die melden bei denen dieser Issue auftritt? Ich möchte mal Wissen ob wir Einzelfälle sind oder ob es generell ein Problem ist.
> 

Hier - auch unter der aktuellsten Version

michael-brueggemann, 2025-03-22:
Ich bin ein totaler Neuling, habe EVCC erst seit einer Woche am laufen. Also sind meine Aussagen mit Vorsicht zu betrachten. Trotzdem möchte ich hier kommentieren.
Mein Setup: Sunny Boy Smart Energy + BYD HVS Akku
Ich hatte einmal kurz Netzladen ausprobiert, dass ging allerdings nicht.

Ich bin mir nicht sicher ob man pauschal "SMA Hybrid Wechselrichter" sagen kann.
SMA hat 2 Modelle: https://www.sma.de/produkte/hybrid-wechselrichter
1-Phasig: Sunny Boy Smart Energy
3-Phasig: Sunny Tripower Smart Energy

Wenn ich nach der Modbus Dokumentation suche (https://www.sma.de/service/downloads) dann bekomme ich für `Sunny Boy SE` und `Sunny Tripower SE` unterschiedliche Dateien.
[Sunny Boy SE](https://files.sma.de/downloads/PARAMETER-SBSExx-50_03-08-12-R.zip?_gl=1*160t2hy*_gcl_au*MTUzODQ3MzQ4OS4xNzQyNjc2NDI1) / [Sunny Tripower SE](https://files.sma.de/downloads/PARAMETER-HTML_STPxx-3SE-40_30109R_V11.zip?_gl=1*uotmu1*_gcl_au*MTUzODQ3MzQ4OS4xNzQyNjc2NDI1)

Wenn ich z.B. von hier https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/sma-hybrid.yaml die Modbus Adressen nehme, dann finde ich die nicht immer in der SMA Doku.

---

Beispiel für "Batterie halten" von hier: https://github.com/evcc-io/evcc/blob/7ff82549a8a40690af8f2c1d2d977b5713d7c482/templates/definition/meter/sma-hybrid.yaml#L244C26-L244C87
`40799 # CmpBMS.BatDschMaxW - Maximale Batterieentladeleistung`

Für den `Sunny Tripower Smart Energy` finde ich den Wert 40799 in der Doku.
Allerdings gibt es in der Doku dafür auch noch eine weitere Adresse;
`CmpBMS.BatDschMaxW - Maximale Batterieentladeleistung - 44437`

In der Doku für den `Sunny Boy Smart Energy` finde ich die Adresse aber nicht und auch nichts für `Batterieentladeleistung`.

Für `40236 # CmpBMS.OpMod - Betriebsart des BMS` finde ich dafür etwas in der `Sunny Boy SE` Doku, aber nichts in der `Sunny Tripower SE` Doku.

Somit stellen sich mir 2 Fragen:
- auf welcher Doku basiert die EVCC Implementierung?
- verhalten sich beide Hybrid-Wechselrichter gleich?

123thomasme, 2025-03-22:
Ich denke, du solltest einen eigenen Issue aufmachen. 

Da es ja nicht das gleiche Verhalten ist, dass beim Schnellladen die Batterie immer mal wieder kurz entladen wird, obwohl die Batterie gesperrt sein sollte.

premultiply, 2025-03-23:
Das `sma-hybrid`-Template ist mit dem STP SE entwickelt und getestet worden.

Eigentlich™️ sollte das aber auch mit SB SE funktionieren wenn SMA nicht mal wieder irgendwas in der aktuellen Firmware des Geräts vergessen oder falsch gemacht hat...

Alternativ kannst du auch einfach mal für die Batterie das SI- und SBS-Template bezogen auf die Batteriesteuerung probieren.

michael-brueggemann, 2025-03-27:
Ich kann bei mir (Sunny Boy Smart Energy + BYD Akku, sma-hybrid Template) auch das Problem des initialen Posts nachstellen (Batterie nicht gesperrt).
Wie im ersten Post beschrieben wird bei mir auch die Batterie kurz gesperrt, der Wechselrichter scheint es dann aber zu ignorieren. Dann ist sie wieder gesperrt und dann wird es wohl wieder ignoriert.
Der Wechselrichter wechselt auch seinen Statuscode von 307 auf 455 (lese ich per https://github.com/littleyoda/ha-pysmaplus aus).
In der SMA Oberfläche wechselt der Gerätestatus auch von "ok" auf "es gibt ein Problem" (weiß jetzt nicht mehr den aktuellen Wortlaut).

Leider ist diese Issue als "not planned" geschlossen. Wobei die aktive Batteriesteuerung / Sperre laut Doku gehen sollte: https://docs.evcc.io/docs/devices/meters#smart-energy-hybrid-wechselrichter

@premultiply schaut ihr euch das noch an?
Werden noch Logs benötigt? @123thomasme hat wohl schon welche per Mail geschickt.
Bei Bedarf kann ich es morgen bei Sonne auch nochmal testen und Logs bereitstellen.

premultiply, 2025-03-27:
Probiere es bitte wie zuvor beschrieben auch mal mit den anderen SMA Templates für die Batterie.

Für mich sieht das derzeit eher nach einem Bug in der SMA Firmware aus.

Was sagt denn SMA dazu?

Ich kenne das Gerät nicht und solange keine Kommunikationsfehler auf Modbus-Ebene auftreten (die z.B. auf falsche Verwedung von Registern hinweisen könnten) ist es für mich quasi unmöglich da irgendwas zu erkennen.

Auf die SMA-Doku selber würde ich mich nicht zu sehr verlassen. Die ist nicht unbedingt vollständig oder aktuell.

michael-brueggemann, 2025-03-27:
Habe es mit dem SBS Template ausprobiert, gleiches Ergebnis.
Die Batterie ist kurzzeitig gesperrt (etwa 1 Minute) und der Strom kommt aus dem Netz, danach kommt er wieder aus der Batterie. Und dann immer abwechselnd.

Der Wechselrichter wechselt dann den Status:
[image]
[image]

Im Protokoll des Wechselrichters sieht es dann so aus:
[image]

Um 12:37:20 geht das System in den Modus "Batterie extern getrennt", etwa eine Minute später verlässt es den Zustand (12:38:21)

Um 12:40:50 beginnt dann die nächste Runde.
(Ich vermute, dass evcc da wieder den "Befehl zum sperren" geschickt hat)

123thomasme, 2025-03-27:
Bei mir hat der fix:
[f0aeb9b](https://github.com/evcc-io/evcc/commit/f0aeb9ba3a986e83b9f85e5c09eadce9180928c1) SMA: fix battery control (https://github.com/evcc-io/evcc/pull/17393)

das Verhalten deutlich verbessert, aber noch nicht vollständig gelöst. Daher denke ich, dass es evcc liegt.

michael-brueggemann, 2025-03-27:
@123thomasme hast du bei dir in der Batterie Konfig die Watchdog Sekunden eingestellt? (ich nicht)
Und auf was steht der "Regelverhalten" Timer von evcc bei dir? (bei mir 30 Sekunden)

Sind das bei dir `Sunny Tripower SE` oder `Sunny Boy SE` Geräte?

123thomasme, 2025-03-28:
Ich habe nichts bei der Batterie Konfig mit Watchdog konfiguriert. Aber ich habe dazu auch nichts in der Doku gefunden. Kannst du mir das mal schicken wo du das gefunden hast?

Regelverhalten habe ich aktuell auf 30 Sekunden. Man könnte mit dem Wert mal spielen und schauen ob sich das Verhalten verbessert.

Mit meiner aktuellen evcc config hat das ganze schon ein Jahr sehr gut funktioniert.

RoughRodent, 2025-04-06:
Bei mir tritt das Problem nun leider auch auf. 
SMA Tripower Hybrid
BYD HVS 10.2
VW ID4
Evcc 0.202.01

Sobald ich auf Schnellladen gehe, wird der Akku mit 10kW entladen, obwohl es in Evcc verboten ist. Es steht auch „gesperrt“ am Akku in der Übersicht.

Sundixx, 2025-06-05:
Genau wie bei Euch ging bei mir die die Akkusperre und Netzladen nicht.

Bei mir funktioniert es jetzt!

Kurze Antwort:
In der WebUi der Sunny Island unter
Anlagen- und Gerätesteuerung > Wechselrichter > Wirkleistungsverfahren > Betriebsart Wirkleistungsvorgabe

Von: 

Aus --> Externe Wirkleistungsvorgabe

umstellen.

Den Hinweis habe ich in der Doku [https://files.sma.de/downloads/SI-Modbus-BA-de-12.pdf](https://files.sma.de/downloads/SI-Modbus-BA-de-12.pdf) [Technische Information - Kommunikation über SMA Modbus®-Schnittstelle (SI4.4M-12 / SI6.0H-12 / SI8.0H-12 / SI4.4M-13 / SI6.0H-13 / SI8.0H-13)](https://files.sma.de/downloads/SI-Modbus-BA-de-12.pdf)
Stand: 28.05.2024 gefunden

Ich musste mich mit **GridguardCode** anmelden!

@premultiply 
Bei der Gelegenheit habe ich noch mal Modbusparameter in der sma-si-modbus mit der Parameterliste Stand: 19.03.2025 verglichen.
Alles noch OK
Parameter list for SI4.4M-12 / SI6.0H-12 / SI8.0H-12 / SI4.4M-13 / SI6.0H-13 / SI8.0H-13 (Sunny Island 4.4M / 6.0H / 8.0H)

Mein aktuelles Setting ist:
evcc: v0.203.6
Sunny Island 6.0H-13 FW: 3.40.3.R
BYD Battery-Box Premium LVL 15.4 kWh
Sunny Home Manager 2.0 FW: 2.16.4.R

Vielleicht hilft dies jemandem, es würde mich freuen.

Viele Grüße aus Stralsund