evcc-io/evcc #18349: EVCC does not update SOC after car connects and until it starts loading.

MartijnvanGeffen13, 2025-01-22:
### Describe the bug

EVCC does not update SOC when car connects and does not load in solar mode , until it actually starts loading. 

Car got connected but SOC state shows the state the car left / disconnected from EVCC, while the cars API is showing updates EVCC does not seem to poll the API where it gets the SOC untill it starts loading cycle, 

Now this is not good due to min SOC at this moment is set to 75% so car should start loading as it actually is 72% but skipps due to old value in EVCC , luckilly small dif now but i noticed same this morning while car was actualy 50% and should have started loading as MINSOC said 75%.

Looking at logs you see the car connected 14:14 
also in log 14:14 it checked the SOC found 77% , (Could be API was still calling Car as it does that on connect of vehicle) . but between 14:14 and now 15:09 no new call to the API was made to update SOC. 
15:16 setting car to minsolar  you see EVCC doing a SOC check setting SOC to 72% as per API and kicking in to charge at full speed as Minsoc is higher than SOC

Screenshot is from 15:09 still no soc update
setting it to minsolar directly updates in next screenshot

[evcc-20250122-151726-debug.log](https://github.com/user-attachments/files/18506708/evcc-20250122-151726-debug.log)

[image]

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: EX30
    title: EX30
    type: custom
    capacity: 51 # in kWh
    soc:
      source: http
      uri: http://192.168.178.201:6060
      method: GET
      headers:
        - content-type: application/jso
      jq: .data.batteryChargeLevel.value
    status:
      source: http
      uri: http://192.168.178.201:6060
      method: GET
      headers:
        - content-type: application/json
      jq: .data.EvccStatus.value
    range:
      source: http
      uri: http://192.168.178.201:6060
      method: GET
      headers:
        - content-type: application/json
      jq: .data.electricRange.value
```

### Log details

```text
log attached in post
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

1.33

StefanSchoof, 2025-01-22:
See https://docs.evcc.io/en/docs/reference/configuration/loadpoints#poll

MartijnvanGeffen13, 2025-01-22:
darn how did i miss that one , thx,