evcc-io/evcc #18365: Soc Cupra via openWB: 400 (Bad Request)

martin-zh, 2025-01-23:
### Describe the bug

Seit ein paar Tagen schlägt die SoC Abfrage des Cupra Born fehl mit folgender Fehlermeldung:

  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Get "https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/35b583de-4e7e-4437-874f-c4ea82ccd9de/vehicles/xxx/mycar": unexpected status: 400 (Bad Request)"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/35b583de-4e7e-4437-874f-c4ea82ccd9de/vehicles/xxx/mycar\": unexpected status: 400 (Bad Request)", grpc_status:2, created_time:"2025-01-23T07:26:59.617427013+01:00"}"

Soc Abfrage via Cupra App funktioniert weiterhin.

Danke fürs anschauen!
martin

### Steps to reproduce

Soc Abfrage auslösen

### Configuration details

```yaml
Cupra Zugangsdaten inkl EVCC Sponsor Token
```

### Log details

```text
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 110, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Get "https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/35b583de-4e7e-4437-874f-c4ea82ccd9de/vehicles/xxx/mycar": unexpected status: 400 (Bad Request)"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/35b583de-4e7e-4437-874f-c4ea82ccd9de/vehicles/xxx/mycar\": unexpected status: 400 (Bad Request)", grpc_status:2, created_time:"2025-01-23T07:26:59.617427013+01:00"}"
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-23:
Da hat wohl Seat die API geändert. Jemand muss herausfinden, was sich konkret geändert hat.

andig, 2025-01-23:
Getestet, geht. Bitte Zugangsdaten an info@evcc.io

niederradenberg, 2025-01-23:
Interessant ist, dass die Cupra we connect integration weiterhin funktioniert.

DrUdoZucker, 2025-01-23:
Was bedeutet dies für Nutzer der openWB, die den SoC ja von EVVC beziehen? Leider kann ich dies aus diesem Beitrag nicht klar erkennen. Danke für jeden Hinweis.

andig, 2025-01-23:
Service neu gestartet.

closed 2025-01-23 by andig

niederradenberg, 2025-01-23:
Wenn du, auf welche Weise auch immer, EVCC nutzt, gibt es auf dem System wo EVCC läuft, die Config Datei evcc.yaml. Dort ist das Fahrzeug konfiguiert u.a. mit cupra login und optional der VIN. Dort muss dann deine Fahrgestellnummer eingetragen werden

martin-zh, 2025-01-23:
Soc Abfrage klappt wieder - danke für die Hilfe!

DrUdoZucker, 2025-01-23:
Danke auf von mir - gute und schnelle Hilfe!

niederradenberg, 2025-01-23:
Ich hab#s gerade nochmal ohne vin versucht, dann kommt der Fehler wieder. Liegt also nicht am service neu starten. @andig könntest du bitte in der Doku den Parameter "vin" von optional zu mandatory ändern. Danke!

andig, 2025-01-23:
Geht bei mir auch ohne.

niederradenberg, 2025-01-23:
Strange…

Oeffly, 2025-02-01:
Leider wird wieder in der openWB beim Modul EVCC für denn Soc beim Cupra Born wieder nur 0% angezeigt. Beim zweiten Fahrzeug anderer Hersteller geht es.

andig, 2025-02-01:
https://community.fly.io/t/dns-resolution-fails-inside-app-part-4/23728/5 ist aktualisiert

Oeffly, 2025-02-02:
Okay, Danke.... aber das was ich auch noch festgestellt habe ist wenn ich denn Sponsor-Key erneuer funktioniert es gefühlt auch.... aber halt nur die 7 Tage? Oder hat sich das ganze mit dem aktualisieren und bei mir überschnitten....

Oeffly, 2025-02-19:
Leider ist der Soc vom Cupra Born wieder auf 0%

andig, 2025-02-19:
@Oeffly tu dir uns mir bitte einen Gefallen und poste die Fehlermeldung, nicht nur ein #metoo.

Oeffly, 2025-02-20:
`2025-02-20 05:35:16,077 - {modules.vehicles.evcc.api:79} - {DEBUG:fetch soc_ev6} - Vehicle id found in config: 1136
2025-02-20 05:35:16,079 - {modules.vehicles.evcc.api:81} - {DEBUG:fetch soc_ev6} - Fetching SoC for vehicle id: 1136
2025-02-20 05:35:16,350 - {modules.common.fault_state:49} - {ERROR:fetch soc_ev6} - EVCC: FaultState FaultStateLevel.ERROR, FaultStr <class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Get "https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/c40c13b5-10a0-450f-95b0-d23f0393fb/vehicles/VSSZZZxxxxxxxx/mycar": unexpected status: 503 (Service Unavailable)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-02-20T05:35:16.300801749+01:00", grpc_status:2, grpc_message:"Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/c40c13b5-10a0-450f-95b0-d23f0393fb/vehicles/VSSZZZxxxxxxxx/mycar\": unexpected status: 503 (Service Unavailable)"}"
>,), Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 86, in fetch_soc
    response = stub.SoC(
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1160, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1003, in _end_unary_response_blocking
    raise _InactiveRpcError(state)  # pytype: disable=not-instantiable
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Get "https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/c40c13b5-10a0-450f-95b0-d23f0393fb/vehicles/VSSZZZxxxxxxxx/mycar": unexpected status: 503 (Service Unavailable)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-02-20T05:35:16.300801749+01:00", grpc_status:2, grpc_message:"Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/c40c13b5-10a0-450f-95b0-d23f0393fb/vehicles/VSSZZZxxxxxxxx/mycar\": unexpected status: 503 (Service Unavailable)"}"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 110, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Get "https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/c40c13b5-10a0-450f-95b0-d23f0393fb/vehicles/VSSZZZxxxxxxxx/mycar": unexpected status: 503 (Service Unavailable)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-02-20T05:35:16.300801749+01:00", grpc_status:2, grpc_message:"Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/c40c13b5-10a0-450f-95b0-d23f0393fb/vehicles/VSSZZZxxxxxxxx/mycar\": unexpected status: 503 (Service Unavailable)"}"
>`

niederradenberg, 2025-02-20:
Bei mir auch wieder der Fehler. Interessant ist, dass die cupra „we Connect“ Integration weiterhin funktioniert. Was machen die anders?