evcc-io/evcc #18367: Standheizung --> Wallbox aktivieren

dMopp, 2025-01-23:
**Is your feature request related to a problem? Please describe.**
Hi, mir ist aufgefallen, das mit meiner Kombi (KIA EV6, Wallbox Commander 2 und Huawei PV Anlage OHNE Batterie), die Standheizung grundsätzlich aus dem Akku kommt und die Wallbox keinen Strom nachlädt. Eingestellt ist bei mir das Nachladen auf 80% basierend auf Strompreis + überschussladen. 

**Describe the solution you'd like**
Es wäre super, wenn EVCC, nach erreichen des Akkustands, den Akku automatisch auf dem Zielwärt hält und nachlädt,

**Describe alternatives you've considered**
 Falls die Information zur Standheizung aus der Fahrzeugapi kommt, das Fahrzeug einfach laden solange geheizt wird, auch wenn das bedeuted, das man am Ende mehr geladen hat als eingestellt. (Bei 100% sollte generell die Wallbox das Laden nicht beenden sondern immer auf Fahrzeuganforderung warten)

**Additional context**
Bei Fragen, bitte fragen. Ich weiß gerade nicht wie ich es besser beschreiben soll. PS: Der neue Hyundai Ioniq5 bietet Borkonditionierung per Knopfdruck an, auch da wäre es gut, wenn der Akku parallel geladen wird.

andig, 2025-01-23:
Das funktioniert falls evcc die Klimatisierung des Fahrzeugs erkennen kann.

andig, 2025-01-23:
> Bei 100% sollte generell die Wallbox das Laden nicht beenden sondern immer auf Fahrzeuganforderung warten

Interessante Idee.

/cc @naltatis

premultiply, 2025-01-23:
Das war eigentlich auch schon mal so.
Ist mir aber auch schon mal aufgefallen dass es wohl nicht funktioniert hatte.
Wahrscheinlich irgendwo kaputt gegangen.

Ein Test dafür wäre wohl nicht schlecht.

andig, 2025-01-23:
> Das war eigentlich auch schon mal so.

Nein, das war noch nie so. Es wird _nicht_ abgeschaltet wenn das Auto 100% erreicht. Es _wird_ abgeschaltet wenn PV weg ist und das ist heir der Fall. Müssten wir diskutieren ob wir das im allgemeinen Fall anders wollen.

dMopp, 2025-01-23:
> Das funktioniert falls evcc die Klimatisierung des Fahrzeugs erkennen kann.

Also in HomeAssistant gibt es " EV6 Air Conditioner ", das zeigt an, ob die Klima läuft. Leider scheint evcc das nicht zu prüfen, da die Wallbox hier stumm bleibt :|

dMopp, 2025-01-23:
> > Das war eigentlich auch schon mal so.
> 
> Nein, das war noch nie so. Es wird _nicht_ abgeschaltet wenn das Auto 100% erreicht. Es _wird_ abgeschaltet wenn PV weg ist und das ist heir der Fall. Müssten wir diskutieren ob wir das im allgemeinen Fall anders wollen.

Mein Input dazu: Ich finde das sollte man dem Nutzer überlassen. Vorkonditionierung/Heizung soll den bevorzugten Lademodus überschreiben (oder halt nicht). Default wäre dann aus und mit dem aktuellen Verhalten zu vergleichen, so habt ihr keinen Stress mit Usern die durchdrehen und die anderen freuen sich über die neue Option  :D

andig, 2025-01-23:
> Leider scheint evcc das nicht zu prüfen

Wenn jemand das API benennen kann können wir das ändern.

> und die anderen freuen sich über die neue Option

Wir wollen keine überbordenden Optionen- daher müssen wir abwägen.

dMopp, 2025-01-23:
https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api damit kann ich hier aufwarten, mehr aktuell nicht  :|

premultiply, 2025-01-23:
> Es wird abgeschaltet wenn PV weg ist und das ist heir der Fall.

Ah!
Ich hatte wahrscheinlich nur den Ladeplaner benutzt.

andig, 2025-01-23:
Auch der schaltet am Ende ab- es  gibt ja kein PV mehr.

VolkerK62, 2025-01-23:
Bzgl Klimatisierung:
Ich hab es bei mir ("alter" Ioniq 5) mal ausprobiert.
Wenn ich die Klimatisierung per Hand (Bluelink App) einschalte, dann geht
`.resMsg.vehicleStatusInfo.vehicleStatus.airCtrlOn` auf `true`

dMopp, 2025-01-23:
> Bzgl Klimatisierung: Ich hab es bei mir ("alter" Ioniq 5) mal ausprobiert. Wenn ich die Klimatisierung per Hand (Bluelink App) einschalte, dann geht `.resMsg.vehicleStatusInfo.vehicleStatus.airCtrlOn` auf `true`
> 
> Fragt sich aber, ob das Sinn macht, weil diese manuelle Aktivierung nur 15 Minuten läuft.

Wenn es ganz doof läuft, dann wohl nicht. Aber wenn er nach Cache Ablauf Datum noch mal nachfragt, dann kann er in der kurzen Zeit zumindest ein paar kWh nachladen. (Ich habe lieber > 80% im Akku als weniger wenn das Auto dafür warm ist :). Ich begrenze die Lademenge AC extra NICHT im Fahrzeug, da es das konfigurieren in EVCC einfacher macht. So kann ich, vor einer langen Fahrt, mal eben schnell auf 100% laden ohne in die App zu müssen (Das erhöht auch den WAF ungemein)

andig, 2025-01-23:
@VolkerK62 hast Du bitte mal komplettes JSON? Das Feld haben wir nicht.

VolkerK62, 2025-01-23:
Ist natürlich auch die Frage, ob das immer drin ist.
Ich hab auch schon JSON von Hyundai gesehen, wo z.B. die "Ladeklappe" nicht mit drin war.
```
{
    "retCode": "S",
    "resCode": "0000",
    "resMsg": {
        "vehicleStatusInfo": {
            "vehicleLocation": {
                "coord": {
                    "lat": 49.314394,
                    "lon": 8.537147,
                    "alt": 0,
                    "type": 0
                },
                "head": 15,
                "speed": {
                    "value": 0,
                    "unit": 0
                },
                "accuracy": {
                    "hdop": 0,
                    "pdop": 0
                },
                "time": "20250123091005"
            },
            "vehicleStatus": {
                "airCtrlOn": false,
                "engine": false,
                "doorLock": true,
                "doorOpen": {
                    "frontLeft": 0,
                    "frontRight": 0,
                    "backLeft": 0,
                    "backRight": 0
                },
                "trunkOpen": false,
                "airTemp": {
                    "value": "02H",
                    "unit": 0,
                    "hvacTempType": 1
                },
                "defrost": false,
                "acc": false,
                "evStatus": {
                    "batteryCharge": false,
                    "batteryStatus": 47,
                    "batteryPlugin": 0,
                    "remainTime2": {
                        "etc1": {
                            "value": 27,
                            "unit": 1
                        },
                        "etc2": {
                            "value": 980,
                            "unit": 1
                        },
                        "etc3": {
                            "value": 245,
                            "unit": 1
                        },
                        "atc": {
                            "value": 245,
                            "unit": 1
                        }
                    },
                    "drvDistance": [
                        {
                            "rangeByFuel": {
                                "evModeRange": {
[... 176 lines cut ...]
                    }
                },
                "smartKeyBatteryWarning": false,
                "washerFluidStatus": false,
                "breakOilStatus": false,
                "sleepModeCheck": false,
                "time": "20250123091006",
                "remoteWaitingTimeAlert": {
                    "remoteControlAvailable": 1,
                    "remoteControlWaitingTime": 168,
                    "elapsedTime": "00:03:11"
                },
                "systemCutOffAlert": 0,
                "tailLampStatus": 0,
                "hazardStatus": 0
            },
            "odometer": {
                "value": 27597.6,
                "unit": 1
            }
        }
    },
    "msgId": "168bd560-d962-11ef-b07e-96cb97543a8f"
}
```

andig, 2025-01-23:
@runtologist gibts das auch im CCS api?

dMopp, 2025-01-23:
> Ist natürlich auch die Frage, ob das immer drin ist. Ich hab auch schon JSON von Hyundai gesehen, wo z.B. die "Ladeklappe" nicht mit drin war.
--snipp--
> ```

Kannst du mir mal, natürlich ohne deine credentials, das API querry schickend as du verwendest? Dann würde ich für meinen EV6 auch mal schauen was raus kommt.

VolkerK62, 2025-01-23:
du hast dein EV6 doch in evcc konfiguriert?
Dann kommst du da ganz einfach ran.
Auto anstecken, dadurch wird die API abgefragt
Dann UI-Log (Menü oben rechts) öffnen.
Dort "Trace" und "Kia" auswählen.

dMopp, 2025-01-23:
> du hast dein EV6 doch in evcc konfiguriert? Dann kommst du da ganz einfach ran. Auto anstecken, dadurch wird die API abgefragt Dann UI-Log (Menü oben rechts) öffnen. Dort "Trace" und "Kia" auswählen.

Perfekt, danke !

mfuchs1984, 2025-01-24:
Du musst den loadpoint auf poll mode connected umstellen.

dMopp, 2025-01-24:
Das zieht aber nicht wirklich an der 12V, da er die Daten ja von der API holt OHNE einen refresh zu erzwingen, oder ?

mfuchs1984, 2025-01-24:
In der 0.133.0 ist die Änderung doch noch gar nicht drin.

dMopp, 2025-01-24:
Ja, aber es geht trotzdem. Das hat mich gerade auch gewundert. Scheinbar liegt es am polling. Er stellt das laden bei beenden der Klima auch direkt wieder ein (zumindest wenn ich unter 80% liege)

mfuchs1984, 2025-01-24:
Das macht keinen Sinn, das muss an was anderem liegen.

dMopp, 2025-01-24:
Erklären kann ich es mir auch nicht. Ich lade gerade auf 80 und versuche es noch mal. Wenn es dann wieder geht —> KANN es nur am Polling liegen. Mehr habe ich nicht verändert

andig, 2025-01-24:
@dMopp ist hier alles OT, bitte beim Thema bleiben.

VolkerK62, 2025-01-24:
@mfuchs1984  danke für den `connected` Hinweis. Damit funktioniert es.

andig, 2025-01-25:
Keine neuen Erkenntnisse, closing.