evcc-io/evcc #18373: Kostal KSEM with Software >=2.6.0: Modbus Address changes to establish Sunspec conformity

illingkipv, 2025-01-23:
This affects KOSTAL KSEM with SW Version 2.6.0 and above.
The Modbus register addresses 40069-40177 receive an offset of +1 in order to establish Sunspec conformity.

Example: 
- SW 2.5.x:
  - 40069 C_SunSpec_DID
  - 40070 C_SunSpec_Length
  - ...
  - 40087 M_AC_Power
  - ...
  - 40177 C_SunSpec_Length

- SW 2.6.0+:
  - 400**70** C_SunSpec_DID
  - 400**71** C_SunSpec_Length
  - ...
  - 40088 M_AC_Power
  - ...
  - 40178 C_SunSpec_Length

illingkipv, 2025-01-23:
@andig FYI: SW Release is estimated around End of February.

andig, 2025-01-23:
Ich weiss leider nicht, was ich mit der Information machen soll. Das KSEM Template verwendet bereits Sunspec- mir ist kein Problem damit bekannt.

illingkipv, 2025-01-23:
> Ich weiss leider nicht, was ich mit der Information machen soll. Das KSEM Template verwendet bereits Sunspec- mir ist kein Problem damit bekannt.

Falls ihr bisher Adressen in dem Bereich genutzt habt, wird das KSEM mit der neuen SW nicht mehr mit EVCC funktionieren. Deshalb die Vorabinfo.

andig, 2025-01-23:
Falls wir die jetzt verwendet hätte _und_ sie bei Kostal falsch wären würde es aktuell nicht funktionieren. Kann also nur besser werden solange Kostal nix kaputt macht...

closed 2025-01-23 by andig