evcc-io/evcc #18376: Ladefreigabe, wenn die Fahrzeugposition im spezifizierten Umkreis zum Ladepunkt ist

efdece, 2025-01-23:
**Is your feature request related to a problem? Please describe.**
Es ist kein Problem, sondern eine Bequemlichkeitsfunktion.

**Describe the solution you'd like**
Beim Testen mit `evcc vehicle` ist mir aufgefallen, dass die aktuelle Fahrzeugposition mit ausgegeben wird.
Wenn man eine Position für einen Ladepunkt angeben könnte, dann wäre eine Freischaltung möglich, wenn das Fahrzeug in einem spezifizierten Umkreis zum Ladepunkt ist und angeschlossen wird.

**Describe alternatives you've considered**
Wenn die Fahrzeugposition extern (API, MQTT, etc.) abfragbar wäre, könnte man es auch ausserhalb von EVCC schalten.

**Additional context**
Ich habe noch nicht getestet was passiert, wenn mein Fahrzeug an einer anderen Wallbox angeschlossen wird und zeitgleich ein fremdes Fahrzeug an meine EVCC kontrollierte Wallbox, wo mein Fahrzeug als Default eingetragen ist.
Wird das Fremdfahrzeug geladen?

andig, 2025-01-23:
> Ladefreigabe

Ist unspezifisch. Ob freigegeben wird hängt ja vom Modus ab. Was erwartest Du also von einer Ladefreigabe? Was bringt die Position wenn wir eine Fahrzeugidentifkation an der WB haben? Wo ist der Vorteil?

efdece, 2025-01-23:
Aktuell habe ich defaultmäßig ein Fahrzeug eingestellt und einen festen Modus. Meist PV.
Wenn das Fahrzeug angesteckt ist, genügend PV vorhanden fängt er ja einfach an zu laden.
Ja, das Fahrzeug ist konfiguriert, aber wie weiß EVCC, dass genau dieses Fahrzeug an der Wallbox hängt und nicht an der vom Nachbarn? Da wird doch nichts über das Ladekabel übertragen oder übersehe ich da was?
Wenn EVCC erkennt, Fahrzeug angeschlossen, aber das konfigurierte Fahrzeug ist zu weit weg, dann sollte das Laden nicht starten.

VolkerK62, 2025-01-23:
Offenischtlich hast du ein "online" Fahrzeug.
Dann findet die Erkennung über den Abgleich des Ladestatus statt.
Wenn die Wallbox meldet, dass ein Fahrzeug angesteckt wurde, dann werden alle konfigurierten Fahrzeuge abgefragt.
Meldet dann genau ein Fahrzeug, dass es "verbunden" ist, dann ist das ein Match.
Das ist natürlich keine 100% sichere Variante.
Wenn bei dir jetzt ein Fremdfahrzeug angeschlossen wird und du gleichzeitg dein Auto an einer anderen Wallbox ansteckst, dann würde das für evcc so aussehen, als wenn du an deiner eigenen Wallbox angesteckt hast.

andig, 2025-01-23:
> oder übersehe ich da was?

Über die Fahrzeug-APIs. Dazu gibts eine Heuristik. _die_ könnte man jenseits von Status ggf. um Location ergänzen.

efdece, 2025-01-23:
Wie gesagt, kam mir beim Testen so in den Sinn als ich die Positionsinfos gesehen habe.
Wäre es denn möglich dem Loadpoint eine Position zu geben und die mit der Fahrzeugposition vor dem Starten abzugleichen? Ist jetzt nicht super wichtig, würde aber eine mögliche Lücke schließen, wenn man es nutzt.

VolkerK62, 2025-01-23:
Wenn du dein Fahrzeug als "default" am Loadpoint eingetragen hast, findet kein Erkennungsprozess statt.
https://docs.evcc.io/docs/reference/configuration/loadpoints#vehicle

andig, 2025-01-23:
> Ist jetzt nicht super wichtig, würde aber eine mögliche Lücke schließen, wenn man es nutzt.

Probier doch erstmal die verfügbaren Optionen durch ;). Erstmal kein akuter Implementierungsbedarf.