evcc-io/evcc #18383: Pushover Message don´t work correcly after last update

Maddin2020, 2025-01-23:
### Describe the bug

After the last update the Pushover messsage don´t work correcty. Before the last version the value are correct, but after the Update i only see the value 0. 

If it possible that you change everythink in the configuration or in the tag name. I have check the documentation but i see no change. In the Log i can see everythink value correct. 

Here Screenshot for a better understanding

[image]
[image]

### Steps to reproduce

1. Start the charging
2. See the pushover message
3. Stop the charging 
4. See the pushover message

### Configuration details

```yaml
messaging:
  events:
    start: # charge start event
      title: Ladevorgang {{.vehicleTitle}} gestartet
      msg: |
        Wallbox {{.title}} startet Ladevorgang von {{.vehicleTitle}} im {{ toString .mode | upper }} Modus.
        --------------------------
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: Ladevorgang {{.vehicleTitle}} beendet
      msg: |
        Wallbox {{.title}} beendet Ladevorgang von {{.vehicleTitle}} 
        mit {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
        --------------------------
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} mit Wallbox {{.title}} verbunden"
      msg: |
        {{.vehicleTitle}} wurde mit der Wallbox {{.title}} mit {{round (divf .pvPower 1000) 2 }} kW PV Leistung verbunden.
        --------------------------
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle disconnected event
      title: "{{.vehicleTitle}} wurde von der Wallbox {{.title}} getrennt"
      msg: |
        {{.vehicleTitle}} wurde von der Wallbox {{.title}} nach {{.connectedDuration}} getrennt.
        --------------------------
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
```

### Log details

```text
the log have no special information
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.133

andig, 2025-01-23:
Probier mal bitte .grid.power etc. geht das?

andig, 2025-01-23:
Ps.: Leistung bei einem beendeten Ladevorgang ergibt wenig Sinn?

closed 2025-01-23 by andig

Maddin2020, 2025-01-23:
hi andig, leider geht .grid.power nicht 

        Wallbox {{.title}} startet Ladevorgang von {{.vehicleTitle}} im {{ toString .mode | upper }} Modus.
        --------------------------
        Netz-Leistung: {{round (divf **.grid.power** 1000) 3 }} kW

Kann es noch was anderes sein ?

andig, 2025-01-23:
Was heisst „geht nicht“? grid.Power?

Maddin2020, 2025-01-23:
leider das gleiche ergebnis es wird nur 0 angezeigt weder grid.Power oder .grid.power es wird immer 0 angezeigt

Maddin2020, 2025-01-23:
Netz-Leistung: {{round (divf .grid.Power 1000) 3 }} kW
Solar-Leistung: {{round (divf .pv.Power 1000) 3 }} kW

Netz-Leistung: {{round (divf grid.Power 1000) 3 }} kW
Solar-Leistung: {{round (divf pv.Power 1000) 3 }} kW

Netz-Leistung: {{round (divf grid.power 1000) 3 }} kW
Solar-Leistung: {{round (divf pv.power 1000) 3 }} kW

hab ich durchprobiert, groß oder klein leider immer Null nur HomePower geht

Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW

andig, 2025-01-26:
Ich hab das kurz getestet:

     msg: "Netz-Leistung: {{ .grid.Power }}"

geht bei mir wunderbar.

Maddin2020, 2025-01-27:
Hi Andig,

danke fürs nachhaken ich habe den Fehler gefunden. Ich brauch einen Neustart damit die yaml übernommen wird ;-) Hab allerdings noch eine Frage. Kann das sein das die Zeile

`{{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}`

übersprungen wird wenn ich eine Entladesperre vom Akku habe. Diese Zeile wird bei mir nicht angezeigt. Oder wie wird ermittelt das batteryConfigured true ist ? Ich habe natürlich ein Akku konfiguriert. 

Außerdem verstehe ich den Syntax der msg noch nicht 

Netz-Leistung: {{round (divf .grid.Power 1000) 3 }} kW

Der Befehl heißt doch es wird gerundet durch 1000 dividiert und 3 Stellen hinter dem Komma angezeigt. Bei mir zeigt er manchmal 3 manchmal 2 Kommazahlen an...

PS: die Doku zur Message müsste man noch anpassen dort steht ja noch grip.Power anstatt .grid.Power

[](https://docs.evcc.io/docs/reference/configuration/messaging#msg)

DANKE FÜR DEN TOLLEN SUPPORT. EVCC ist echt super!