evcc-io/evcc #18385: Corrupt link in configuration

jori0001, 2025-01-23:
### Describe the bug

In der Konfiguration sind die Links zu myteslamate falsch. 
Bitte korrigieren. Danke. Ich liebe die Software einfach. 🥰

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Na
```

### Log details

```text
Na
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-23:
@naltatis fürs UI brauchen wir einen Trick?

jori0001, 2025-01-24:
Hi, ich meinte hier nicht das Problem, dass der Tesla nicht mehr erkannt wird, sondern die grünen Links unter den Eingabefeldern.
Hier wird im Link hinten das ")." mitgegeben. Von daher nur eine Kleinigkeit.

closed 2025-02-07 by andig