evcc-io/evcc #18391: Tesla wird nicht mehr erkannt

jori0001, 2025-01-24:
### Discussed in https://github.com/evcc-io/evcc/discussions/18382

<div type='discussions-op-text'>

<sup>Originally posted by **jori0001** January 23, 2025</sup>
Habe die aktuelle EVCC Version und auch alles bzgl. Tesla wie beschrieben angepasst. Hat auch wunderbar funktioniert. Jetzt, nach ein paar Tagen, hab ich ihn mal wieder eingesteckt und ich bekomme nur den Fehler

cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired

Ich habe auf myteslamate die Token erneuert und eingetragen. Klappt trotzdem nicht. 
Jemand ne Idee, was ich machen könnte?</div>

dakky, 2025-01-24:
Logmeldung dazu:
```
evcc  | [lp-1  ] INFO 2025/01/24 07:35:29 car connected
evcc  | [main  ] ERROR 2025/01/24 07:35:30 vehicle status: Get "https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/<REDACTED>/vehicle_data": oauth2: "login_required" "The refresh_token is expired."
```

auch ein test im config UI => Fahrzeug bearbeiten => Prüfen gibt ein einen fehler

```
cannot create vehicle type 'template': cannot create vehicle type 'tesla': Get "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region": oauth2: "login_required" "The refresh_token is expired."
```

andig, 2025-01-24:
Token ist abgelaufen, vmtl. mehrfach benutzt. An der evcc Logik hat sich nichts geändert.

BlueSnake830, 2025-01-31:
Bei mir ist der selbe Fehler aufgetreten. Der Token wird aber nur von evcc genutzt.

andig, 2025-02-01:
Habt ihr berücksichtigt, dass der Token in myteslamate abgemeldet werden muss? /cc @naltatis

BlueSnake830, 2025-02-01:
Guter Punkt, muss ich mal schauen. Das mit dem abmelden bei myteslamate habe ich entweder überlesen, oder es fehlt in der Anleitung.

dakky, 2025-02-02:
hmm ne in teslamate hab ich nichts ausgetragen. Steht auch nirgendwo? Was genau muss ich da machen?

ich glaube die Doku braucht an der stelle eh noch etwas mehr liebe ;-) ich hab mir das so aus einer Mischung von eurer Doku und der teslamate Doku zusammen getragen

Heimdall-One, 2025-02-03:
Bei mir verbindet sich EVCC auch mit ausgetragenem Teslamate nicht mit dem Model Y seit dem Update.

naltatis, 2025-02-08:
Doku (bzw. erst einmal der erklärende Blogartikel) hat soeben etwas mehr Liebe und Details erhalten.
https://github.com/evcc-io/docs/pull/723

Perspektivisch werden wir noch einen besseren Ort/Mechanismus für gerätespezifische Anleitungen einführen. Für jetzt muss es der Blog-Artikel tun :D

closed 2025-02-08 by naltatis