evcc-io/evcc #18400: Fahrzeug wird nicht geweckt, bevor ein Befehl gesendet wird (Proxy über TeslaMate)

D4rkH0ur, 2025-01-24:
### Describe the bug

Ich habe EVCC Version 0.133.0 im Einsatz und nutze TeslaMate als Proxy. Mein Fahrzeug (Tesla Model Y) wird nicht aufgeweckt, bevor ein Befehl wie set_charging_amps gesendet wird. Stattdessen schläft das Fahrzeug ein, was zu folgenden Fehlern führt:
500 Internal Server Error: Server error: `POST <redacted_url>client-tfqx9dzdpsy-teslamate-cmd:4443/api/1/vehicles/<redacted_vin>/command/set_charging_amps` resulted in a `500 Internal Server Error` response:
{"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}

### Steps to reproduce

1.Konfiguration: EVCC mit TeslaMate-Proxy nutzen.
2.Fahrzeug schläft ein.
3.Ladebefehl senden, ohne dass das Fahrzeug zuvor aufgeweckt wurde.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: <redacted>
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <redacted>

interval: 30s # control cycle interval

sponsortoken: <redacted>

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 16  
  host: <redacted>
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 30s
  name: grid1

- type: template
  template: huawei-dongle-powersensor 
  id: 16  
  host: <redacted>
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 30s  
  name: pv2

- type: template
  template: huawei-dongle-powersensor 
  id: 16  
  host: <redacted>
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 30s 
  name: battery3
  storageunit: 1 # optional
  maxchargepower: 5000 # Leistungslimit für Netzladung. (optional)
  capacity: 15 # Akkukapazität in kWh (optional)
  forceaccharging: true # AC-Laden bleibt aktiv zum Laden des Speichers aus anderen AC Quellen für Wechselrichterkaskaden. Verhindert Stand-by. (optional)
[... 242 lines cut ...]
[site  ] DEBUG 2025/01/24 17:00:45 grid currents: [8.76 7.3 6.67]A
[site  ] DEBUG 2025/01/24 17:00:46 battery 1 soc: 40%
[site  ] DEBUG 2025/01/24 17:00:47 site power: 4961W
[lp-1  ] DEBUG 2025/01/24 17:00:47 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2025/01/24 17:00:47 charger status: B
[lp-1  ] ERROR 2025/01/24 17:00:50 max charge current 16A: 500 Internal Server Error: Server error: `POST <redacted_url>client-tfqx9dzdpsy-teslamate-cmd:4443/api/1/vehicles/<redacted_vin>/command/set_charging_amps` resulted in a `500 Internal Server Error` response:
{"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}

[site  ] DEBUG 2025/01/24 17:01:01 ----
[lp-1  ] DEBUG 2025/01/24 17:01:01 charge power: 0W
[lp-1  ] DEBUG 2025/01/24 17:01:01 charge currents: [0.1 0.2 0.3]A
[site  ] DEBUG 2025/01/24 17:01:01 pv 3 power: 0W
[site  ] DEBUG 2025/01/24 17:01:02 grid power: 4929W
[site  ] DEBUG 2025/01/24 17:01:03 battery 1 power: -49W
[site  ] DEBUG 2025/01/24 17:01:04 pv 1 power: 0W
[site  ] DEBUG 2025/01/24 17:01:05 pv 2 power: 0W
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-24:
Ja, da hat Tesla das Backend geändert. Da jetzt Status 500 kommt wird der Fehler von evcc nicht erkannt.

andig, 2025-01-24:
Closed in https://github.com/evcc-io/evcc/commit/0a83391856b6092c5c69677d09c3a2ca9f3f9940

closed 2025-01-24 by andig

mucki12, 2025-02-04:
@andig 
Wahrscheinlich werde ich mir jetzt einen Rüffel einfangen und eigentlich ist es auch nicht meine Art, aber gibt es eventuell eine Timeline wann die nächste final mit diesem RQ erscheint?
Alle, die es jetzt mir Müh und Not schaffen die doch umfangreichen Änderungen für Tesla umzusetzen stehen ansonsten vor dem nächsten mit diesem RQ natürlich schon gelöstem Problem, da wohl die Minderheit die nightly einsetzt...