evcc-io/evcc #18402: Tariffs: Enever - API token limit exceeded

lenvm, 2025-01-24:
### Describe the bug

I am getting the following log output

```
[tariff] ERROR 2025/01/24 19:26:27 jq: query failed: cannot iterate over: string ("API token limit exceeded ...")
[main ] ERROR 2025/01/24 19:26:27 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'custom': jq: query failed: cannot iterate over: string ("API token limit exceeded ...")
```

Apparently the Enever API request has been triggered more than 250 times this month.

Now EVCC cannot retrieve the tariff, until the Enever API is reset on the first of the month, and then again when the request has been triggered 250 times.

A possible solution would be to reduce the number of API calls, to stay under the limit of 250 calls per month.

### Steps to reproduce

1. Add to the config file:
```
tariffs:
  grid:
    type: template
    template: Enever
    token: secret
    provider: secret
```
2. Let EVCC run for multiple days
3. Check log for the error reported

### Configuration details

```yaml
tariffs:
  grid:
    type: template
    template: Enever
    token: secret
    provider: secret
```

### Log details

```text
[tariff] ERROR 2025/01/24 19:26:27 jq: query failed: cannot iterate over: string ("API token limit exceeded ...")
[main ] ERROR 2025/01/24 19:26:27 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'custom': jq: query failed: cannot iterate over: string ("API token limit exceeded ...")
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-01-25:
evcc makes 1 request per hour, enever requires 2. If 48 requests per day (!) are too much for enever I'm afraid we can't support it. I'd suggest talking to enever to reduce this limit.

closed 2025-01-25 by andig

lenvm, 2025-01-25:
Can we think of a way to reduce the number of requests, to 1 or 2 per day, around 16:00 as then the tariffs for tomorrow are available?

andig, 2025-01-25:
Feel free to open a PR. I don't see a good way to do this generically via templates.

wgentine, 2025-01-25:
This integration doesnt work long enough and reach the token limit.