evcc-io/evcc #18440: Tesla: Extend Command Proxy Support Beyond myteslamate.com

tkoenig89, 2025-01-26:
Building on the support for the Tesla Fleet API and Commands (via Proxy) introduced in #17982, I would like to propose extending the feature to support additional command proxies beyond `myteslamate.com`.

### Current State
Currently, it is possible to configure a custom `commandProxy` and `proxyToken` as follows:

```yaml
vehicles:
  - my_tesla
    ...
    commandProxy: http://tesla-commands.local/
    proxyToken: eyJhbGciOiJ...
```

This setup works well initially. However, issues arise when the `proxyToken` expires and requires a refresh.

### Use Case
I am using the official [Tesla Command Proxy](https://github.com/teslamotors/vehicle-command) with a custom domain. Since this official proxy forwards requests using the provided access token, the need for an additional `proxyToken` becomes redundant in some cases.

### Proposed Solution
To improve flexibility and usability, I propose the following enhancements:

1. **Support for Token Reuse:**
  If a `commandProxy` is provided without a `proxyToken` (or with a flag like `reuseAccessToken: true`), the system could leverage the existing access and refresh tokens.
  This would simplify setups that use proxies like the official Tesla Command Proxy, where access and refresh tokens suffice for authorization.

2. **Handle Proxy Token Refresh:** 
  If a `proxyToken` is configured, implement automatic token refresh similar to the existing process for `accessToken`.
  This ensures continued functionality without manual intervention when the token expires.

### Rationale
The current implementation for `myteslamate.com` likely stores the access and refresh tokens on their server and uses an API token (proxyToken) for authentication. Expanding support to handle refresh tokens for other proxies should align closely with the existing logic, requiring minimal changes.

By extending this functionality, users can integrate alternative proxies seamlessly while maintaining reliability.

Thank you for considering this feature request. I'd be happy to provide more details or assist with testing if needed!

andig, 2025-01-26:
I absolutely understand this request. However, there is only limited demand for it. Seems the core team won‘t work on this. But we‘d be happy to take a PR (basically just expose the oauth2 config).

closed 2025-01-27 by andig

martin-eberle, 2025-03-15:
Since myteslamate is quite unstable (at least in my use case) I would like such a solution.

andig, 2025-03-15:
Feel free to provide a PR

tkoenig89, 2025-03-16:
@martin-eberle I managed to find a solution that works quite well for me. Maybe this works for you as well:

[https://github.com/L480/tesla-http-api](https://github.com/L480/tesla-http-api) is doing what myteslamate does in the 'normal' setup. It proxies the commands to tesla and handles token refresh (which was my original pain point). 

I did fork the repository and made some more settings available via environment variables: [https://github.com/tkoenig89/tesla-http-api](https://github.com/tkoenig89/tesla-http-api). The container is not on docker-hub at the moment, i just built it locally for my setup.

My evcc.yaml setup looks like this. 'proxyToken' is the api token for the tesla-http-api. 

```yaml
    commandProxy: http://tesla-http-api:8080/ #http://tesla-commands.local/
    proxyToken: njdsjkf8i43934kfdkf9432k34590
```

My redacted docker-compose.yaml:

```yaml
services:
  evcc:
    command:
      - evcc
    container_name: evcc
    image: evcc/evcc:0.133.0
    ports:
      - 7070:7070/tcp
      - 8887:8887/tcp
    volumes:
      - /local/evcc/evcc.yaml:/etc/evcc.yaml
      - /local/evcc/db:/root/.evcc
    restart: unless-stopped

  tesla-http-api:
    container_name: tesla-http-api
    hostname: tesla-http-api
    image: tkoenig89/tesla-http-api:latest
    user: "93761"
    environment:
      TZ: Europe/Berlin
      TESLA_REFRESH_TOKEN: EU_30sd
      TESLA_CLIENT_ID: 9716f6
      TESLA_PRIVATE_KEY_FILE: /var/lib/tesla-http-api/private-key.pem
      ENABLE_API_TOKEN: "true"
      API_TOKEN: njdsjkf8i43934kfdkf9432k34590
      TESLA_REFRESH_TOKEN_FILE: /var/lib/tesla-http-api/refresh-token
    volumes:
      - /local/tesla-api/:/var/lib/tesla-http-api/
    restart: unless-stopped
```

Hope this is helpful.

tkoenig89, 2025-11-01:
Hey @johnrogers ,

thats exactly what i see when i try upgrading to the latest version. 
My setup is still running v0.133 which i never bothered to update until a few days ago. I tried updating, and was running into this 403 error. I did not investigate a lot at the time, but it sure must be the same issue.

For some reason it has been changed in this PR [https://github.com/evcc-io/evcc/pull/18962](https://github.com/evcc-io/evcc/pull/18962).

We might get around this by disabling the check of token on the proxy. Which should be okay if the proxy is only accessible within a secure network.

```yaml
ENABLE_API_TOKEN: "false"
```

johnrogers, 2025-11-02:
> Hey [@johnrogers](https://github.com/johnrogers) ,
> 
> thats exactly what i see when i try upgrading to the latest version. My setup is still running v0.133 which i never bothered to update until a few days ago. I tried updating, and was running into this 403 error. I did not investigate a lot at the time, but it sure must be the same issue.
> 
> For some reason it has been changed in this PR [#18962](https://github.com/evcc-io/evcc/pull/18962).
> 
> We might get around this by disabling the check of token on the proxy. Which should be okay if the proxy is only accessible within a secure network.
> 
> ENABLE_API_TOKEN: "false"
> [Update] Tested and works. I'm now running this setup on 0.209.6 now.

Ah, good to know. My proxy instance is hosted publicly on a VPS so disabling the token check isn’t wise. I’ve forked the repo for now and adjusted it to accept X-Authorization and opened an issue: https://github.com/L480/tesla-http-api/issues/7

I’ll attempt a PR this week and let you know. Thank you.

L480, 2025-11-02:
FYI: https://github.com/L480/tesla-http-api/pull/10

tesla-http-api now also supports the X-Authorization header for authentication.

johnrogers, 2025-11-04:
> Add
> 
> ```
> —log-headers
> ```
> 
> To see headers in the trace log, too.

Thanks @andig! Where does that flag go, do I add it to extra parameters of the Docker container, or in the config file?

andig, 2025-11-06:
Looking at the code, it *seems* that the logging http client/transport is passed down into the command proxy. If that doesn't work (and it seems it doesn't) it needs somebody with debugger to take a look...

tkoenig89, 2025-11-06:
@johnrogers Not sure this is helpful here, I stumbled over this in the past and also when I updated my compose stack:

evcc does not use the proxy exclusively to talk to the tesla cars. There are some requests (maybe wakeup?) that are done without the proxy. For this to work i always had to put a fresh access/refresh token in my evcc.yml file as well. I'm at work and don't have my config at hand, but this always caused a headache for me until i remembered i had to update these as well. 

That might be some issue/miss-configuration on my side, but It seems like evcc does only keep the refresh token in memory and loses a valid refresh token on restart (as long as the refresh token is beyond its lifetime).

So tldr: ensure that you have valid tesla tokens inside your proxy and evcc.

tkoenig89, 2025-11-07:
@johnrogers i did not debug it at all, but from the pull-requests code, i would guess that there might be two headers present:

```go
token := r.Header.Get("Authorization")
if token == "" {
    token = r.Header.Get("X-Authorization")
}
```

In case there is anything in the "Authorization" header, the "X-Authorization" header is not used.
Your Fork does only access the "X-Authorization" header. So maybe evcc does send something in the "Authorization" as well?

I did a version of this myself (with help of buddy copilot) and have a very verbose token check, that also ensures that only a single header is filled or both having the same content. [my-fork](https://github.com/tkoenig89/tesla-http-api/blob/966307020304cd6763a6b835f46c4714fd0051fe/cmd/tesla-http-api/main.go#L42)