evcc-io/evcc #18441: Fahrzeug-Priorisierung im PV-Modus auch wenn alle Fahrzeuge aktuell < Min. SOC

Steph6431, 2025-01-26:
Wenn mehrere Fahrzeuge im PV-Modus connected sind und aufgrund einer Limitierung nicht beide gleichzeitig laden können, sollte das priorisierte Fahrzeug auch mit Prio geladen werden. 

Das nicht priorisierte Fahrzeug sollte zugunsten des priorisieren Fahrzeugs den Ladevorgang unterbrechen bis das priorisierte Fahrzeug sein Min. SOC Ziel erreicht hat und dann wieder den Ladevorgang fortsetzen. 

Siehe: https://github.com/evcc-io/evcc/discussions/18439

closed 2025-01-26 by andig

Steph6431, 2025-01-26:
Bei mir nicht: https://github.com/evcc-io/evcc/discussions/18436

andig, 2025-01-26:
Dann mach gerne ein Issue. Bitte kein Suchspiel. Danke.

Steph6431, 2025-01-26:
Issue hatte ich eröffnet und Details verlinkt. Wollte niemanden überfordern.

VolkerK62, 2025-01-28:
@andig  Info zur Vollständigkeit:
Loadpoint mit Prio1 lädt auf Basis minSoc
Loadpoint mit Prio2 wird angesteckt und will auch auf Basis minSoc laden
Das Einschalten von Loadpoint mit höherer Prio wird durch Lastmanagement verhindert, weil Lastmanagement (noch) keine Prio berücksichtigt.
Hier müsste erst die bestehende Ladung mit nierdriger Prio beendet werden, damit die Ladung mit höherer Prio starten kann.

andig, 2025-01-28:
MinSoc ist wie schnellladen. Da gelten keine Prioritäten.