evcc-io/evcc #18442: Javascript error displaying the charging sessions in Qt 5.15 LTS WebView/WebEngineView

Simpson474, 2025-01-26:
### Describe the bug

My setup is a little bit uncommon as I use evcc within a WebView or WebEngineView of Qt 5.15 LTS . Everything was working fine before adding a tariff in the configuration causing evcc to display the charging cost. In the Qt log, the following message is printed on displaying the charging sessions and selecting the price tab:
```
js: RangeError: maximumFractionDigits value is out of range.
```
The price graph is not visible, hovering over the price graph causes further exceptions:
```
js: Uncaught TypeError: Cannot read property 'hitRadius' of undefined
```

The issue is very likely related to a quite old report at [Stack Overflow](https://stackoverflow.com/questions/41045270/range-error-with-tolocalestring-with-maximumnumber-of-digits-0). The issue seems to be gone in Firefox and Safari, but the QtScript Javascript implementation in Qt still throws the exception. The following patch solves the issue and the price graph is fully functional:

```diff
diff --git a/assets/js/mixins/formatter.js b/assets/js/mixins/formatter.js
index 2bb21bc1..375a62ee 100644
--- a/assets/js/mixins/formatter.js
+++ b/assets/js/mixins/formatter.js
@@ -268,6 +268,7 @@ export default {
         style: "currency",
         currency,
         currencyDisplay,
+	minimumFractionDigits: decimals ? undefined : 0,
         maximumFractionDigits: decimals ? undefined : 0,
       }).format(amout);
```

### Steps to reproduce

1. Create a Qt 5.15 LTS QML Application and add a WebView to display evcc
2. Select charging sessions
3. Select the price tab

### Configuration details

```yaml
meters:
- name: grid1
  type: custom
  power:
    source: calc
    add:
    - source: http
      method: GET
      uri: http://127.0.0.1:8002/getPlainValue/modbus.1.inputRegisters.30867_Metering_GridMs_TotWOut
      scale: -1
    - source: http
      method: GET
      uri: http://127.0.0.1:8002/getPlainValue/modbus.1.inputRegisters.30865_Metering_GridMs_TotWIn
- name: bat1
  type: custom
  power:
    source: http
    method: GET
    uri: http://127.0.0.1:8002/getPlainValue/modbus.0.inputRegisters.30775_GridMs_TotW
  soc:
    source: http
    method: GET
    uri: http://127.0.0.1:8002/getPlainValue/modbus.0.inputRegisters.30845_Bat_ChaStt
  batterymode:
    source: switch
    switch:
    - case: 1 # normal
      set:
        source: http
        method: GET
        uri: http://127.0.0.1:8002/set/0_userdata.0.sunny-boy-storage.evcc_BatteryMode?value=1
    - case: 2 # hold
      set:
        source: http
        method: GET
        uri: http://127.0.0.1:8002/set/0_userdata.0.sunny-boy-storage.evcc_BatteryMode?value=2

- name: pv1
  type: custom
  power:
    source: http
    method: GET
    uri: http://127.0.0.1:8002/getPlainValue/0_userdata.0.sunny-tripower.no-sna.GridMs_TotW

tariffs:
  grid:
    type: template
    template: awattar
    region: DE
    charges: 0.17697
    tax: 0.19
    formula: math.Max((price * (1 + tax) + charges), 0.0)

chargers:
  name: wallbox1
  type: template
  template: go-e-v3
  host: 192.168.178.25

vehicles:
  - name: vehicle1
    type: template
    template: cupra
    title: CUPRA Born
    user: ***
    password: ***
    capacity: 58

site:
  title: Home
  meters:
    grid: grid1
    pv: pv1
    battery: bat1

loadpoints:
- title: Garage
  charger: wallbox1
  vehicle: vehicle1
```

### Log details

```text
The evcc log is probably not helpful for a client-side Javascript issue.
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0 (0a833918)

andig, 2025-01-27:
Not sure that‘s a relevant platform to support?

naltatis, 2025-01-27:
Easy fix and in line with the rest of our number formatting.

closed 2025-01-27 by andig