evcc-io/evcc #18479: Fronius Verto support

Astretowe, 2025-01-28:
**Is your feature request related to a problem? Please describe.**
evcc already supports Fronius inverters in general.
The "Verto" series mostly works using template: fronius-gen24.
However, the Verto series features 4 MPPTs.
At the same time, evcc polls data from MPPT1+2 only, which seems to be in line with the available data according to the Fronius documentation I could find.

This leads to a situation where the displayed PV output is not correct (missing MPPT3+4).
In turn, the price calculation seems to be incorrect.
Calculation of charging limits doesn't seems to be affected as that seems to be based on meter values, so this is only a minor or cosmetical issue.

**Describe the solution you'd like**
I have noticed that the sunspec document for the Fronius inverters, one of the fields is "DC Power" and seems to hold the sum of the power of each individual MPPT. This should be a catch-all solution to support both, existing 2MPPT models and as well as the 4MPPT Verto models.

andig, 2025-01-29:
Happy to take a PR.

premultiply, 2025-02-13:
Looks just like an standard SunSpec-compatible pv-only inverter. This is no hybrid inverter that does require special handling for battery path.
Just use the SunSpec inverter template.
https://docs.evcc.io/docs/devices/meters#sunspec-wechselrichter

Feel free to comment again if this does not fit.

closed 2025-02-13 by premultiply

andig, 2025-02-13:
@premultiply sollten wir den WR dann- analog OCPP- dort im Template mit hinzufügen? Für nicht-Techies ist das ja nicht offensichtlich?