evcc-io/evcc #18481: Ladeplanung wird nicht aktualisiert bei neuem SoC

artner1988, 2025-01-29:
### Describe the bug

Folgende Konfiguration:

evcc bekommt den SoC des Fahrzeugs beim Einstecken (also wenn durch Homeassistant erkannt), sowie alle 15 Minuten per MQTT (über Homeassistant) mitgeteilt. Jetzt kann es passieren das Homeassistant noch den alten Wert vom wegfahren gespeichert hat (meistens 99%) und diesen beim Einstecken übermittelt. spätestens beim nächsten übermitteln kommt dann aber der richtige Wert.

Problem: wenn die Ladeplanung einmal etwas geplant hat, dann wird das nicht mehr durch einen neuen SoC geändert - dieser ändert sich auf der Hauptseite aber auch nicht mehr. In der Konfiguration sieht man aber dass EVCC sehr wohl den aktuellen SoC weiß.
Man kann eine Richtigstellung dadurch provozieren, dass man kurz lädt (Min+PV oder schnell), dann aktualisiert er den SoC auch auf der Hauptseite und der Ladeplan wird aktualisiert

[image]

[image]

[image]

[image]

### Steps to reproduce

1.Ladeplan aktiv
2.SoC wird übermittelt (per MQTT)
3.neuer SoC wird übermittelt - nichts ändert sich am Ladeplan
...

### Configuration details

```yaml
[main  ] INFO 2025/01/29 08:47:38 evcc 0.133.0
[main  ] INFO 2025/01/29 08:47:38 using config file: /root/evcc.yaml
[db    ] INFO 2025/01/29 08:47:38 using sqlite database: /root/.evcc/evcc.db

Fehlermeldung:

    failed configuring loadpoints: missing loadpoints

Konfiguration (/root/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.1.55  
  port: 502  
  name: grid1

site:
  title: Mein Zuhause
  meters:
    grid: grid1

Version: `0.133.0`
```

### Log details

[... 118 lines cut ...]
Jan 29 05:16:43 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:16:43 charge power: 2425W
Jan 29 05:16:43 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:16:43 detected active phases: 3p
Jan 29 05:16:43 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:16:43 charger status: C
Jan 29 05:16:43 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:16:43 soc estimated: 100.00% (vehicle: 98.50%)
Jan 29 05:16:43 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:16:43 vehicle soc: 100%
Jan 29 05:17:13 EVCC evcc[23397]: [site  ] DEBUG 2025/01/29 05:17:13 ----
Jan 29 05:17:13 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:17:13 charge power: 2436W
Jan 29 05:17:13 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:17:13 detected active phases: 3p
Jan 29 05:17:13 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:17:13 charger status: C
Jan 29 05:17:13 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:17:13 soc estimated: 100.00% (vehicle: 98.50%)
Jan 29 05:17:13 EVCC evcc[23397]: [lp-1  ] DEBUG 2025/01/29 05:17:13 vehicle soc: 100%
Jan 29 05:17:43 EVCC evcc[23397]: [site  ] DEBUG 2025/01/29 05:17:43 ----
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

mfuchs1984, 2025-01-29:
Evcc fragt den SoC standardmäßige nur beim Anstecken und während eines Ladevorgangs ab. Um das zu ändern, musst du den poll mode des loadpoints auf connected konfigurieren. Besser wäre aber, direkt den korrekten SoC bereitzustellen.

artner1988, 2025-01-29:
Ich hätte versucht es laut Doku anzupassen - aber die Änderung ist offensichtlich falsch

vorher - funktioniert:
vehicles:
  - name: byd
    type: custom
    title: BYD Atto 3 # display name for UI
    capacity: 60.48 # battery capacity (kWh)
    soc: # battery soc (%)
      source: mqtt # plugin type
      topic: evcc/soc1
-----------------------
nachher - funktioniert nicht:
vehicles:
  - name: byd
    type: custom
    title: BYD Atto 3 # display name for UI
    capacity: 60.48 # battery capacity (kWh)
    soc: # battery soc (%)
      source: mqtt # plugin type
      topic: evcc/soc1
      poll:
        mode: connected
        interval: 5m

mfuchs1984, 2025-01-29:
Was bedeutet "funktioniert nicht" genau?

mfuchs1984, 2025-01-29:
Ah,ist auch falsch

```
soc:
  poll:
    mode: connected
    interval: 5m
```

artner1988, 2025-01-29:
danke - jetzt funktionierts wie erwartet - danke für die Hilfe

mfuchs1984, 2025-01-29:
Dann schließe bitte das issue.

closed 2025-01-29 by premultiply

loryanstrant, 2025-04-07:
Thank you. After spending way too many hours trying to make various BT & BLE OBD work, I've given up and just ordered the WiCAN unit. :-)