evcc-io/evcc #18482: Target Charging behavior does not respect 1p 3p settings

michaelheiml, 2025-01-29:
### Describe the bug

Hi,

Problem: when target charging is set, the start time is **always** calculated based on the 1p ampere setting, even if 3p instead of 1p charging is selected. This leads to the unwanted behavior, that charging starts, runs for a while (battery heats up etc.), then suddenly stops and continues at a much later point (again heating up battery etc.) to finish at the right time.

Example (see screenshots): in the evening, I set target charging to next day morning 8am. It is required to charge ~20 to 25 kWh.
**With 1p behavior is:** charging starts at 2am, charges until 8am @5kW (average) = 6h * 5kW = 30kWh (this is OK, but with a lot of waste/loss of energy due to slow and long charging)
**With 3p behavior is:** charging starts at 2:30am, charges ~1h @10kW (average) = ~10kWh, then stops, then starts again at 6am and charges until 7am = ~1h @ 10kW = ~10kWh => all in all ~20kWh but broken up in 2 charging sessions, each one heating up and cooling down the battery (not good).

What should happen IMHO is that with 3p, charging starts at around 5am until 7am average @12kW = 2h * 12kW = 24kWh in one piece.

I have: evcc + go-E + Zoe 41kW 2017.

See attached screenshots and notice that even though 3p is selected in the settings menu, evcc calculates the start time as if 1p is selected:
evcc charge sessions: green columns are the 2 charging sessions from 2:30-3:30 and 6:00-7:00 (approximately)
evcc charging: see that evcc actually charges with 3p @ 9,5 kW (battery is almost full therefore not full 17kW)
evcc plan: is set as described, notice that "charging time" calculates with "@5,8KW" although in the (next) setting screen, 3p is selected
evcc settings: notice that 3p (4,8-17,3kW) is selected

[image]
[image]
[image]
[image]

### Steps to reproduce

1. set target charging to 8am morning with a quite empty battery
2. manually set to 3p charging (make sure 3p can charge with much more Ampere than 1p)
3. notice that target charging starts much too early because it calculates start time as if 1p was selected

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: 192.168.1.110  
  port: 502  
  name: grid1
- type: template
  template: fronius-gen24 
  usage: pv  
  host: 192.168.1.110  
  port: 502  
  name: pv2
- type: template
  template: fronius-gen24 
  usage: battery  
  host: 192.168.1.110  
  port: 502  
  name: battery3

chargers:
- type: template
  template: go-e 
  host: 192.168.1.10  
  name: wallbox4

vehicles:
- type: template
  template: renault 
  title: Anja  
  user: xxx
  password: xxx
  vin: xxx  
  capacity: 41  
  name: ev3
  mincurrent: 7
  maxcurrent: 25
  phases: 1
- type: template
  template: renault 
  title: Michael  
  user: xxx
  password: xxx
  vin: xxx  
  capacity: 41  
  name: ev4
  mincurrent: 7
  maxcurrent: 25
  phases: 1

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv

site:
  title: Hattenberg
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2868 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.12464 # [currency]/kWh
```

### Log details

```text
Does not apply
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

michaelheiml, 2025-01-29:
If I remember right, I set phases:1 because I want to prevent automatic phase switching due to the PV. The PV delivers 5-8kW on average, so a lot of switching between 1p and 3p occurs. It is better to only charge with 1p (there have been quite some discussions here in the past about this).

So do I understand this right: If I remove phases:1, I have the constant phase switching issue. If I leave phases:1, target charging does not calc time correct. How do I handle this?

mfuchs1984, 2025-01-29:
I think, you could change the loadpoints enable properties to reduce the number of phase switches 
https://docs.evcc.io/docs/reference/configuration/loadpoints#enable

michaelheiml, 2025-01-29:
Thanks for the link but sorry, I do not understand.
So I should remove phases:1 and then add:
enable:
  threshold: 0
  delay: 1m

but with which value to achieve this:
PV charge only with 1p, if there is more than 5,6kW available export the rest to the grid (cap charging at 5,6kW + do not switch to 3p charging).

mfuchs1984, 2025-01-29:
The larger the enable delay, the longer evcc waits before switching to 3 phases when the power changes. This will avoid switching back an forth when power is around the threshold for 3 phases. Same for disable but  for switching back to 1 phase.

closed 2025-01-29 by andig