evcc-io/evcc #18484: Stromtarif electricitymaps-free liefert unplausible Daten im Planer

An-Wi, 2025-01-29:
### Describe the bug

Eingestellt wie in Doku beschrieben bekomme ich folgende Anzeige mit "32.700 ct/kWh" im Planer:

[image]

Den Token für electricitymaps-free habe ich bereits zum x-ten mal neu generiert - zuletzt mit Anmeldung via Email und Konfiguration der Zone:

[image]

Meine vorherige Konfiguration mit `co2: | type: template | template: grünstromindex` und `planner: | type: fixed | price: ...` hatte problemlos funktioniert.

Im Log konnte ich keine Fehlermeldung bzgl. "creating tariff co2 failed" feststellen.

Das Problem scheint mit dem aktuellen Netz-CO2-Wert zu korrelieren, denn dort stand letzte Stunde 327g.
Jetzt ist er auf 306g und im Planer wird "**30.600 ct/kWh**" angezeigt!

[image]

### Steps to reproduce

1. evcc.yaml von co2 "grünstromindex" auf "electricity-maps" umgestellt
2. Neustart evcc
3. Planer aufgerufen
...

### Configuration details

```yaml
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.2797       #0.2961  #0.2797  #0.312  #0.355 # EUR/kWh
#    zones:
#      - days: Mo-So
#        hours: 0-6
#        price: 0.2797   #0.2961  #0.2797  #0.312  #0.2455 # EUR/kWh
#      - days: Mo-So
#        hours: 22-0
#        price: 0.2797   #0.2961  #0.2797  #0.312  #0.2455 # EUR/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1427 # EUR/kWh

    # type: octopusenergy
    # tariff: AGILE-FLEX-22-11-25 # Tariff code
    # region: A # optional
  co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
#    type: template
#    template: grünstromindex # GrünStromIndex (Germany only)
#    zip: "07552"

    type: template
    template: electricitymaps-free
    token: zY********gO
    zone: DE

    # type: electricitymaps # https://app.electricitymaps.com/map
    # uri: <uri>
    # token: <token>
    # zone: DE

#planner:
#    type: fixed
#    price: 0.2797
[... 53 lines cut ...]
[site  ] INFO 2025/01/29 09:57:13     vehicle 1: range âœ“ finish âœ“ status âœ“ climate âœ— wakeup âœ—
[site  ] INFO 2025/01/29 09:57:13     vehicle 2: range âœ— finish âœ— status âœ— climate âœ— wakeup âœ—
[lp-1  ] INFO 2025/01/29 09:57:13 loadpoint 1:
[lp-1  ] INFO 2025/01/29 09:57:13   mode:        pv
[lp-1  ] INFO 2025/01/29 09:57:13   charger:     power âœ“ energy âœ“ currents âœ“ phases âœ“ wakeup âœ—
[lp-1  ] INFO 2025/01/29 09:57:13   meters:      charge âœ“
[lp-1  ] INFO 2025/01/29 09:57:13     charge:    power âœ“ energy âœ“ currents âœ“
[lp-1  ] DEBUG 2025/01/29 09:57:13 phase timer inactive
[lp-1  ] DEBUG 2025/01/29 09:57:13 pv timer inactive
[lp-1  ] INFO 2025/01/29 09:57:13 vehicle updated: unknown -> Ariya
[site  ] WARN 2025/01/29 09:57:13 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2025/01/29 09:57:13 ----
```

### What type of operating system or environment does evcc run on?

Windows

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

andig, 2025-01-29:
Was sagt

    evcc tariff --log trace,db:error

andig, 2025-01-29:
@naltatis warum kommen wir hier auf ct/kWh? Falscher `tariff` im Template?

andig, 2025-01-29:
Closed in https://github.com/evcc-io/evcc/commit/86c66080b3660b09cf10e4c44755aadff7623ec7

closed 2025-01-29 by andig