evcc-io/evcc #18494: Tesla: vehicle unavailable: vehicle is offline or asleep

marcsowen, 2025-01-30:
### Describe the bug

Since switching to the new Tesla Fleet API via Teslamate, evcc is unable to start charging the vehicle when it's sleeping. It works perfectly when the car is awake. I get the following error:

`ERROR 2025/01/30 09:57:33 max charge current 32A: 500 Internal Server Error: Server error: `POST https://client-tdtpzamudtd-teslamate-cmd:4443/api/1/vehicles/5YJ3E7EB1MF971033/command/set_charging_amps` resulted in a `500 Internal Server Error` response: {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
`
After entering this command, the vehicle wakes up and evcc continues to work as expected:

`$ sudo evcc vehicle --log trace,db:error --wakeup
[main  ] INFO 2025/01/30 09:59:09 evcc 0.133.0
[main  ] INFO 2025/01/30 09:59:09 using config file: /etc/evcc.yaml
[tesla ] TRACE 2025/01/30 09:59:09 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2025/01/30 09:59:10 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2025/01/30 09:59:10 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2025/01/30 09:59:10 {"response":[{"id":1493004313417487,"vehicle_id":211580603406,"vin":"5YJ3E7EB1MF971033","color":null,"access_type":"OWNER","display_name":"LG S 322E","option_codes":"AD15,AF00,APFB,APH4,AU3P,BC3B,BT42,CDM0,CH15,CODE,DRLH,DV4W,FC02,FG31,FM3B,GLFR,HL32,HM30,ID3W,IL31,LT03,MDL3,MR31,PMNG,PC30,REEU,RF3G,RS3H,S3PB,SA3P,SC04,STCP,SU3C,T3HS,TM00,TW00,UT3P,W41B,WR00,ZINV,MI02,PL30,SLR0,ST33,BG31,I36M,OSSB,AUF1,RSF1,ILF1,FGF1,CPF1,P3WS,HP31,PT01,RL31,FD00,RD01,SWF1,RA01","granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"1493004313417487","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"pagination":{"previous":null,"next":null,"current":1,"per_page":1,"count":1,"pages":1},"count":1}
[tesla ] TRACE 2025/01/30 09:59:10 POST https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/5YJ3E7EB1MF971033/wake_up
[tesla ] TRACE 2025/01/30 09:59:10 {"response":{"id":1493004313417487,"user_id":220718464065,"vehicle_id":211580603406,"vin":"5YJ3E7EB1MF971033","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"1493004313417487","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}}`

### Steps to reproduce

1. Wait for the Tesla vehicle to fall asleep
2. Start charging via evcc
3. The error response 

### Configuration details

```yaml
$ sudo evcc dump --cfg
[sudo] password for sowen: 
[main  ] INFO 2025/01/30 10:32:07 evcc 0.133.0
[main  ] INFO 2025/01/30 10:32:07 using config file: /etc/evcc.yaml
[db    ] INFO 2025/01/30 10:32:07 using sqlite database: /var/lib/evcc/evcc.db

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: https
  host: evcc.xxx.de # .local suffix announces the hostname on MDNS
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.178.104  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
[... 105 lines cut ...]

---

$ sudo evcc vehicle --log trace,db:error --wakeup
[main  ] INFO 2025/01/30 09:59:09 evcc 0.133.0
[main  ] INFO 2025/01/30 09:59:09 using config file: /etc/evcc.yaml
[tesla ] TRACE 2025/01/30 09:59:09 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2025/01/30 09:59:10 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2025/01/30 09:59:10 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2025/01/30 09:59:10 {"response":[{"id":1493004313417487,"vehicle_id":211580603406,"vin":"5YJ3E7EB1MF971033","color":null,"access_type":"OWNER","display_name":"LG S 322E","option_codes":"AD15,AF00,APFB,APH4,AU3P,BC3B,BT42,CDM0,CH15,CODE,DRLH,DV4W,FC02,FG31,FM3B,GLFR,HL32,HM30,ID3W,IL31,LT03,MDL3,MR31,PMNG,PC30,REEU,RF3G,RS3H,S3PB,SA3P,SC04,STCP,SU3C,T3HS,TM00,TW00,UT3P,W41B,WR00,ZINV,MI02,PL30,SLR0,ST33,BG31,I36M,OSSB,AUF1,RSF1,ILF1,FGF1,CPF1,P3WS,HP31,PT01,RL31,FD00,RD01,SWF1,RA01","granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"1493004313417487","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"pagination":{"previous":null,"next":null,"current":1,"per_page":1,"count":1,"pages":1},"count":1}
[tesla ] TRACE 2025/01/30 09:59:10 POST https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/5YJ3E7EB1MF971033/wake_up
[tesla ] TRACE 2025/01/30 09:59:10 {"response":{"id":1493004313417487,"user_id":220718464065,"vehicle_id":211580603406,"vin":"5YJ3E7EB1MF971033","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"1493004313417487","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}}
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.133.0

closed 2025-01-30 by marcsowen

horst789github, 2025-01-31:
was war die lösung? habe das gleiche problem

mucki12, 2025-02-02:
@horst789github 
Denke mal die Lösung war die nightly zu benutzen, in der dieser Fehler gefixt sein sollte. https://github.com/evcc-io/evcc/issues/18400