evcc-io/evcc #18495: DNS Fehler bei Cupra SOC Abruf via openWB

martin-zh, 2025-01-30:
### Describe the bug

Cupra SOC Abfrage via openWB funktioniert derzeit nicht mit DNS Fehler:

Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 110, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Get "https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/xxx/vehicles/xxx/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-01-30T12:03:15.346947924+01:00", grpc_status:2, grpc_message:"Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/xxx/vehicles/xxx/mycar\": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving"}"
>

Test via nslookup kann den Host ola.prod.code.seat.cloud.vwgroup.com ab auflösen

### Steps to reproduce

Soc via openWB abrufen

### Configuration details

```yaml
openWB korrekt konfiguriert
```

### Log details

```text
Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v5/users/xxx/vehicles/xxx/mycar\": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving"}
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-30:
Ich starte den Server mal durch- schauen wir obs dann geht.

martin-zh, 2025-01-31:
Der Fehler ist nach wie vor vorhanden.

andig, 2025-01-31:
I‘ll need to open provider ticket

martin-zh, 2025-01-31:
Danke - das Problem scheint mit dem Troubleshooting des Providers gelöst zu sein. SoC Abfrage ist nun erfolgreich