evcc-io/evcc #18503: Wrong PV readings from Solax X3-Ultra

spn91, 2025-01-30:
### Describe the bug

PV readings for the Solax X3-Ultra (Gen5 Hybrid inverter) are not reproducible. You can compare the readings from the screenshot and the readings from the logs. The screenshots are from the Home Assistant Solax Modbus integration. They are similar to the readings from the Solax Cloud. 

Both, the HA Solax integration and evcc are reading the values from the HA modbus proxy addon. It seems that evcc reads a wrong register, or is doing some wrong calculations (if the value is calculated by evcc). I have only a simple site without any other smart meter or load on this site yet, as you can see in the configuration. Also, the bigger the PV power, the bigger the difference between the readings from HA and evcc gets.

Unfortunately the nightly version isn't provided via HA, so please excuse me that I wasn't able to test it with the nightly version.

[image]

### Steps to reproduce

Nothing special needed, just using the Solax X3-Ultra inverter, which is one of the Solax Gen5 Hybrid inverters.

### Configuration details

```yaml
## minimal configuration example
log: info
levels:
  lp-1: trace
  site: trace
site:
  title: Home # display name for UI
  meters:
    pv:
      - solax_x3_ultra

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Parkplatz # display name for UI
    vehicle: tesla_m3 # default vehicle
    charger: twc3

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  # replace with your real grid meter
  # replace with your real pv system
  - name: solax_x3_ultra
    type: template
    template: solax
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.1 # Hostname
    port: 502 # Port
  # replace with your real battery
  - name: my_pv
    type: template
    template: demo-meter
    usage: pv
    power: 7000 # 4 kW production

chargers:
  - name: twc3
    type: template
    template: twc3
    host: 192.168.2.1:8080 # IP-Adresse oder Hostname

# replace with your real vehicle (optional)
# see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: tesla_m3
    type: template
    template: tesla
    title: Tesla Model 3
    capacity: 60 # Akkukapazität in kWh (optional)
    clientId: aaa # Kunden ID deiner [Tesla Developer App](https://developer.tesla.com/dashboard).
    accessToken: aaa # von [myteslamate.com](https://app.myteslamate.com/).
    refreshToken: aaa # von [myteslamate.com](https://app.myteslamate.com/).
    proxyToken:  aaaa

# enter your real grid tariff and feed-in price
# see https://docs.evcc.io/docs/devices/tariffs

# mqtt broker
mqtt:
  broker: 192.168.1.1:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: evcc
  password: evcc
```

### Log details

```text
[site  ] DEBUG 2025/01/30 15:30:23 ----
[lp-1  ] DEBUG 2025/01/30 15:30:23 charge power: 0W
[lp-1  ] DEBUG 2025/01/30 15:30:23 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/30 15:30:23 pv 1 power: 4647W
[site  ] DEBUG 2025/01/30 15:30:23 site power: -4647W
[lp-1  ] DEBUG 2025/01/30 15:30:23 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2025/01/30 15:30:23 charger status: A
[site  ] DEBUG 2025/01/30 15:30:33 ----
[lp-1  ] DEBUG 2025/01/30 15:30:33 charge power: 0W
[lp-1  ] DEBUG 2025/01/30 15:30:33 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/30 15:30:33 pv 1 power: 4570W
[site  ] DEBUG 2025/01/30 15:30:34 site power: -4570W
[lp-1  ] DEBUG 2025/01/30 15:30:34 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2025/01/30 15:30:34 charger status: A
[site  ] DEBUG 2025/01/30 15:30:43 ----
[lp-1  ] DEBUG 2025/01/30 15:30:43 charge power: 0W
[lp-1  ] DEBUG 2025/01/30 15:30:43 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/30 15:30:43 pv 1 power: 4610W
[site  ] DEBUG 2025/01/30 15:30:43 site power: -4610W
[lp-1  ] DEBUG 2025/01/30 15:30:43 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2025/01/30 15:30:43 charger status: A
[site  ] DEBUG 2025/01/30 15:30:53 ----
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133.0

spn91, 2025-01-31:
Some more infos regarding inverter firmware

ARM: 018.02-000.09
DSP: 017.05

spn91, 2025-02-05:
Looking at the Code from the Solax HA Modbus Integration I found out that they are using 

register=0x6C for inverter power L1
register=0x70 for inverter power L2
register=0x74 for inverter power L3

and then calculating the sum of the three values to get the total inverter power. Searching in the solax template for these registers, I'm unable to find them, even when converting the hex values to decimal representation.

Comparing the values from evcc and HA, it seems that one of the three phases is not counted in evcc.

andig, 2025-02-09:
MPPT ist the only way to get PV (production!) power for hybrid inverters. Without further details available I‘m afraid there‘s nothing we can do.

spn91, 2025-02-09:
Hmm, but MPPT power is not the „real“ power, which is exported to the house. And as I said, there is one MPPT missing in evcc. The Ultra has 3 of them.

andig, 2025-02-09:
It‘s an approximation. There is no other „total“ AC power for hybrid inverters.

andig, 2025-02-09:
Do you want to open a PR for the third mppt?

spn91, 2025-02-09:
What’s with the registers I have posted above?

Yes, i will open a PR

premultiply, 2025-02-13:
Please ask the vendor support for the current Modbus Protocol document for you inverter model and attach it here.

premultiply, 2025-02-16:
Fixed by #18761

closed 2025-02-16 by premultiply