evcc-io/evcc #18513: UI Loadpoints: deprecated: soc setting is ignored, please remove

TobiasHuber1980, 2025-01-31:
feedback zu <https://github.com/evcc-io/evcc/pull/18137>

weiter konfigurieren geht.
kommt wohl daher:
`[main ] INFO 2025/01/31 13:30:09 evcc 0.133.0 (864c2cf5)`
`[main ] INFO 2025/01/31 13:30:09 using config file: /etc/evcc.yaml`
`[db ] INFO 2025/01/31 13:30:09 using sqlite database: /var/lib/evcc/evcc.db`
`[main ] INFO 2025/01/31 13:30:10 listening at :7070`
`[lp-1 ] WARN 2025/01/31 13:30:12 deprecated: soc setting is ignored, please remove`

[Slack Message](https://evccgroup.slack.com/archives/D05KF6DEPQC/p1738326529250369)

andig, 2025-01-31:
@TobiasHuber1980 bitte kein Suchspiel. Steps to reproduce?

TobiasHuber1980, 2025-01-31:
0.133.0+1738324237 installieren
(Experimental) Konfiguration -> neuen Loadpoint hinzufügen -> openWB Pro auswählen -> IP-Adresse eintragen, auf prüfen klicken.
Dann kommt :
```
Status: fehlgeschlagen
soc: not available
```

andig, 2025-01-31:
> deprecated: soc setting is ignored, please remove

@naltatis irgendwoher kommt hier soc.poll in `migrateSettings`.

naltatis, 2025-02-03:
Die Deprecations haben mit dem beschriebenen Fehler der OpenWB Pro nichts zu tun. Sind aber irreführend.

Zum Thema: @andig du hast doch eine OpenWB Pro. Kannst du mal prüfen wieso der "soc: not available" Fehler beim Test fliegt?

naltatis, 2025-02-09:
Der Part mit den Deprecation Warnings sollte mit https://github.com/evcc-io/evcc/pull/18638 gefixt sein. 
Bleibt nur noch die Frage nach dem SoC Fehler bei der OpenWB Pro.

andig, 2025-02-10:
Der not available Fehler MUSS ignoriert werden. Den werfen Apis wenn sie temporär erwartbar nicht funktionieren. Im Backend werden die unterdrückt, UI/Api müssen das auch tun.

andig, 2025-02-10:
Der zweite Fehler ist, dass fehlerhafte Autos nicht abgespeichert werden können. Das muss auch gehen, ebenso andere Geräte.

closed 2025-02-10 by andig