evcc-io/evcc #18516: UI: Incorrect phases visualization

naltatis, 2025-01-31:
### Describe the bug

UI shows 3 phases but only one is used.

Likely a regression from https://github.com/evcc-io/evcc/pull/18137

[image]

### Steps to reproduce

-

### Configuration details

```yaml
{
        "batteryBoost": false,
        "chargeCurrent": 6,
        "chargeCurrents": [
          0.007000000216066837,
          0.009999999776482582,
          5.945000171661377
        ],
        "chargeDuration": 8133,
        "chargePower": 1328,
        "chargeRemainingDuration": 39414,
        "chargeRemainingEnergy": 21666.667,
        "chargeTotalImport": 22076.107,
        "chargedEnergy": 5286.452,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": null,
        "chargerPhases1p3p": true,
        "chargerSinglePhase": false,
        "chargerStatusReason": "unknown",
        "charging": true,
        "connected": true,
        "connectedDuration": 0,
        "disableDelay": 120,
        "disableThreshold": 700,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 20,
        "effectiveMinCurrent": 6,
        "effectivePlanId": 2,
        "effectivePlanSoc": 50,
        "effectivePlanTime": "2025-02-01T08:00:00+01:00",
        "effectivePriority": 2,
        "enableDelay": 120,
        "enableThreshold": -700,
        "enabled": true,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 20,
        "minCurrent": 6,
        "mode": "pv",
        "phaseAction": "inactive",
        "phaseRemaining": 0,
        "phases": 0,
        "phasesActive": 1,
        "phasesConfigured": 0,
        "phasesEnabled": 1,
        "planActive": false,
        "planEnergy": 0,
        "planOverrun": 0,
        "planProjectedEnd": null,
        "planProjectedStart": null,
        "planTime": null,
        "priority": 2,
        "pvAction": "disable",
        "pvRemaining": 119,
        "sessionCo2PerKWh": 0.954,
        "sessionEnergy": 5286.452,
        "sessionPrice": 0.6,
        "sessionPricePerKWh": 0.113,
        "sessionSolarPercentage": 98.166,
        "smartCostActive": false,
        "smartCostLimit": null,
        "smartCostNextStart": null,
        "title": "Vorne",
        "vehicleClimaterActive": null,
        "vehicleDetectionActive": false,
        "vehicleLimitSoc": 80,
        "vehicleName": "tesla3",
        "vehicleOdometer": 0,
        "vehicleRange": 261,
        "vehicleSoc": 54,
        "vehicleWelcomeActive": false
      },
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

nightly

andig, 2025-01-31:
Hier werden ja 3 genutzt, aber eben nur gering?

alex90alex, 2025-02-01:
Das Problem habe ich seit der aktuellen nightly auch. Ich verwende eine easee wallbox. Beim Ladevorgang wird auch im einphasigen Betrieb das dreiphasige Laden in evcc visualisiert.

Tombra1889, 2025-02-01:
Aus meiner Sicht sagt die Grafik etwas anders aus. Es wird zwar auf einer Phase geladen, daher ist der Balken auf der rechten Seite auch etwas dunkler, es steht aber ein "Überschuss" zumindest in der Visualisierung zur Verfügung, der aktuell auf den anderen Phasen als mögliche Energie dargestellt wird. Eigentlich müsste hier die mögliche Energie zunächst der aktiven Phase zugeordnet werden.

andig, 2025-02-02:
Hier gehts nicht um zugeordnete Energie sondern gemessene Ströme. Und die sind real >0.

andig, 2025-02-02:
@naltatis was für ein Charger ist das? Kann man das mit einem Demo Charger mit alter/neuer Version prüfen?

naltatis, 2025-02-03:
Beobachtung: Wir haben den Speicherort der Phasen mit dem https://github.com/evcc-io/evcc/pull/18137 verändert. Vorher wurden Phasenänderungen eines yaml-konfigurierten LPs via API/UI in `settings.lp1.phasesConfigured` gespeichert. Mit der Änderung ist der neue Speicherort settings.lp1.phases`. Das führt dazu dass eine vorher gemacht Einstellung mit dem Update auf die neue Version verloren geht. Hier müssen wir entweder eine Migration bauen oder die Änderung zurücknehmen.

naltatis, 2025-02-05:
Ok, seems like we have two different effects here:

**1. changed easee behavior since firmware v328**

> New phase priority logic: Optimized load balancing for better performance. The load balancer now prioritizes L1-L3 (instead of L2-L3) for single-phase charging on IT 3-phase systems and N-L3 (instead of N-L1) for single-phase charging on TN 3-phase systems.

https://release.easee.com/firmware/chargers/v328

**2. breaking change in https://github.com/evcc-io/evcc/pull/18137 that leads to loss of user preference** 

See comment above.

Let's focus on **1.** in this issue. I'll create a new one for *2.*.

naltatis, 2025-02-05:
@alex90alex Can you confirm that you're running v328 and can you check if this effect also happens with the stable evcc release

closed 2025-02-05 by andig

premultiply, 2025-02-06:
To vehicle only pin L1 (1p) or L1+L2+L3 (3p) is allowed.
Easee can't change that or they may destroy OBCs as it is not allowed in anyway else.

I think they are talking ONLY about the grid side of the charger if this is related at all.

naltatis, 2025-02-06:
I don't know if this change is grid side only. But I'll check the Tesla api next time this happens. I think it also shows phase currents.