evcc-io/evcc #18522: Charger installation phase rotation configure

Habazor, 2025-01-31:
Hello,

When multiple charger are installed on same power supply there is need to install charger different phase.

Chargers that dont have dynamic phase swiching needs done this by installing charger to different phase.

Examble Mennekes or Etrel from modbus charger are sending phase values always on same order L1-L2-L3.

Charger connector:  CP1: L1-L2-L3  CP2: L1-L2-L3  CP3: L1-L2-L3

Installation phase  :  CP1: **L1**-L2-L3  CP2: **L3**-L1-L2  CP3: **L2**-L3-L1

On this examble CP2 modbus L1 value, load is on L3 phase and CP3 modbus L1 value load is on L2. 

To get loadbalance working there is need to configure load to right installation phase.

andig, 2025-02-01:
How is this related to evcc functionality? /cc @premultiply

Habazor, 2025-02-01:
> How or why would this be related to evcc functionality? /cc [@premultiply](https://github.com/premultiply)

Here is two exambles how this impact on evcc loadmanagement:

**1. Chargers are on same nested circuit (garage)**

If we have three chargers on same circuit and all have 1p 16A car charging.

Circuit maxcurrent: 20A

From charger evcc gets all load values (16A) to L1 = 48A and will limit charging currents to all chargers 20/3=6A

When chargers has installed to different phases real load to circuit is 16 / 16 / 16 A and there is no need to limit current.

**2. Chargers are connected to circuit with external load (house)**

If we have three chargers with two 1p 16A car charging and one 3p 16A

Circuit maxcurrent: 35A

External load is 16 / 5 / 5

In this situation CP1/CP2 chargers (1p/3p) has load on same phase L1 and have 19A / 2 = 9,5A current to use.

CP3 charger 1p 16A charging real load is on L3 so no need to limit this charger.

andig, 2025-02-07:
We currently don‘t support phase-accurate currents. Maybe at a later time.

closed 2025-02-07 by andig