evcc-io/evcc #18538: Timeout der Sonnebatterie im Rahmen der nächtlichen Wartung stoppt Netzladung

marleyan333, 2025-02-01:
### Describe the bug

Es geht mir nochmal, um das bekannte Problem mit der sonnenbatterie und Netzladung. Diese wird jede Nacht um 2:49 Uhr aufgrund der systemeigenen Wartung abgebrochen und nicht wieder aufgenommen, wodurch meine Batterie nur bis maximal 79% geladen wird. 

Bisher gab es dafür in evcc keinen workaround. 

Bringe diese Diskussion noch mal hoch, nachdem ich mal die logfiles DeepSeek habe prüfen lassen und mir Lösungsvorschläge für das Problem der nächtlichen Wartungen durch Sonnen und anschließende Timeout-Fehler anzeigen lassen habe. Die von DeepSeek präsentierte Lösung sieht so aus:

„Das Problem lässt sich durch eine Zeitverschiebung der EVCC-Abfrageintervalle (z.B. 5-Minuten-Puffer nach Sonnen-Wartungsfenstern) in der Cron-Konfiguration entschärfen.“

jetzt sagt mir eine Cron-Konfiguration gar nichts, aber ich denke mal euch Profis schon. DeepSeek bietet sogar eine konkreten Code-Vorschlag:

Zeitversatz einfügen
Verschiebung um 5 Minuten nach dem Wartungsfenster (02:50 → 02:55):
50 2 * * * /usr/bin/evcc --cmd "trigger update"
55 2 * * * /usr/bin/evcc --cmd "trigger update"
Alternativ eine Pufferzonen-Konfiguration:
Für variable Wartungszeiten (z.B. 02:45-03:15) einen Zeitpuffer mit Zufalls-Offset:

Jede Nacht zwischen 03:20-03:25 starten (35-40 Min Puffer)

20-25 3 * * * sleep $((RANDOM % 300)) && /usr/bin/evcc --cmd "reload"

Wie gesagt, ich bin Laie, vielleicht steht da oben auch totaler Blödsinn. Dachte aber nen Versuch,euch das zu zeigen, wäre es wert.

### Steps to reproduce

1. Netzladung mit sonnenbatterie um 2:49
2. abbruch wegen systemwartung durch sonnen
2. timeout-Fehler verhindert Ansteuerung durch evcc
3. ladubg wird auch nach Abschluss der Wartung nicht mehr automatisch aufgenommen 
3.
...

### Configuration details

```yaml
So
```

### Log details

```text
So
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.133

andig, 2025-02-01:
Du kannst das durch externe Autoamtisierung lösung. Dass Sonnen die Kundeneinstellung hier grob unterbricht und dann vergisst ist ein Fehler den der Hersteller beheben sollte.

andig, 2025-02-01:
PS.: die artificial-stupidity Vorschläge sind frei erfunden und vollkommen nutzlos...

closed 2025-02-01 by andig