evcc-io/evcc #18543: Vehicle status no longer shown

andig, 2025-02-02:
I have a vehicle that is not propertly connected:

    [main  ] ERROR 2025/02/02 09:21:23 creating vehicle db:1 failed: cannot create vehicle type 'template': cannot create vehicle type 'etron': cannot get vehicles: Message: Post "https://app-api.live-my.audi.com/vgql/v1/graphql": dial tcp: lookup app-api.live-my.audi.com: i/o timeout, Locations: [], Extensions: map[code:request_error], Path: []

I would expect that it shows as disconnected in the UI. However, it does not.

naltatis, 2025-02-04:
@andig steps to reproduce, please.

andig, 2025-02-04:
Vehicle with invalid credentials or broken api.

naltatis, 2025-02-05:
Creating a vehicle with invalid credentials via UI is not possible. Any other way to reproduce?

closed 2025-02-08 by andig